mongoose-os-apps/shelly-homekit #1420: Setting HomeKit slider to fully closed / open does the opposite with Shelly Plus 2PM.

tzickel, 2024-07-17:
I installed in a rolling shutter Shelly Plus 2PM and put the latest firmware 2.12.1 (20240623-075044/2.12.1-g417cbdf).

If I put the sliders in the Home app to the state "Closed" (It doesn't say 0% open), it will jump to "Opened" state (again, doesn't say 100% open). And if I slide it all the way to the other side to the "Opened" state, it jumps all the way to "Closed" state...

If I tell Siri though to close / open the blind, it works fine...

tzickel, 2024-07-17:
Here is the logs when moving from 3% open to closed, and instead if jumps to opened:
```
9288601099 shelly_main.cpp:483     Up 9288.59, HAP 0/1/16 ns 1, RAM: 210028/199268; st 50; 4.1: c:1 mp:133.81 ip:5.00 mt_ms:20190 cp:3.84 tp:3.84 md:0 lmd:1
9292897440 shelly_hap_window_c:375 WC 1: HAPSetTgtPos 1.00 cur 3.84 tgt 3.84 lmd 1
9292902880 shelly_hap_window_c:350 WC 1: Tgt pos 3.84 -> 1.00 (HAP)
9292908576 shelly_hap_window_c:330 WC 1: State: idle -> move (0 -> 20)
9292976979 shelly_output.cpp:64    Output 1: off -> on (move)
9292983029 shelly_hap_window_c:330 WC 1: State: move -> rampup (20 -> 22)
9293077634 shelly_hap_window_c:552 P = 108.74 -> 133.81
9293083702 shelly_hap_window_c:330 WC 1: State: rampup -> moving (22 -> 23)
9293578867 shelly_hap_window_c:350 WC 1: Tgt pos 1.00 -> 1.43 (fixup)
9293584021 shelly_output.cpp:64    Output 1: on -> off (moving)
9293589791 shelly_hap_window_c:330 WC 1: State: moving -> stop (23 -> 24)
9293798651 mgos_sys_config.c:323   Saved to conf9.json
9293805873 shelly_hap_window_c:330 WC 1: State: stop -> stopping (24 -> 25)
9293818826 shelly_hap_window_c:330 WC 1: State: stopping -> idle (25 -> 0)
9293829817 shelly_hap_window_c:375 WC 1: HAPSetTgtPos 0.00 cur 1.43 tgt 1.43 lmd 2
9293835654 shelly_hap_window_c:350 WC 1: Tgt pos 1.43 -> 100.00 (HAPalt)
9293841229 shelly_hap_window_c:330 WC 1: State: idle -> move (0 -> 20)
9293877098 shelly_output.cpp:64    Output 2: off -> on (move)
9293882980 shelly_hap_window_c:330 WC 1: State: move -> rampup (20 -> 22)
9293977556 shelly_hap_window_c:552 P = 107.67 -> 133.81
9293983719 shelly_hap_window_c:330 WC 1: State: rampup -> moving (22 -> 23)
9294279178 shelly_hap_window_c:340 WC 1: Cur pos 2.36 -> 2.85, P = 134.18
9295079219 shelly_hap_window_c:340 WC 1: Cur pos 6.32 -> 6.82, P = 134.28
9295879451 shelly_hap_window_c:340 WC 1: Cur pos 10.28 -> 10.78, P = 134.21
9296601488 shelly_main.cpp:483     Up 9296.59, HAP 0/1/16 ns 1, RAM: 209964/199268; st 51; 4.1: c:1 mp:133.81 ip:5.00 mt_ms:20190 cp:14.24 tp:100.00 md:1 lmd:1
```

tzickel, 2024-07-17:
Ok, I updated my poor old shelly 2.5 to newest firmware now, and it has the same bug too :(

markirb, 2024-07-17:
https://github.com/mongoose-os-apps/shelly-homekit/compare/2.10.1..master#diff-83af29c79c513510ea62e6345632c7e86a0db353bb445dc58c70803d73b24e67

this is the diff from 2.10.1. not much actually... but this should make a difference. looks like your 3 to 0 is interpreted as a tap (HAPalt) and not a target position?

tzickel, 2024-07-17:
I have no idea what it means. All I'm doing is sliding the slider in iOS 17.5.1 to the edge.

tzickel, 2024-08-23:
Ahh, the let's auto close stuff instead of fixing them way, good luck.

timoschilling, 2024-08-29:
If we look on the protocol, we can't see a bug, that's why the issue is closed, otherwise we would add a `confirmed-bug` label and the issue would stay open.

Please make a video where we can see the log, you window and the iOS UI during the change.

You can test the following thing: tell siri to open it to 25%, 50%, 75%, close and open, does that work as expected?

We often have this kind of issue and it's strange behavior of iOS. I have some original certificated Velux roller shutter and even they have the problem in some iOS versions. The shutters didn't get software updates so it must be get broken and fixed by iOS.

naanlizard, 2024-09-10:
Happening for me as well, on 5 brand new Shelly Plus 2PM flashed with the latest firmware as of this post

gmuth, 2024-09-28:
If this would be an iOS bug using (close) via the shell-homekit-webapp should work. however this does not work either. Even if iOS wound send confusing commands this should led go a wrong position state.

markirb, 2024-10-02:
I agree that this seems to be a Bug in this change quoted above:

https://github.com/mongoose-os-apps/shelly-homekit/compare/2.10.1..master#diff-83af29c79c513510ea62e6345632c7e86a0db353bb445dc58c70803d73b24e67

This needs a very deep look

markirb, 2024-10-03:
I specifically think it is this commit: https://github.com/mongoose-os-apps/shelly-homekit/commit/92795664ecb75b58f8e4e3343dc84f7a727d5ed6

So I need you help. I build a version without this, can you try if this is the cause?

[shelly-homekit-ShellyPlus2PM.zip](https://github.com/user-attachments/files/17241081/shelly-homekit-ShellyPlus2PM.zip)
[shelly-homekit-Shelly25.zip](https://github.com/user-attachments/files/17241082/shelly-homekit-Shelly25.zip)

gmuth, 2024-10-03:
Markus, thanks a lot! 
Yesterday I removed the device from homekit and reverted back to 2.11.2 for further monitoring (and looking into logs each time use open or close. I'll install this version, add the device to homekit and see what happens. In the past the issue come up once or twice a week when using automated opening and closing (which I need to setup again :-)

gmuth, 2024-10-07:
The issue still exists:

I triggered "open shutter" using HomeKit with iOS 17.7:

```
311813350965 shelly_debug.cpp:231    No log file, sending new entries
311817312869 shelly_main.cpp:477     Up 311817.30, HAP 0/1/12 ns 1, RAM: 27012/18440; st 33; 4.1: c:1 mp:106.62 ip:5.00 mt_ms:23812 cp:0.00 tp:0.00 md:0 lemd:2 lhmd:2
311825313116 shelly_main.cpp:477     Up 311825.30, HAP 0/1/12 ns 1, RAM: 27012/18440; st 33; 4.1: c:1 mp:106.62 ip:5.00 mt_ms:23812 cp:0.00 tp:0.00 md:0 lemd:2 lhmd:2
311833313050 shelly_main.cpp:477     Up 311833.30, HAP 0/1/12 ns 1, RAM: 27012/18440; st 33; 4.1: c:1 mp:106.62 ip:5.00 mt_ms:23812 cp:0.00 tp:0.00 md:0 lemd:2 lhmd:2
311841313153 shelly_main.cpp:477     Up 311841.30, HAP 0/1/12 ns 1, RAM: 27012/18440; st 33; 4.1: c:1 mp:106.62 ip:5.00 mt_ms:23812 cp:0.00 tp:0.00 md:0 lemd:2 lhmd:2
311849313042 shelly_main.cpp:477     Up 311849.30, HAP 0/1/12 ns 1, RAM: 27012/18440; st 33; 4.1: c:1 mp:106.62 ip:5.00 mt_ms:23812 cp:0.00 tp:0.00 md:0 lemd:2 lhmd:2
311849574219 shelly_hap_window_c:373 WC 1: HAPSetTgtPos 100.00 cur 0.00 tgt 0.00 lhmd 0
311849580259 shelly_hap_window_c:350 WC 1: Tgt pos 0.00 -> 100.00 (HAP)
311849586455 shelly_hap_window_c:330 WC 1: State: idle -> move (0 -> 20)
```

* no more output was received anymore - **the log stream has died**
  it looks like the open-event was received: it started to handle the event
  output was switched on (log entry is missing though) roller did open completely!
  **my guess: shelly firmware crashed and automatically rebootet** (how can I verify a crash?)

* now the state is inconsistent: real shutter is open, but state says 0=closed.

* now closing is impossible:

```
200901207 shelly_main.cpp:477     Up 200.89, HAP 0/1/12 ns 1, RAM: 26996/21280; st 37; 4.1: c:1 mp:106.62 ip:5.00 mt_ms:23812 cp:0.00 tp:0.00 md:0 lemd:0 lhmd:0
208901179 shelly_main.cpp:477     Up 208.89, HAP 0/1/12 ns 1, RAM: 26996/21280; st 37; 4.1: c:1 mp:106.62 ip:5.00 mt_ms:23812 cp:0.00 tp:0.00 md:0 lemd:0 lhmd:0
216901012 shelly_main.cpp:477     Up 216.89, HAP 0/1/12 ns 1, RAM: 26996/21280; st 37; 4.1: c:1 mp:106.62 ip:5.00 mt_ms:23812 cp:0.00 tp:0.00 md:0 lemd:0 lhmd:0
218824721 shelly_hap_window_c:373 WC 1: HAPSetTgtPos 54.00 cur 0.00 tgt 0.00 lhmd 0
218830659 shelly_hap_window_c:350 WC 1: Tgt pos 0.00 -> 54.00 (HAP)
218836818 shelly_hap_window_c:330 WC 1: State: idle -> move (0 -> 20)
218855296 shelly_hap_window_c:330 WC 1: State: move -> rampup (20 -> 22)
218944785 shelly_hap_window_c:554 P = 0.00 -> 106.62
219044599 shelly_hap_window_c:554 P = 0.00 -> 106.62
219145083 shelly_hap_window_c:554 P = 0.00 -> 106.62
219244648 shelly_hap_window_c:554 P = 0.00 -> 106.62
219344705 shelly_hap_window_c:554 P = 0.00 -> 106.62
219444764 shelly_hap_window_c:554 P = 0.00 -> 106.62
219544702 shelly_hap_window_c:554 P = 0.00 -> 106.62
219644718 shelly_hap_window_c:554 P = 0.00 -> 106.62
219744801 shelly_hap_window_c:554 P = 0.00 -> 106.62
219845005 shelly_hap_window_c:554 P = 0.00 -> 106.62
219944877 shelly_hap_window_c:554 P = 0.00 -> 106.62
220044645 shelly_hap_window_c:554 P = 0.00 -> 106.62
220144619 shelly_hap_window_c:554 P = 0.00 -> 106.62
220244729 shelly_hap_window_c:554 P = 0.00 -> 106.62
220345669 shelly_hap_window_c:554 P = 0.00 -> 106.62
220444566 shelly_hap_window_c:554 P = 0.00 -> 106.62
220544776 shelly_hap_window_c:554 P = 0.00 -> 106.62
220644603 shelly_hap_window_c:554 P = 0.00 -> 106.62
220744269 shelly_hap_window_c:554 P = 0.00 -> 106.62
220844551 shelly_hap_window_c:554 P = 0.00 -> 106.62
220944664 shelly_hap_window_c:554 P = 0.00 -> 106.62
221044677 shelly_hap_window_c:554 P = 0.00 -> 106.62
221145021 shelly_hap_window_c:554 P = 0.00 -> 106.62
221244949 shelly_hap_window_c:554 P = 0.00 -> 106.62
221347129 shelly_hap_window_c:554 P = 0.00 -> 106.62
221445053 shelly_hap_window_c:554 P = 0.00 -> 106.62
221544565 shelly_hap_window_c:554 P = 0.00 -> 106.62
221644914 shelly_hap_window_c:554 P = 0.00 -> 106.62
221745466 shelly_hap_window_c:554 P = 0.00 -> 106.62
221845003 shelly_hap_window_c:554 P = 0.00 -> 106.62
221944781 shelly_hap_window_c:554 P = 0.00 -> 106.62
222044731 shelly_hap_window_c:554 P = 0.00 -> 106.62
222145234 shelly_hap_window_c:554 P = 0.00 -> 106.62
222244920 shelly_hap_window_c:554 P = 0.00 -> 106.62
222344685 shelly_hap_window_c:554 P = 0.00 -> 106.62
222444812 shelly_hap_window_c:554 P = 0.00 -> 106.62
222544775 shelly_hap_window_c:554 P = 0.00 -> 106.62
[... 19 lines cut ...]
224227816 shelly_hap_window_c:330 WC 1: State: stopping -> idle (25 -> 0)
224244723 shelly_hap_window_c:330 WC 1: State: idle -> error (0 -> 100)
224345801 shelly_hap_window_c:350 WC 1: Tgt pos 54.00 -> 0.00 (error)
224352357 shelly_hap_window_c:330 WC 1: State: error -> idle (100 -> 0)
224901309 shelly_main.cpp:477     Up 224.89, HAP 0/1/12 ns 1, RAM: 26948/21280; st 37; 4.1: c:1 mp:106.62 ip:5.00 mt_ms:23812 cp:0.00 tp:0.00 md:0 lemd:0 lhmd:0
232900699 shelly_main.cpp:477     Up 232.89, HAP 0/1/12 ns 1, RAM: 26996/21280; st 37; 4.1: c:1 mp:106.62 ip:5.00 mt_ms:23812 cp:0.00 tp:0.00 md:0 lemd:0 lhmd:0
```

* it looks like the inconsistent state is known to the shelly
the message „Failed to start moving“ indicates that something is wrong.
it says State: idle -> error (0 -> 100)
it says Stats: error -> idle (100 -> 0)

* a workaorund could be implemented
 when start moving fails the internal state should be stopped
 issuing "stop" in state "stopped" does not make sense (assuming state=stopped after reboot)
 it could assume something is wrong
[... cut ...] ion is invalid it could reset cp=5 (or 50 to support both directions)
 this should enable closing again

* detect inconsistent state
when output is switched on and no power is consumed it can assume the shutter-end-switch has kicked in to avoid damaging the shutter-motor
this hypothesis can be testet: switch output on opposite direction (tp=tp +/- 5) and check power consumption again

* another solution for power users would be to have a URL to set the cp manually: e.g. http://shell25/set?cp=100

gmuth, 2024-10-07:
Issues look related:
https://github.com/mongoose-os-apps/shelly-homekit/issues/844
https://github.com/mongoose-os-apps/shelly-homekit/issues/920

nextgenoperations, 2024-10-07:
I am also seeing the same issue on a Shelly  shellyswitch25.. for me it started when I upgrade to 2.11.2, was find on 2.10.1

If I simply click the square icon on the home app for my shutter to open or close it or use Siri it all works fine. but I enter the 'slider' page and try and use the slider to open / close the shutter partially, it all goes wrong and shutter never full opens or closes... if I try to full open, it almost opens, but then starts to close again... odd.

markirb, 2024-10-10:
What I really take from this is that devices should not reboot when getting triggered. Do you have WiFi Power savings on maximum (IIRC it is 1 for plus devices but 2 for non-plus)?

gmuth, 2024-10-12:
Sorry, I did not note the WiFi Power savings setting - I did not change it - so I guess 2.

As for the ticket I suggest two follow up tickets and a label for this common issue like "hardware issue".
The problem looks like the reason is partly a hardware and partly a software issue.
Because the original firmware doesn't seem to have that problem it looks like the shelly-homekit firmware configures or uses the hardware in a different was. There was a change in the past concerning power management to reduce power consumption and cpu-temperature. It think this configuration has a higher probability to trigger a hardware-failure/CPU-crash.

@markirb, instead of classifying this as bug, I would like to close this one and open two enhancement tickets to look for workaorunds to solve this issues: one to set CP manually, another one to detect "i-crashed-while-openning" and "have-to-adjust-cp".

nextgenoperations, 2024-10-15:
Just to add I don't see my Shelly 2.5 rebooting when I try to open the shutter with the slider, but happy to re-test if required.

What I see is on 2.10.1 everything was working fine and then on 2.11.2 it broke.

gmuth, 2024-10-29:
I've been monitoring the roller shutter device and its behavior using version 2.10.1 for two weeks now and did not have a single issue. So I can also confirm that the behavior identified in combination with the individual setup (shutter motor) and the 2.12 versions has higher probability to crash/reboot during switching - leaving the shelly with a messed up position state.

_Software Workaround_
Downgrade to **Firmware 2.10.1**

_Hardware Workaround_
Install **RC Snubbers** to "effectively mitigate voltage spikes that can occur during switching of inductive loads"

markirb, 2024-10-29:
Unfortunately +2PM is not available in 2.10.1

However I see that all GEN1 devices/ESP8266 seem to have a of memory leak apparent in RPC calls  in versions >2.10.1, which might explain crashes.

Still does not fully explain the +2PM behavior that seems to be the same? or is it another way?