mongoose-os-apps/shelly-homekit #1424: 2.12.1. Shelly plus 2pm calibration doesn’t work

DanielZornhagen, 2024-07-23:
The Shelly Plus 2PM has been updated to the HomeKit version and defined as a Roller Shutter. During calibration, it only runs for a maximum of 5.09 seconds before ending the calibration. The maximum distance the shutter can move is about 30% of the total travel distance. New calibrations slightly change the maximum runtime. The same error occurs even after resetting and updating again. Turning off the power for several minutes does not affect the calibration time either. The input mode was set to separate - momentary, as it involves two push buttons. Version 2.12.1.

markirb, 2024-07-23:
Can you post logs please?

DanielZornhagen, 2024-07-25:
Where can I get the logs from?

santigvz, 2024-08-10:
Exactly the same error here and also ir afecta to the wall switch which now can not move more than the 30%…

Also there is a reported error that does not allow to come back to the custom FW. Help appreciated!

DanielZornhagen, 2024-09-05:
the problem still exists. Here is the log file:

80260214 shelly_debug.cpp:231    No log file, sending new entries
80267279 mgos_http_server.c:180  0x3ffce4d4 HTTP connection from 192.168.189.14:51103
82014535 shelly_main.cpp:483     Up 82.00, HAP 0/1/16 ns 1, RAM: 209300/196404; st 56; 4.1: c:1 mp:0.00 ip:5.00 mt_ms:5090 cp:49.27 tp:49.27 md:0 lmd:2
90014553 shelly_main.cpp:483     Up 90.00, HAP 0/1/16 ns 1, RAM: 209300/196404; st 55; 4.1: c:1 mp:0.00 ip:5.00 mt_ms:5090 cp:49.27 tp:49.27 md:0 lmd:2
96527903 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
96834580 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
98014520 shelly_main.cpp:483     Up 98.00, HAP 0/1/16 ns 1, RAM: 209300/196404; st 55; 4.1: c:1 mp:0.00 ip:5.00 mt_ms:5090 cp:49.27 tp:49.27 md:0 lmd:2
101751706 W (102397) wifi:<ba-add>idx:1 (ifx:0, 52:e6:36:4f:8e:51), tid:6, ssn:2, winSize:64
104616618 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
105742524 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
106014800 shelly_main.cpp:483     Up 106.00, HAP 0/1/16 ns 1, RAM: 209048/196404; st 55; 4.1: c:1 mp:0.00 ip:5.00 mt_ms:5090 cp:49.27 tp:49.27 md:0 lmd:2
106473904 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
107490622 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
108514786 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
109536180 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
110560754 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
111683082 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
112516311 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
113535389 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
114014606 shelly_main.cpp:483     Up 114.00, HAP 0/1/16 ns 1, RAM: 209048/196404; st 56; 4.1: c:1 mp:0.00 ip:5.00 mt_ms:5090 cp:49.27 tp:49.27 md:0 lmd:2
114558354 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
115489630 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
116494826 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
117518404 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
118557664 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
119772962 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
120488634 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
121718412 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.189.14:51101
121726757 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
121796384 shelly_hap_window_c:330 WC 1: State: idle -> precal0 (0 -> 10)
121895992 shelly_hap_window_c:450 Begin calibration
122038667 mgos_sys_config.c:323   Saved to conf9.json
122044209 shelly_output.cpp:64    Output 1: off -> on (precal0)
122050106 shelly_hap_window_c:330 WC 1: State: precal0 -> cal0 (10 -> 11)
122069935 shelly_main.cpp:483     Up 122.06, HAP 0/1/16 ns 1, RAM: 207500/196404; st 56; 4.1: c:0 mp:0.00 ip:5.00 mt_ms:5090 cp:49.27 tp:49.27 md:0 lmd:2
122395456 shelly_hap_window_c:466 WC 1: P0 = 0.000
123195393 shelly_hap_window_c:466 WC 1: P0 = 0.000
123408920 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
123995472 shelly_hap_window_c:466 WC 1: P0 = 0.000
124758312 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
124795554 shelly_hap_window_c:466 WC 1: P0 = 0.000
125507385 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
125595587 shelly_hap_window_c:466 WC 1: P0 = 0.000
126395512 shelly_hap_window_c:466 WC 1: P0 = 0.000
126448952 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
127096320 shelly_output.cpp:64    Output 1: on -> off (cal0)
127102330 shelly_hap_window_c:330 WC 1: State: cal0 -> postcal0 (11 -> 12)
127196948 shelly_hap_window_c:330 WC 1: State: postcal0 -> precal1 (12 -> 13)
127295660 shelly_output.cpp:64    Output 2: off -> on (precal1)
127301567 shelly_hap_window_c:330 WC 1: State: precal1 -> cal1 (13 -> 14)
127566613 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
127895503 shelly_hap_window_c:497 WC 1: P1 = 0.000
128476554 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
128695396 shelly_hap_window_c:497 WC 1: P1 = 0.000
129495447 shelly_hap_window_c:497 WC 1: P1 = 0.000
129508063 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
130014481 shelly_main.cpp:483     Up 130.00, HAP 0/1/16 ns 1, RAM: 209436/196404; st 56; 4.1: c:0 mp:0.00 ip:5.00 mt_ms:5090 cp:49.27 tp:49.27 md:0 lmd:2
130295424 shelly_hap_window_c:497 WC 1: P1 = 0.000
130524755 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
131095454 shelly_hap_window_c:497 WC 1: P1 = 0.000
131548498 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
131896754 shelly_hap_window_c:497 WC 1: P1 = 0.000
132396145 shelly_output.cpp:64    Output 2: on -> off (cal1)
132403662 shelly_hap_window_c:503 WC 1: calibration done, move_time 5089, move_power 0.000
132409572 shelly_hap_window_c:330 WC 1: State: cal1 -> postcal1 (14 -> 15)
132451212 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
132628663 mgos_sys_config.c:323   Saved to conf9.json
[... cut ...]
136873635 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
137592795 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
138014434 shelly_main.cpp:483     Up 138.00, HAP 0/1/16 ns 1, RAM: 209436/196404; st 56; 4.1: c:1 mp:0.00 ip:5.00 mt_ms:5089 cp:50.85 tp:50.85 md:0 lmd:1
138826204 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.189.14:51101
146014719 shelly_main.cpp:483     Up 146.00, HAP 0/1/16 ns 1, RAM: 209436/196404; st 56; 4.1: c:1 mp:0.00 ip:5.00 mt_ms:5089 cp:50.85 tp:50.85 md:0 lmd:1

markirb, 2024-09-13:
The logs show that the measured power is always 0... :(

DanielZornhagen, 2024-09-16:
Is there a solution, how the problem can be solved?

markirb, 2024-09-16:
I am under the impression that you have the first Hardware Version of the device. Maybe Powermetering does not work there yet?
Can you show me the output of 
http://A.B.C.D/rpc/Config.Get

zuklior, 2024-09-21:
Hi
Same problem here.. is there a walk around??

markirb, 2024-09-21:
Can you give me the output I was asking for? You might need to reboot the device also?

DanielZornhagen, 2024-09-21:
{"debug":{"udp_log_addr":"","udp_log_level":3,"mbedtls_level":1,"level":2,"file_level":"","event_level":3,"stdout_uart":0,"stderr_uart":0,"factory_reset_gpio":-1,"mg_mgr_hexdump_file":""},"sys":{"esp32_adc_vref":0,"esp32_adc_width":3,"tz_spec":"","wdt_timeout":30,"pref_ota_lib":""},"i2c":{"unit_no":0,"enable":true,"freq":100000,"debug":false,"sda_gpio":33,"scl_gpio":25},"i2c1":{"unit_no":1,"enable":false,"freq":100000,"debug":false,"sda_gpio":22,"scl_gpio":23},"device":{"id":"ShellyPlus2PM-98B0DC","license":"","mac":"","public_key":"","sn":""},"conf_acl":"*","file_logger":{"enable":false,"dir":"/","prefix":"log_","max_file_size":16000,"max_num_files":1,"timestamps":true,"level":3,"include":"","syslog_enable":true,"rpc_service_enable":false,"buf_size":1000,"buf_line":true},"dns_sd":{"enable":true,"adv_only":false,"host_name":"shellyplus2pm-48551998b0dc","txt":"fw_type=homekit"},"hap":{"salt":"","verifier":"","setup_id":"","mfi_uuid":"","mfi_token":""},"http":{"enable":true,"listen_addr":"80","document_root":"/","index_files":"","ssl_cert":"","ssl_key":"","ssl_ca_cert":"","upload_acl":"*","hidden_files":"","auth_domain":"","auth_file":"","auth_algo":1,"extra_headers":""},"update":{"timeout":600,"commit_timeout":0,"key0":"","key1":"","key2":"","key3":"","key4":"","key5":"","key6":"","key7":"","sig_req_mask":-1,"url":"","interval":0,"extra_http_headers":"","ssl_ca_file":"ca.pem","ssl_client_cert_file":"","ssl_server_name":"","enable_post":true},"rpc":{"enable":true,"http_enable":true,"service_sys_enable":true,"max_frame_size":4096,"max_queue_length":25,"max_non_persistent_channels":20,"default_out_channel_idle_close_timeout":10,"acl":"","acl_file":"","auth_domain":"","auth_file":"","auth_algo":1,"uart":{"uart_no":0,"baud_rate":115200,"fc_type":2,"dst":""},"ws":{"enable":true,"server_address":"","reconnect_interval_min":1,"reconnect_interval_max":60,"ssl_server_name":"","ssl_cert":"","ssl_key":"","ssl_ca_cert":""}},"sntp":{"enable":true,"server":"time.google.com","use_dhcp_ntp_server":true,"retry_min":1,"retry_max":30,"update_interval":7200},"wifi":{"ap":{"enable":false,"ssid":"ShellyPlus2PM-48551998B0DC","pass":"","hidden":false,"channel":6,"max_connections":10,"ip":"192.168.33.1","netmask":"255.255.255.0","gw":"","dhcp_start":"192.168.33.2","dhcp_end":"192.168.33.100","trigger_on_gpio":-1,"disable_after":0,"hostname":"","bandwidth_20mhz":false,"protocol":"BGN","keep_enabled":false},"sta":{"enable":true,"ssid":"Darex-NT","pass":"4216 8512 9272 WX22 5210","bssid":"","channel":0,"last_bssid":"52:e6:36:4f:8e:51","last_channel":1,"user":"","anon_identity":"","cert":"","key":"","ca_cert":"","ip":"","netmask":"","gw":"","nameserver":"","dhcp_hostname":"shellyplus2pm-48551998b0dc","protocol":"BGN","listen_interval_ms":0},"sta1":{"enable":false,"ssid":"","pass":"","bssid":"","channel":0,"last_bssid":"","last_channel":0,"user":"","anon_identity":"","cert":"","key":"","ca_cert":"","ip":"","netmask":"","gw":"","nameserver":"","dhcp_hostname":"shellyplus2pm-48551998b0dc","protocol":"BGN","listen_interval_ms":0},"sta2":{"enable":false,"ssid":"","pass":"","bssid":"","channel":0,"last_bssid":"","last_channel":0,"user":"","anon_identity":"","cert":"","key":"","ca_cert":"","ip":"","netmask":"","gw":"","nameserver":"","dhcp_hostname":"shellyplus2pm-48551998b0dc","protocol":"BGN","listen_interval_ms":0},"sta_rssi_thr":-95,"sta_connect_timeout":15,"sta_roam_rssi_thr":-80,"sta_roam_interval":600,"sta_ps_mode":1},"shelly":{"name":"shellyplus2pm-48551998b0dc","mode":1,"cfg_version":6,"legacy_hap_layout":false,"overheat_on":100,"overheat_off":90,"reboot_counter":0,"wifi_connect_reboot_timeout":0},"_const":{"rpc_acl":"[{\"ch_type\": \"UART\", \"acl\": \"*\"},{\"method\": \"Shelly.GetInfo\", \"acl\": \"*\"},{\"method\": \"*\", \"ch_type\": \"HTTP\", \"acl\": \"admin\"},{\"method\": \"*\", \"ch_type\": \"WS_in\", \"acl\": \"admin\"}]"},"k_apparent":1750000,"max_power":1800,"sw1":{"name":"Shelly SW1","enable":true,"out_inverted":false,"in_mode":1,"in_inverted":false,"state":false,"svc_type":0,"hk_state_inverted":false,"valve_type":1,"initial_state":3,"auto_off":false,"auto_off_delay":0.000000,"state_led_en":-1,"persist_state":false},"in1":{"type":3,"inverted":false,"ssw":{"name":"Input 1","in_mode":0},"sensor":{"name":"Shelly S1","in_mode":0,"idle_time":15}},"sw2":{"name":"Shelly SW2","enable":true,"out_inverted":false,"in_mode":1,"in_inverted":false,"state":false,"svc_type":0,"hk_state_inverted":false,"valve_type":1,"initial_state":3,"auto_off":false,"auto_off_delay":0.000000,"state_led_en":-1,"persist_state":false},"in2":{"type":3,"inverted":false,"ssw":{"name":"Input 2","in_mode":0},"sensor":{"name":"Shelly S2","in_mode":0,"idle_time":15}},"in3":{"type":3,"inverted":false,"ssw":{"name":"Shelly SSW3","in_mode":0},"sensor":{"name":"Shelly S3","in_mode":0,"idle_time":15}},"wc1":{"name":"Window 1","in_mode":0,"swap_inputs":false,"swap_outputs":false,"calibrated":true,"idle_power_thr":5.000000,"move_power":0.000000,"move_time_ms":5089,"max_ramp_up_time_ms":5000,"obstruction_power_coeff":2.500000,"obstruction_duration_ms":3000,"obstruction_time_coeff":1.500000,"current_pos":50.854786},"gdo1":{"name":"Garage Door","close_sensor_mode":0,"open_sensor_mode":0,"out_mode":0,"move_time_ms":30000,"begin_move_time_ms":7000,"pulse_time_ms":300},"ts1":{"name":"Shelly TS1","unit":0,"update_interval":1},"ts2":{"name":"Shelly TS2","unit":0,"update_interval":1},"ts3":{"name":"","unit":0,"update_interval":1},"ssw1":{"name":"Input 1","in_mode":0},"ssw2":{"name":"Input 2","in_mode":0},"factory":{"batch":"2211-Broadwell","model":"SNSW-002P16EU","version":1,"calib":{"done":true,"gains0":{"avgain":1600971.375000,"aigain":491898.968750,"awgain":3527589.750000,"phcala":0.000000,"bvgain":1598085.125000,"bigain":498927.562500,"bwgain":3536512.750000,"phcalb":0.000000}}}}

markirb, 2024-09-22:
Thank you I reduced your info as it maybe contained too private information. But I have what I needed:

It looks indeed like you have Version 1 of the hardware SNSW-002P16EU. 

There is no proper reset for the Power IC (or it's unknown?) in this version. Also that's the only known difference between version 1 and 2. 

You could try if a reboot/power cut helps for now.
Otherwise I have to look on another way to reset...

DanielZornhagen, 2024-09-23:
Sadly a power cut doesn't help. Even a reset  to facorysettings doesn't help. If you find a solution, ist would be great. 

Within the Shelly-App the Calibration works fine.

markirb, 2024-09-24:
Hi,

ok that would have been an easy fix. :( 

Sadly it is hard to Pinpoint this issue. The Chip itself is reset via software. And I think it is found otherwise your device should not be able to perform any firmware Update.

So far I have no idea why this should not work. When you switch to "normal mode" i.e. non window cover you see the power showing, right? Maybe channels are switched somehow? Can you try to play around in this "normal mode"?

[image]

markirb, 2024-09-28:
One more thing: did you have the device on anything other than stock or this fw? It might also be that your calibration partition is wrong.

Can you downgrade to stock? And if so does it show power measuerements?

DanielZornhagen, 2024-10-20:
I’ve only used the Shelly with the stock firmware and this custom firmware. In the Shelly app, the power measurement works perfectly and is displayed correctly on the dashboard. The only place it doesn’t work is with this custom firmware.

markirb, 2024-10-21:
I see. 
Then it seems indeed that the previous hardware version of 2pm (v1) does not yet work with the code. I do not know how to fix this without the hardware unfortunately.

DanielZornhagen, 2024-10-22:
So the only solution I can choose is to buy the Shelly +2PM in a new version and replace all 15 Shellys?

markirb, 2024-10-22:
That is not the solution 🙃

I can say that I cannot fix this as long as I do not know the exact problem. 

Can you please remove the plastic casing to confirm that the assumption is valid that we have a v1 board.

I can only look at the problem when I have the same hardware. Which cannot be bought anymore.

markirb, 2024-10-24:
Maybe you can try this: This avoids IRQ pin conflict.
Note that 2.13 can only be flashed from 1.4.X stock

[Plus2PM-esp32-2.13.0.zip](https://github.com/user-attachments/files/17510940/Plus2PM-esp32-2.13.0.zip)

DanielZornhagen, 2024-10-30:
I tried the firmware. Unfortunately it doesn't solve the Problem.

markirb, 2024-11-07:
Ok thanks for testing. I guess this leaves the only option to send me the hardware if you want it running

finthamoussu, 2025-01-10:
> Good catch. Easy fix. Try this:
> 
> (on Mac right click save as)
> 
> [shelly-homekit-ShellyPlus2PM.zip](https://github.com/user-attachments/files/18377166/shelly-homekit-ShellyPlus2PM.zip)

👏 This fixes the issue. Power messurement is now in the correct switch which fixes calibration.

Thanks for the quick bug fix.

markirb, 2025-01-10:
Okay. Closing this, is available in prerelease, and will be in next release