mongoose-os-apps/shelly-homekit #1427: Plus Plug S OTA update fails

JakobWe, 2024-08-01:
I have a newly bought Shelly Plus Plug S. The currently installed firmware is ```20240726-114505/1.4.0-gb2aeadb```.
Installing the homekit software does not work - I have the feeling I am missing something obvious? Do you need more information to properly assess this issue?

Here are the logs:
[Using file upload method](https://github.com/user-attachments/files/16454636/diagnostics-shelly-plusplugs-debug-log-file-upload.txt)
[Using http://A.B.C.D/ota?url=http://shelly.rojer.cloud/update method](https://github.com/user-attachments/files/16454637/diagnostics-shelly-plusplugs-debug-log-ota-url.txt)

The Python script determines that the plug is not a shelly device:
[image]
But the plug is reachable via the ip in a browser on the same machine:
[image]

markirb, 2024-08-01:
Update from 1.4 is not yet supported, they repartition the flash. please go back to 1.3.x

closed 2024-08-01 by markirb

hadess, 2025-01-25:
> Update from 1.4 is not yet supported, they repartition the flash. please go back to 1.3.x

It would be great if the flasher tool said "Host: X.X.X.X uses an unsupported firmware" rather than "is not a Shelly device."

hadess, 2025-01-25:
> It would be great if the flasher tool said "Host: X.X.X.X uses an unsupported firmware" rather than "is not a Shelly device."

Oh, updating using the OTA works, just not from the flasher script. I filed this issue about it: https://github.com/mongoose-os-apps/shelly-homekit/issues/1618