mongoose-os-apps/shelly-homekit #1428: Unable to install HomeKit after upgrading to 1.4.0 firmware

johnmrichey, 2024-08-02:
I have tried all the methods for installing HomeKit and they all fail.  I am also unable to go back to an older firmware which I posted about on the Shelly Forum.

Any suggestion please to get HomeKit installed on Shelly 1 Plus running firmware 1.4.0?

https://community.shelly.cloud/topic/1852-unable-to-downgrade-to-other-firmware-after-updating-to-version-140/#comment-8946

johnmrichey, 2024-08-02:
I am unable to get the Shelly back to an older version firmware after updating to 1.4.0.  Any suggestion how to do this please as I have tried everything mentioned out there.

markirb, 2024-08-03:
I can only tell from experience some weeks ago on 1.4.0b on Plus2PM:

Stock:
1.4.0b downgrade to 1.3.3 was possible
1.4.0b to Shelly-HomeKit fails because of rootfs size but should not break anything.

Shelly-HomeKit:
Revert to 1.4.0b fails correctly and expectedly due to app size but does not break anything.

I do not know yet how Shelly-HomeKit should support the partition size change:

The best thing would be to implement the migration like Shelly does. But this is tedious work, as it has to work with the failsafe OTA mechanism. 

Also it is tedious for everybody to revert to 1.3.3 then upgrade to 1.4.0 then flash Shelly-HomeKit with new partition sizes variant.

It could also be done with a split for two different partition layouts but that is not nice for users.

johnmrichey, 2024-08-03:
I was not able to ever migrate to HomeKit, so I don't believe failing to downgrade to 1.3.3 has anything to do with trying to migrate to HomeKit.

Regardless I am still not able to downgrade after updating to 1.4.0 on the 1P Shelly.

markirb, 2024-08-03:
You need to downgrade to 1.3.3 do you have that version? What is the error message?

johnmrichey, 2024-08-03:
Do you have the zip or the URL please?

mtran83, 2024-08-03:
I'm also facing the same issue after updating to FW v1.4.

johnmrichey, 2024-08-04:
The Shelly forum says it's not possible to downgrade from 1.4.0. Nope no way was the last comment on my post.

rmmmrm, 2024-08-09:
@johnmrichey contact shelly suport and they send the zip. It worked for me for all my shelly's ony the shelly plus 1 is not working because i cannot flash the shelly-homekit

rmmmrm, 2024-08-09:
@johnmrichey @mtran83 
Down here you have the 1.3.3 firmware for these shelly models
Regards

[PlusI4-1.3.3.zip](https://github.com/user-attachments/files/16561121/PlusI4-1.3.3.zip)
[Plus1-1.3.3.zip](https://github.com/user-attachments/files/16561122/Plus1-1.3.3.zip)
[PlusHT-1.3.3.zip](https://github.com/user-attachments/files/16561123/PlusHT-1.3.3.zip)
[Plus2PM-1.3.3.zip](https://github.com/user-attachments/files/16561124/Plus2PM-1.3.3.zip)

mtran83, 2024-08-09:
thanks @rmmmrm 

How are you updating these firmware using the above shared zip files? Via the Shelly Web UI?

rmmmrm, 2024-08-10:
Hello @mtran83
Yes via shelly 
[image]

ivofernandespt, 2024-08-18:
> @johnmrichey @mtran83 Down here you have the 1.3.3 firmware for these shelly models Regards
>
> [PlusI4-1.3.3.zip](https://github.com/user-attachments/files/16561121/PlusI4-1.3.3.zip) [...]

Anyone have this version for Plus 1PM?

markirb, 2024-08-21:
http://rojer.me/files/shelly/stock/1.3.3/

closed 2024-08-21 by markirb