mongoose-os-apps/shelly-homekit #1429: Momentary Switch with Shelly Plus 1

johnmrichey, 2024-08-04:
I have the Shelly HomeKit setup as Momentary switch but when I get the contacts across the dry contacts of 0|I it's not acting as a momentary switch.

I have tried all the various combinations but it seems to be always the same regardless.

Any suggestions please.
[image]

johnmrichey, 2024-08-04:
I posted this to the Shelly forum as it seems this is an issue with stock 1.4.0 as well in my testing.

Shelly Plus 1 as Momentary Switch https://community.shelly.cloud/topic/1870-shelly-plus-1-as-momentary-switch/

Sternsoldat, 2024-08-04:
HeyCan you explain what exactly happen with you O|I contact?-send from my mobile device-

31760082-ED0C-4298-8F53-5814634BBB71.png (view on web)

johnmrichey, 2024-08-04:
As a momentary switch which is normally open it should be open when you release the button.  When I use Garage Door type that behaves correctly and return to the open state but it forces the HAP type to Garage Door in HomeKit.

When I use Switch type Momentary the dry contacts toggle to closed and stay in the closed position, not until you press the button again do it then open.

johnmrichey, 2024-08-04:
The voltmeter across the contacts when pressed indicates continuity and stays that way until you press the button again.

johnmrichey, 2024-08-04:
You just need to turn on Auto Off and it works fine.

Problem solved.

johnmrichey, 2024-08-16:
Auto Off does not make it work properly as an actual momentary switch.  Any ideas how to make this work as normally open actual momentary switch please?

markirb, 2024-09-14:
Can you please add more detail why auto off does not work? It should be exactly what you need

dmarthur, 2024-09-15:
@johnmrichey - I was having this same problem even after I activated the auto-off feature. I disconnected the live wire to the relay, waited 10 seconds, and reconnected it. That fixed the problem for me.

closed 2024-10-07 by markirb