mongoose-os-apps/shelly-homekit #1430: Shelly 1 goes regularly off-line

andrea4381, 2024-08-04:
I’ve configured a Shelly 1 with homekit firmware (last versione available). Everything works fine for 3-4 days and then it goes offline and I have to reconfigure it again. With standard Shelly firmware this doesn’t happen. Is anyone facing the same problem?

markirb, 2024-08-18:
Goes offline means it is in AP Mode?
What is the uptime when you see it in AP Mode?

andrea4381, 2024-08-18:
Reset as “factory reset”, with the factory WiFi active and it’s possible to access just with 192.168.33.1.

markirb, 2024-08-18:
Ok, then I guess the reset sequence toggles. 

Does any of those apply:

https://github.com/mongoose-os-apps/shelly-homekit/wiki/Recovery

Might be that your input is somehow shaky... ?

andrea4381, 2024-08-19:
I'm using it on a shelly 1 first version, so no reset button. This means at least power down-up is require to activate the AP mode and this shouldn't be the case. But I will investigate more. Thanks for your support!

jan666, 2024-09-02:
I had the same issue with a Shelly Plus 1. I used a cheap 12 Volt power supply. Tested with stock shelly firmware - same issue. So I contacted the support and in the end it was the cheap 12 Volt power supply. I switched to https://www.amazon.de/dp/B0B31FLPTV and now it works. Maybe you're using a bad 12 Volt power supply, too?

andrea4381, 2024-09-04:
No I used the 24V that the gate controller provides. I used it an old Shelly 1, first generation. I switched now to a Shelly 1 Plus and up to now it works much better it seems.ù
If it will happen again I'll try to use an external power supply, thanks!

closed 2024-09-13 by markirb