mongoose-os-apps/shelly-homekit #1431: Shelly Plus 2PM with Shelly Homekit 2.12.1 fails to revert to stock

sergherrero, 2024-08-04:
Shelly Plus 2PM with build 20240623-075044/2.12.1-g417cbdf

Using the web gui after clicking on 'revert' nothing occurs. The logs show that the firmware update failed.

```
81548633111 shelly_ota.cpp:90       Starting firmware update
81548646145 esp32xx_ota_backend:209 App: Plus2PM.bin -> app_1 enc 0, FS: fs.img -> fs_1 enc 0
81548658428 mgos_ota_core.c:685     0.12% total, bootloader.bin 512 of 25280
81548791935 mgos_ota_core.c:685     1.34% total, partition-table.bin 512 of 4096
81548827649 mgos_ota_core.c:685     1.54% total, otadata.bin 512 of 8192
81548882079 mgos_ota_core.c:1051    Update finished, result -1 (File is too big)
```

markirb, 2024-08-04:
Yes. Unfortunately revert to stock is currently broken as it tries to revert to 1.4.0 which has other partition sizes. 

Revert to 1.3.3 is possible if you know how to get the file. 

This can and will be fixed, but I currently do not have access to the resources to fix the problem. in two weeks I will.

lucatk, 2024-08-05:
does someone have a firmware link for 1.3.3 for the Shelly Plus 2PM?

ReToCode, 2024-08-06:
@rojer any idea how to get the 1.3.3 version of https://updates.shelly.cloud/update/Plus2PM? This URL seems to give only the latest one. Same for the https://rojer.me/files/shelly/stock/ update.sh script.

ReToCode, 2024-08-07:
Found it, you can download it from 

```
{"stable": {"version": "1.3.3", "build_id": "20240625-122830/1.3.3-gbdfd9b3", "url": "https://fwcdn.shelly.cloud/gen2/Plus2PM/eea874bcfee2b4876901948159b80bd9d2fc719300982f3ee489fa2168d400ea"}, "time": 1723008265}
```

I was able to revert to the original 1.3.3 firmware with that.

cc @lucatk

venity, 2024-08-07:
> Found it, you can download it from
> 
> ```json
> {
>   "stable": {
>     "version": "1.3.3",
>     "build_id": "20240625-122830/1.3.3-gbdfd9b3",
>     "url": "https://fwcdn.shelly.cloud/gen2/Plus2PM/eea874bcfee2b4876901948159b80bd9d2fc719300982f3ee489fa2168d400ea"
>   },
>   "time": 1723008265
> }
> ```
> 
> I was able to revert to the original 1.3.3 firmware with that.
> 
> cc @lucatk

You are a lifesaver. Thank you!

Anyone has the 1.3.3 version for Plus 1?

ReToCode, 2024-08-07:
Should be the following, but can't test that, as I don't have a 1 Plus.

```
{
  "stable": {
    "version": "1.3.3",
    "build_id": "20240625-122719/1.3.3-gbdfd9b3",
    "url": "https://fwcdn.shelly.cloud/gen2/Plus1PM/cc34adf8e45a3765b3f05efcd9a4322efd99c50c52ec9434fa51beb3b56217e1"
  },
  "time": 1723013136
}
```

venity, 2024-08-07:
> Should be the following, but can't test that, as I don't have a 1 Plus.
>
> ```
> {
>   "stable": {
>     "version": "1.3.3",
>     "build_id": "20240625-122719/1.3.3-gbdfd9b3",
>     "url": [...]

I need for Plus1 instead of Plus1PM but anyway thanks for the quick reply.

ReToCode, 2024-08-07:
Probably:
```
{
  "stable": {
    "version": "1.3.3",
    "build_id": "20240625-122558/1.3.3-gbdfd9b3",
    "url": "https://fwcdn.shelly.cloud/gen2/Plus1/ddd5a7b49ff3e65240d1264eb531f82da2aa86d3d05d045c5226a81e7ea2e43d"
  },
  "time": 1723015391
}
```

venity, 2024-08-07:
> Probably:
> 
> ```
> {
>   "stable": {
>     "version": "1.3.3",
>     "build_id": "20240625-122558/1.3.3-gbdfd9b3",
>     "url": "https://fwcdn.shelly.cloud/gen2/Plus1/ddd5a7b49ff3e65240d1264eb531f82da2aa86d3d05d045c5226a81e7ea2e43d"
>   },
>   "time": 1723015391
> }
> ```

Thanks again, it works perfectly!

markirb, 2024-08-07:
Anyone knows how to get generally get the url? Probably headers?

sergherrero, 2024-08-07:
@ReToCode Thanks!

ReToCode, 2024-08-08:
> Anyone knows how to get generally get the url? Probably headers?

Yepp, but you need pretty exact headers from an existing device, so I used an other (still working) device which was on a previous version to get the headers. Basically something this:

```
curl -k -H "X-MGOS-Device-ID: <your-value>" -H "User-Agent: Plus2PM/0.14.1 (MongooseOS/202210021802; esp32)" -H  "X-MGOS-FW-Version: esp32 0.14.1 20230308-091102/0.14.1-g22a4cb7" https://updates.shelly.cloud/update/Plus2PM
```

Sillyjan, 2024-08-09:
Is there a way tot get 1.3.3 for The shelly Plug S (https://updates.shelly.cloud/update/PlusPlugS) as well?

rmmmrm, 2024-08-09:
Hello,
Check my post
Regards

https://github.com/mongoose-os-apps/shelly-homekit/issues/1428#issuecomment-2277831464

markirb, 2024-08-18:
> Yepp, but you need pretty exact headers from an existing device, so I used an other (still working) device which was on a previous version to get the headers. Basically something this:
>
> ```
> curl -k [...]

That does not really work for me. Can you post the response for you? Did you find out what the important bits are?

markirb, 2024-08-19:
Never mind i think I found it. Revert will work soon again....

markirb, 2024-08-21:
Please if you are looking for 1.3.3 files they are here:

http://rojer.me/files/shelly/stock/1.3.3/

closed 2024-08-21 by markirb