mongoose-os-apps/shelly-homekit #1432: Flashing +PlugS?

guzzzi, 2024-08-05:
Hello,
I'm not sure what I'm doing wrong, or if I'm in the right place to ask a question.
I would like to flash some Shelly Plus Plug S. I did it as in the instructions, unfortunately nothing happens when I enter the URL with the Shelly IP.
I have now tried with the stock version 1.3.3 and 1.4.0, unfortunately unsuccessfully.
Does anyone have any tips for me or can tell me where I can ask for this, should I be wrong here?

Sorry if I am wrong here.

closed 2024-08-06 by guzzzi

andi-abc, 2024-08-11:
Hi i am having the same problem. How did you solve the problem?

guzzzi, 2024-08-11:
V1.4.0 isn't possible to flash. I tried to get a Downgrade from Shelly, but they won't send it out, I think we have to wait for an Update of the custom Firmware.
For Plus Plug S in V1.3.3 I set up the local "Tiny Web server" and download the correct custom Firmware from here and put the File into the "www" Folder and flashed it from my local Machine with the OTA method described here.
For me, it didn't work with the Online method, but local worked with V1.3.3 Plus Plug S Hardware.