mongoose-os-apps/shelly-homekit #1433: Plus 1 Upgrade seems to finish, but didnt upgrade

Littlericket, 2024-08-09:
Hi all,

i have 2 out of 10 shelly 1 plus at my home which are not correctly upgrading. I dont know if they both were in the same double-package and have some common differences with the other ones in this home.

When applying the upgrade, the process runs successfully, but after rebooting the shelly is still running the previous version, even if the logs tell me that everything went fine.

Tried downgrading, upgrading (from 2.11.2 to 2.12.1) , reverting to stock. The shelly always comes back with the previous version 2.11.2.

Below the log entries when starting the upgrade.

```
1587473164 mgos_ota_core.c:656     FW: Plus1 esp32 2.12.1 20240623-075054/2.12.1-g417cbdf sig 0x00
1592449201 shelly_ota.cpp:90       Starting firmware update
1592463683 esp32xx_ota_backend:209 App: Plus1.bin -> app_0 enc 0, FS: fs.img -> fs_0 enc 0
1592470759 mgos_mongoose.c:66      New heap free LWM: 191904
1592477697 mgos_mongoose.c:66      New heap free LWM: 190360
1592485015 mgos_ota_core.c:685     0.15% total, bootloader.bin 512 of 23728
1592492514 mgos_ota_core.c:1202    Update state: 0
1592749625 mgos_ota_core.c:685     1.59% total, partition-table.bin 512 of 3072
1592757215 mgos_ota_core.c:1202    Update state: 0
1592787058 mgos_ota_core.c:685     1.78% total, otadata.bin 512 of 8192
1592794606 mgos_ota_core.c:1202    Update state: 0
1592994993 esp32xx_ota_backend:303 Writing FS @ 0x190000
1593271487 mgos_ota_core.c:685     2.27% total, fs.img 512 of 458752
1593279087 mgos_ota_core.c:1202    Update state: 0
1596631647 mgos_ota_core.c:685     18.13% total, fs.img 262656 of 458752
1596639355 mgos_ota_core.c:1202    Update state: 0
1599408737 esp32xx_ota_backend:399 fs_0 verified (429e0768a2f79d22bfe9fcbb5aaf057f1977d88b)
1599793739 esp32xx_ota_backend:299 Skip writing app (digest matches)
1599808646 mgos_ota_core.c:685     30.03% total, Plus1.bin 512 of 1156448
1599816325 mgos_ota_core.c:1202    Update state: 0
1600441530 shelly_main.cpp:449     Up 1600.43, HAP 0/0/16 ns 0, RAM: 200624/190360; st 44; disabled
1601097220 mgos_ota_core.c:685     45.89% total, Plus1.bin 262656 of 1156448
1601104796 mgos_ota_core.c:1202    Update state: 0
1602365390 mgos_ota_core.c:685     61.75% total, Plus1.bin 524800 of 1156448
1602373363 mgos_ota_core.c:1202    Update state: 0
1603624269 mgos_ota_core.c:685     77.61% total, Plus1.bin 786944 of 1156448
1603631862 mgos_ota_core.c:1202    Update state: 0
1604920368 mgos_ota_core.c:685     93.46% total, Plus1.bin 1049088 of 1156448
1604928924 mgos_ota_core.c:1202    Update state: 0
1605447556 mgos_ota_core.c:820     Reached the end of archive
1605459862 esp32xx_ota_backend:463 Setting boot partition to app_0
1605469123 [0;32mI (1606158) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=30944h (198980) map[0m
1605586052 [0;32mI (1606278) esp_image: segment 1: paddr=0004096c vaddr=3ffb0000 size=03630h ( 13872) [0m
1605603000 [0;32mI (1606288) esp_image: segment 2: paddr=00043fa4 vaddr=40080000 size=0c074h ( 49268) [0m
1605638405 [0;32mI (1606328) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=d1f40h (859968) map[0m
1606097992 [0;32mI (1606788) esp_image: segment 4: paddr=00121f68 vaddr=4008c074 size=085b0h ( 34224) [0m
1606125380 [0;32mI (1606818) esp_image: segment 5: paddr=0012a520 vaddr=50000000 size=00010h (    16) [0m
1606134834 [0;32mI (1606828) esp_ota_ops: New OTA data 0: seq 0x0000005b, st 0x10, CRC 0x12ad02a7[0m
1606194324 mgos_ota_core.c:1051    Update finished, result 1 (Update applied, rebooting)
1606200533 mgos_ota_core.c:1202    Update state: 0
```

markirb, 2024-08-18:
Mhhh everything looks okay. 

This does not look like a software issue to me unfortunately... :( 

I think somehow the device bots the new version and then reverts, because it did not boot successfully...

mike-jl, 2024-09-02:
same problem here..

Littlericket, 2024-09-10:
> Mhhh everything looks okay.
> 
> This does not look like a software issue to me unfortunately... :(
> 
> I think somehow the device bots the new version and then reverts, because it did not boot successfully...

I had similiar issues when trying to flash mongoose for the first time tho. It took me several attempts that the shelly finally booted into the mongoose os. It mostly came back with the original shelly firmware. I dont know if these shellys maybe need a special formatting of the firmware image, which differs than the current known layering for plus 1? But that would mean that I could never have flashed these shellys in the first place. I don‘t know whats happening. 
Still, I have one shelly plus which can not be flashed to mongoose. Maybe there‘s something going on with different hardware units?

markirb, 2024-09-10:
> I had similiar issues when trying to flash mongoose for the first time tho. It took me several attempts that the shelly finally booted into the mongoose os. It mostly came back with the original [...]

Exactly, this is the strange thing, the reason has to be somehow non-deterministic.

> I don‘t know whats happening. Still, I have one shelly plus which can not be flashed to mongoose. Maybe there‘s something going on with different hardware units?

There definitely seems to be something. But without a serial bootlog this won't be a problem we can debug. Unfortunately I have no device where this seems to happen.

> Interestingly, the flash-shelly python tool also doens‘t correctly detect these units. I can see them when running in verbose, but it seems there is no action being taken after being discovered with mdns. This happens with units on mongoose and on original firmware, which always come back as before.

This makes it even more strange... ?!

Littlericket, 2024-09-10:
@markirb I can send you one of the failing ones if you would like. I have some st links but not really into shelly debugging. 
I could imagine that these shellys maybe have two boot partitions and one is maybe read-only, whereas the other inactive one can be written on. Maybe the partition has to be switched to before a firmware upgrade. But thats only a thing I know from higher priced STM boards.

Peeeep, 2024-09-13:
I had the same issue and no indication of an error in the logs. Got it to work by first reverting to stock (1.3.3), factory resetting from the shelly ui, then flashing again using `http://A.B.C.D/ota?url=http://shelly.rojer.cloud/update`.

markirb, 2024-09-15:
I think it has to do with the filesystem somehow. On startup of the new fw it tries to copy over the old config. If it cannot to that a revert is triggered. 

That's why it can work on a freshly reset one.

xsaranaut, 2024-09-27:
I ran into this problem right now. Tried to reset to stock firmware - successfully. Then tried to update to 2.12.2 via ota, which wasn't successful either. So I captured the log, maybe it helps:

`Connected.
11:35:02
shelly_debug.cpp:236 Streaming logs to 192.168.172.189:62794
11:35:02
shos_rpc_inst.c:222 No handler for Shelly.GetInfo
11:35:10
shos_rpc_inst.c:222 No handler for Shelly.GetInfo
11:35:11
shelly_ota_source_ht:33 Update URL: http://shelly.rojer.cloud/update (0x3ffe431c)
11:35:53
shelly_http_client.:304 0x3ffe5154: HTTP GET http://shelly.rojer.cloud/update
11:35:53
shelly_ota.cpp:248 Starting update (0x3ffe3d44), to=600 cd=0/ct=0 ignore_same=0
11:35:53
shelly_http_client.:607 0x3ffe5154: Finished; bytes 416, code 302, redir 1/3, auth 0, status ABORTED: Abandoned
11:35:53
shelly_http_client.:304 0x3ffe5154: HTTP GET http://shelly.rojer.cloud/shelly-homekit-ShellyPlus1.zip
11:35:53
shelly_ota.cpp:375 Plus1-2.12.2/manifest.json: 1805 bytes (hdr 56 desc 0 CRC 7f043517)
11:35:53
shelly_ota.cpp:554 fw: Plus1 ver 2.12.2 20240912-182009/2.12.2-gfff6707
11:35:53
shelly_ota.cpp:182 OTA status in_progress, 0 %
11:35:53
shelly_update.cpp:332 incoming fw signatures: 00
11:35:53
shelly_ota_esp32_ba:512 Will write to slot 0
11:35:53
shelly_ota.cpp:375 Plus1-2.12.2/bootloader.bin: 23728 bytes (hdr 57 desc 0 CRC dc7241b3)
11:35:53
shos_init.c:94 New min heap free: 106808
11:35:53
shelly_notification:208 Event from sys: {"component":"sys", "event":"ota_begin", "msg":"in_progress", "ts":1727429753.37}
11:35:53
shos_init.c:94 New min heap free: 106652
11:35:53
shelly_ota_esp32_ba:709 Boot: cur 010000ff, new 00040127, min 00000000, update? 0
11:35:53
shelly_ota.cpp:375 Plus1-2.12.2/partition-table.bin: 3072 bytes (hdr 62 desc 0 CRC e2333c22)
11:35:53
shelly_ota_esp32_ba:620 New app_0: 1572864@0x10000
11:35:54
shelly_ota_esp32_ba:620 New fs_0: 458752@0x190000
11:35:54
shelly_ota.cpp:261 Aborting update: backend error while processing data: -1 (-11: PT update is req'd but unsupported by the new fw)
11:35:54
shelly_ota.cpp:182 OTA status error, 1 %
11:35:54
shelly_notification:208 Event from sys: {"component":"sys", "event":"ota_error", "msg":"error", "ts":1727429754.01}
11:35:54
shelly_http_client.:607 0x3ffe5154: Finished; bytes 33120, code 200, redir 0/2, auth 0, status CANCELLED: Cancelled`

markirb, 2024-09-27:
> I ran into this problem right now. Tried to reset to stock firmware - successfully. Then tried to update to 2.12.2 via ota, which wasn't successful either. So I captured the log, maybe it helps: [...]

This is not this issue. You are trying to upgrade from 1.4.x which is not supported ATM. See readme

xsaranaut, 2024-09-27:
Sorry, overlooked the 1.3.3 remark. Trying to downgrade it and then install 2.12.2 again.

Thank you @markirb

lindingerm, 2024-09-29:
Same problem here. Two ShellyPlus1 updated succesfully to 2.12.2. One is not updating. It stay on 2.12.1 (20240623-075054/2.12.1-g417cbdf)

Here is my Log:
```
348571358 shelly_ota.cpp:90       Starting firmware update
348584642 esp32xx_ota_backend:209 App: Plus1.bin -> app_0 enc 0, FS: fs.img -> fs_0 enc 0
348596050 mgos_ota_core.c:685     0.15% total, bootloader.bin 512 of 23728
348603842 mgos_ota_core.c:1202    Update state: 0
348725539 mgos_ota_core.c:685     1.59% total, partition-table.bin 512 of 3072
348733377 mgos_ota_core.c:1202    Update state: 0
348759798 mgos_ota_core.c:685     1.78% total, otadata.bin 512 of 8192
348767598 mgos_ota_core.c:1202    Update state: 0
348964620 esp32xx_ota_backend:303 Writing FS @ 0x190000
349221496 mgos_ota_core.c:685     2.27% total, fs.img 512 of 458752
349229388 mgos_ota_core.c:1202    Update state: 0
349236405 mgos_mongoose.c:66      New heap free LWM: 193308
349408982 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:50612 user admin
349474491 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:63829 user admin
350410630 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:50612 user admin
350476622 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:63829 user admin
351714840 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:50612 user admin
351797030 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:63829 user admin
352541662 mgos_ota_core.c:685     18.13% total, fs.img 262656 of 458752
352549547 mgos_ota_core.c:1202    Update state: 0
352566306 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:50612 user admin
352802096 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:63829 user admin
353411653 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:50612 user admin
353475792 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:63829 user admin
353891901 shelly_main.cpp:483     Up 353.88, HAP 0/0/16 ns 0, RAM: 202160/193308; st 47; disabled
354413228 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:50612 user admin
354502078 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:63829 user admin
355243288 esp32xx_ota_backend:399 fs_0 verified (27a5f139e357555037b1d626e6304909684d6576)
355630948 esp32xx_ota_backend:299 Skip writing app (digest matches)
355649924 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:50612 user admin
355670619 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:63829 user admin
355681480 mgos_mongoose.c:66      New heap free LWM: 193008
355688790 mgos_ota_core.c:685     30.03% total, Plus1.bin 512 of 1156448
355696555 mgos_ota_core.c:1202    Update state: 0
355709858 mgos_mongoose.c:66      New heap free LWM: 192344
356414630 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:50612 user admin
356474506 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:63829 user admin
356964765 mgos_ota_core.c:685     45.89% total, Plus1.bin 262656 of 1156448
356972639 mgos_ota_core.c:1202    Update state: 0
357416681 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:50612 user admin
357474422 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:63829 user admin
358232438 mgos_ota_core.c:685     61.75% total, Plus1.bin 524800 of 1156448
358240348 mgos_ota_core.c:1202    Update state: 0
358415224 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:50612 user admin
358475636 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:63829 user admin
359417391 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:50612 user admin
359473618 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:63829 user admin
359544691 mgos_ota_core.c:685     77.61% total, Plus1.bin 786944 of 1156448
359552543 mgos_ota_core.c:1202    Update state: 0
360418348 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:50612 user admin
360473104 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:63829 user admin
360811171 mgos_ota_core.c:685     93.47% total, Plus1.bin 1049088 of 1156448
360819046 mgos_ota_core.c:1202    Update state: 0
361316518 mgos_ota_core.c:820     Reached the end of archive
361329952 esp32xx_ota_backend:463 Setting boot partition to app_0
361339198 [0;32mI (361959) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=30944h (198980) map[0m
361454721 [0;32mI (362079) esp_image: segment 1: paddr=0004096c vaddr=3ffb0000 size=03630h ( 13872) [0m
361472355 [0;32mI (362089) esp_image: segment 2: paddr=00043fa4 vaddr=40080000 size=0c074h ( 49268) [0m
361507640 [0;32mI (362129) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=d1f40h (859968) map[0m
361974378 [0;32mI (362599) esp_image: segment 4: paddr=00121f68 vaddr=4008c074 size=085b0h ( 34224) [0m
362001665 [0;32mI (362619) esp_image: segment 5: paddr=0012a520 vaddr=50000000 size=00010h (    16) [0m
362011065 [0;32mI (362629) esp_ota_ops: New OTA data 1: seq 0x00000013, st 0x10, CRC 0x9141b478[0m
362064461 mgos_ota_core.c:1051    Update finished, result 1 (Update applied, rebooting)
362070898 mgos_ota_core.c:1202    Update state: 0
362075688 mgos_ota_core.c:1096    Update requires reboot
362128400 mgos_mongoose.c:66      New heap free LWM: 190448
362147604 shelly_main.cpp:483     Up 362.14, HAP 0/0/16 ns 0, RAM: 206420/190448; st 47; disabled
362418018 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:50612 user admin
362474344 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.31:63829 user admin
```

lindingerm, 2024-10-03:
I was able to work around the problem as follows:
- Go to "Security Settings" and removed the password
- Downloaded the mongoose phyton script from [here](https://github.com/mongoose-os-apps/shelly-homekit/archive/master.zip)
- Run the script with following arguments: `flash-shelly.py -am keep`

The device is now updated successfully to 2.12.2

markirb, 2024-10-07:
Can anyone confirm that this is related to password being set?

Littlericket, 2024-10-07:
@markirb I have no passwords set.

PaulmichlJ, 2024-10-09:
since today I'm in the same trap. reverted two shelly plus 1 to ad the addon with DS18B20. Added 4 of them successfully under 1.3.3. Did the same on two different shelly plus 1 . . . 
trying then to flash "back" to homekit firmware fails without error message - just boomerang back always with 1.3.3 exactly as the other reported :-(
tried everything from this stream here including the password add/remove . . .

HOPE that there is someone with a clue what could prevent a homekit firmware flash after a stock revert

THANK's in advance for any help.

[image]

markirb, 2024-10-09:
> since today I'm in the same trap. reverted two shelly plus 1 to ad the addon with DS18B20. Added 4 of them successfully under 1.3.3. Did the same on two different shelly plus 1 . . . trying then to flash "back" to homekit firmware fails without error message - just boomerang back always with 1.3.3 exactly as the other reported :-( tried everything from this stream here including the password add/remove . . .
> 
> HOPE that there is someone with a clue what could prevent a homekit firmware flash after a stock revert
> 
> THANK's in advance for any help.
> 
> Btw: I tried also an "update" to 1.4.2 which worked well and an "downgrade" to 1.3.3 which also runes fine . . . but NO homekit anymore :-(
> 
> UPDATE: I removed the addon an bum - the homekit update runes fine - BUT If I add the addon again - the device isn't able to boot anymore. removed the addon again and the device reboots. revered back to 1.3.3 the addon is working fine again.
> 
> FINAL SOLUTION: removed on DS18B20 and tried everything again with just three of them - all fine. therefore I will use two shellys to cover the need of five measure points . . . .
> 
> would be great if the developers could verify that and confirm the only three DS18B20 are supported under there homekit firmware which is really a great game changer from my point of view. Thank you for your great work so far !
> 
> ![image](https://private-user-images.githubusercontent.com/91956867/375045657-eeb8969f-af13-498f-bfc3-61a09db0c2bb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg0OTg5MjUsIm5iZiI6MTcyODQ5ODYyNSwicGF0aCI6Ii85MTk1Njg2Ny8zNzUwNDU2NTctZWViODk2OWYtYWYxMy00OThmLWJmYzMtNjFhMDlkYjBjMmJiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA5VDE4MzAyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkwOWJlYTMwNjZjNjc4MmVlZGY3NTYxYTY0NDc1ODczMmUyZWE4ZDk1NGZiNmIyNmEwNWYzYzExZmJkMDYyZTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.PDBvRUhBizU1b31iM4W8a9oc7yfp6kGb8A462LWxMms)

Yes currently the limit is three. As this used to be the old limit. I never got around to test more, but the limit is rather artificial at least from software perspective. For hardware (addon differences) there can be one. Can you open a new issue from this please? It is not hard to remove the limit. Just wondering why it would crash...

PaulmichlJ, 2024-10-09:
> Yes currently the limit is three. As this used to be the old limit. I never got around to test more, but the limit is rather artificial at least from software perspective. For hardware (addon [...]

thank you for your quick confirmation of my tests - just did it gain on a brand new device - same as you confirmed in the meantime. I'm fine with that limit - just didn't realized the "trap" early enough ;-)

markirb, 2024-10-10:
> thank you for your quick confirmation of my tests - just did it gain on a brand new device - same as you confirmed in the meantime. I'm fine with that limit - just didn't realized the "trap" early [...]

can you test this: 
[shelly-homekit-ShellyPlus1.zip](https://github.com/user-attachments/files/17322895/shelly-homekit-ShellyPlus1.zip)

this should have maximum of 5 for Plus devices

markirb, 2024-10-10:
As for the original Problem of this post. Can you upgrade to 1.4.X and then use the newest beta release:

https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.13.0-beta1

please provide feedback here

Littlericket, 2024-10-10:
@markirb I was able to flash all my shellys with the new firmware version. But besices the first stock reset (from HA to shelly 1.3.3), i had to factory reset afterwards. The shelly on 1.3.3 was visible and detected by the shelly app, but it wasnt making a cloud connection. I tried opening the web UI (which worked), tried enabling cloud but it did always reset to unchecked after a few secs. 
After the additional firmware reset on 1.3.3 I was able to add the shelly device and upgrade to 1.4.2. On 1.4.2, I was able to upgrade to 2.13. Also the ones which failed (resulting in this ticket).

I have one thing to add, I‘ve tested the password option in https://github.com/mongoose-os-apps/shelly-homekit/issues/1433#issuecomment-2395953385 but didnt succeed to upgrade afterwards. Seems like something got broken with the partitions on some of the shellys. I don‘t want to know what really caused this but it works with 2.13. Thats all what matters.  :D

Thanks for your efforts!

PaulmichlJ, 2024-10-11:
> can you test this: [shelly-homekit-ShellyPlus1.zip](https://github.com/user-attachments/files/17322895/shelly-homekit-ShellyPlus1.zip)
>
> this should have maximum of 5 for Plus devices

IT's fine :-) all five DS18B20 appears immediately also in HomeKit - THANK YOU !!
Would it be possible to include that "five" also in your new HomeKit beta?

markirb, 2024-10-20:
I am closing this now. Currently I think it had to do with filesystem incompatibilities. Should be fixed if you use the latest firmware (with possible upgrade path through 1.4.X) if not then we can discuss in a new thread or reopen this one

closed 2024-10-20 by markirb