mongoose-os-apps/shelly-homekit #1434: Shelly Plus 1 not flashing Shelly-homekit

rmmmrm, 2024-08-09:
Hello all

Im having an issue flshing shelly-homekit on my shelly plus 1.

Already downgraded my firmware to 1.3.3 
[Plus1-1.3.3.zip](https://github.com/user-attachments/files/16561771/Plus1-1.3.3.zip)

The shelly-homekit firmware installs but at the end when the device reboots its still the same shelly 1.3.3.
Also tryed to downgrade to 0.14 and didnt work

Can anyone help me with this issue?  

heres the log
[diagnostics-shelly-plus1-data (1).txt](https://github.com/user-attachments/files/16561784/diagnostics-shelly-plus1-data.1.txt)
[diagnostics-shelly-plus1-debug-log (1).txt](https://github.com/user-attachments/files/16561787/diagnostics-shelly-plus1-debug-log.1.txt)

Regards

markirb, 2024-08-22:
This is strange. There is no error in upgrade process.

You say you cannot downgrade to 0.14 also?
This might be a hint.... 

Unfortunately the only way to know the reason would be a serial log after the device tries to boot up in the new firmware.
It seems to revert itself afterwards because it thinks it cannot successfully initialize.

markirb, 2024-09-10:
I will close this, lets continue in https://github.com/mongoose-os-apps/shelly-homekit/issues/1433

There seems to be something but not known yet.

closed 2024-09-10 by markirb