mongoose-os-apps/shelly-homekit #1438: shelly Plug S non funziona flash via OTA

Sindachetto84, 2024-08-18:
ciao a tutti, 
non riesco più a fare il flash via OTA del mio shelly Plug S dopo reset factory. ho provato ad utilizzare tutti i downgrade del firmware ma non funziona con nessuno.
grazie per il supporto.

markirb, 2024-08-22:
Does not work is not an issue description. Provide at least logs and from which firmware you try to update.

Sindachetto84, 2024-08-22:
> Non funziona non è una descrizione del problema. Fornire almeno i registri e da quale firmware si tenta di aggiornare.

Hi Markirb,  sorry but I don't know how to share the logs, I tried with all the downgrades available but I can't flash the HomeKit firmware.  Thanks

markirb, 2024-08-23:
Which downgrades do you mean?

Here ist howtio get logs:
https://community.shelly.cloud/topic/1707-how-to-enable-debug-mode-on-your-shelly/

ilGenu, 2024-08-25:
Here's the log

[diagnostics-shelly-plusplugs-debug-log-3.txt](https://github.com/user-attachments/files/16741667/diagnostics-shelly-plusplugs-debug-log-3.txt)

markirb, 2024-08-25:
Seems like the automatic url is not yet working. Please download file manually from releases tab.

markirb, 2024-09-12:
shoudl work now

closed 2024-09-12 by markirb