mongoose-os-apps/shelly-homekit #1439: +2PM with Shelly Plus Add-On Temperature Sensor DS18B20 not available

Perth-93, 2024-08-18:
I have a Shelly + 2PM with Shelly Plus Add-On. A Ds18B20 is attached to the Add-On a worked flawlessly und the original Firmware 1.4 i.e. it’s was visible via the built.in Webserver and the Shelly App without further configuration.

After switching to HomeKit Firmware 2.12.1 the sensor is not available any more. The firmware does obviously
Recognize the Shelly Plus Add-On as there is an additional Input setting available (“Input 3”). But there are only the following options available: “Disabled”, “Stateless Switch”, “Motion Sensor”, “Occupation Sensor”, “Contact Sensor”, “Doorbell”, “Leak Sensor”, “Smoke Sensor”, “Carbon Monoxide Sensor”, “ Carbon Dioxide Sensor”. 
No Temperature Sensor, although the DS18B20  was recognized under the original firmware.

Is this a bug or is there a special wiring required as there are different option to attach a DS18B20 to the Shelly Plus Add-On?

Everything else ist fine, so thanks for making Shelly devices work with HomeKit!

markirb, 2024-08-19:
Hi,

nothing special should be required.

If it detects the add-on then the first step seems correct. 

I thought I have reports with Plus Addons with DS sensors running, so there should be no general problem. So it might be the driver not supporting your flavor? :(

This is hard to debug, even more so remotely....

Perth-93, 2024-08-19:
Hi,

thx for your fast reply. Your answer and its hints at possible underlying problems makes me ask a few short questions in return:

Isn‘t the driver the same as in the original firmware?

Are there really sigifinicantly different flavors of the DS18B20 - I thought it was very basic and standardized?

Is there a chance that it only works on on of the three possible connection options on the Shelly Add-On. So far I have tried the (seen from top) left connectors, where you can connect up to three DS18B20 parallel and the right ones, where you can only connect one DS18B20. I haven‘t tried yet the middle connectors where you could connect on DHT22 (all as layer out in the manual to the original firmware)?

Could resetting the HomeKit firmware help detecting the sensor?

Best regards

😀

markirb, 2024-08-21:
> Hi,
> 
> 
> 
> thx for your fast reply. Your answer and its hints at possible underlying problems makes me ask a few short questions in return:
> 
> 
> 
> Isn‘t the driver the same as in the original firmware?
> 

It could, but we do not know. It was not for sure some time ago, as the one-wire code via separate pin was done by me.

> 
> Are there really sigifinicantly different flavors of the DS18B20 - I thought it was very basic and standardized?
> 
> 

Thought so too...

> Is there a chance that it only works on on of the three possible connection options on the Shelly Add-On. So far I have tried the (seen from top) left connectors, where you can connect up to three [...]
This makes now difference for sure.

> 
> Could resetting the HomeKit firmware help detecting the sensor?
> 
> 

No. Only thing I can think of is unreliable power at boot. But that is far fetched.

> 
> Best regards
> 
> 
> 
> 😀

If I come up with time or an idea then I will write here.
Would be interesting if anybody has the same problem or a working sensor.

Perth-93, 2024-08-23:
Update:
I bought a different brand of DS18B20 and tried two different sensors at all three available connection options, completely resetting the Shelly each time. Still I had no luck, no temperature was read out and shown in the built-in webserver. 
So maybe the plus 2PM does not really support a DS18B20 temperature sensor, as opposed to the original firmware? 
I hope somebody with debugging skills and a DS18B20 ca give it a try…

markirb, 2024-08-24:
Ok, thanks for trying then it seems a difference on Plus2PM.

Anyway When I have released Mini Support I think I can get an Addon and try. But my time is very limited currently.

ballmercurve, 2024-09-02:
I'm having the same issue. Plus2PM with the add-on and a DS18B20. Only options in the WebUI are discrete switches, nothing as far as temperature. 
[image]

Perth-93, 2024-09-02:
Hi ballmercurve,

thx for posting your experience, too.

So we keep our fingers crossed  that markirb will be able to spend some time for looking into this in the near future…

markirb, 2024-09-08:
Hi,

I think I found it. Was too easy to see it and explains why it only did not work on +2PM.

can you try this:

[shelly-homekit-ShellyPlus2PM.zip](https://github.com/user-attachments/files/16922589/shelly-homekit-ShellyPlus2PM.zip)

Perth-93, 2024-09-08:
Hi,

I uploaded it, but so far I cannot see any difference. 

Should the temperatur of the sensor appear somewhere in the built-in GUI (how is it being shown on the 1PM+, which allegedly works)?

markirb, 2024-09-08:
Exactly should work like the other pluses. At least for me it does now and it did not before

20240908-142515/2.12.2-beta3-1-g82596f1-master

Perth-93, 2024-09-08:
Where or how do you see the temperature in the built in GUI?

markirb, 2024-09-08:
[image]

Perth-93, 2024-09-08:
Thx for the clarification.
Just to be sure I reverted to the the original firmware 1.3.3 (which is easy now) and there is non sensor available either. So I assume the problem is now on my side which I probably created when testing several sensors. I have to check the hardware and make the sensor to reappear in the original firmware. After that I will most probably see it in the provided beta, too.
There a obviously different „types“ of temperatures in HomeKit. Thermostats have a big icon and you can click on it to set a new temperature. Does the temperature via the Shelly have the option to be used as a thermostat?

closed 2024-09-12 by markirb

Perth-93, 2024-09-13:
I can confirm that a DS18B20 now works flawlessly when attached to a Shelly +2 with a Shelly + Add-On
Thx Mark!