mongoose-os-apps/shelly-homekit #1442: Version 1.4.2 on Shelly plus 2PM

MacUser341, 2024-08-24:
Hello
As I understand the HomeKit support isn’t working with version 1.4 and up and I need to revert to 1.3.3

first is there any ETA to release an update that will support the new version?

second.  I’ve searched a lot and I don’t understand how to revert the 1.4 stock firmware to the 1.3.3 stock firmware. I’ve seen the link to download but I can’t seem to find a manual - any help will be appreciated.

markirb, 2024-08-25:
Hi,

There is no ETA, as supporting this completely is not an easy task; otherwise this would be done already.

They change partition sizes (64kb go from fs to app) during upgrade which is... a real pain to reprogram.

Currently I see three ways to support this which all have their downsides.

1. is to provide two different versions of this firmware for each partition layout. The downside would be that we can never really use the larger app partition, but we probably do not need the space. However it is unclear if the automatic updater can differentiate between devices on two different partition tables. And to explain to people that they need other versions depending on their current partition layout is also a pain.

2. is two have only a firmware for the new partition sizes. This would require everyone on the old Shelly-HomeKit version do three updates for each device if they want a new version. (Revert to 1.3.3 then go to 1.4.x then to Shelly HomeKit again). This is not very nice

3. try out if using a the smaller fs is accepted by stock. Then we would not need to care from which stock fw we update. However a revert to stock would need the version for the 1.3.3 if the partition table is still old.

You see.... not very cool situation. What would you do?

In Shelly stock firmware you have a form where you can upload zips in the firmware section. Just put the file there and hit upload/update.

Br

Baionar, 2024-08-27:
Bonjour
Je profite du post pour remercier tous les développeurs et faire un petit retour d'expérience à MacUser341.
Après avoir téléchargé les 2 zip (1.3.3 + version Shelly Homekit), j'ai accédé à l'adresse IP de mon shelly avec mon naviguateur Edge PC. La page affichée et les fonctions sont claires. A la rubrique Firmware je suis revenu de la version officielle du moment à la version 1.3.3 puis une fois l'opération terminée chargement du firmware Shelly Homekit. Ensuite paramétrage des autres rubriques via la même interface.
Déroulement sans pbm, cela a été rapide et les résultats sont OK.
Bravo et merci encore aux développeurs

oli-d, 2024-09-28:
> Hi,
> 
> There is no ETA, as supporting this completely is not an easy task; otherwise this would be done already.
> 
> They change partition sizes (64kb go from fs to app) during upgrade which is... a real pain to reprogram.
> 
> Currently I see three ways to support this which all have their downsides.
> 
> 1. is to provide two different versions of this firmware for each partition layout. The downside would be that we can never really use the larger app partition, but we probably do not need the space. However it is unclear if the automatic updater can differentiate between devices on two different partition tables. And to explain to people that they need other versions depending on their current partition layout is also a pain.
> 2. is to have only a firmware for the new partition sizes. This would require everyone on the old Shelly-HomeKit version do three updates for each device if they want a new version. (Revert to 1.3.3 then go to 1.4.x then to Shelly HomeKit again). This is not very nice
> 3. try out if using a the smaller fs is accepted by stock. Then we would not have to care from which stock fw we update. However a revert to stock would need the version for the 1.3.3 if the partition table is still old.
> 
> You see.... not very cool situation. What would you do?
> 
> In Shelly stock firmware you have a form where you can upload zips in the firmware section. Just put the file there and hit upload/update.
> 
> Br

If this is a public poll, my vote would be for suggested version 2. 😊

Yes, it is not a super nice upgrade path for people (like me) using older devices. But 1.4 is out for quite a while, it's the new way forward, all new devices come stock with that. At the moment, life is hard for everyone getting a new device. Including newbies that need explanations on downgrading stock etc. If alternative 2 would be implemented, the flashing for the new devices would be seamless again, very easy for newbies. Also keep in mind that the old devices will die at some point in the future, so the audience that suffers from alternative 2 is getting smaller and smaller over time anyways...

markirb, 2024-10-01:
You are right with the argumentation for 2.

I think there are three points to add, that are some work...

There should be a (python) automated way to not have to flash the old ones 3 times. 

It has to be certain that no config bit is lost during the upgrades to stock.

We should have a description in web interface why the upgrade fails on plus devices for old users to not provoke support requests when they try to update via webinterface. There should be a description for the process for automated (python) and manual (maybe urls with autodetection of device) update way

Salle88, 2024-10-08:
Is there an update planned ?

oli-d, 2024-10-26:
Sorry for the late reply. Got to testing it today (with the proper 2.13.0 release) and it worked like a charm!

Thanks for the great work to everyone involved! ♥