mongoose-os-apps/shelly-homekit #1444: Shelly Plus 1 Doorbell 12 V AC - only "long" press is detected

jan666, 2024-08-26:
Hello,

I used a Shelly Plus 1 as Doorbell for a few weeks. My old doorbell uses 12 Volt AC so I have running via 12 V transformer rectifier. The physical switch still as 12 V AC which is connected to SW. This works without problems in the stock shelly firmware. But only without eco mode.

I used it via Home Assistant to get notifications.

Now I noticed it's possible to add my setup to Homekit without the need for home assistant. So I installed the Homekit firmware and set it up. Everything looks great but then I noticed I started missing rings. So I began to test and noticed I have to press the switch longer than with the shelly stock firmware to trigger it. So when you just press it for a very short time it does not work. I just reverted to shelly firmware and the problem is gone. 

Is this a bug? I'd prefer to use the homekit firmware but I guess I will miss most rings.

markirb, 2024-08-26:
Hi,

maybe can you post some logs?

There is a detection for long and short presses, which might somehow lead to that.

jan666, 2024-08-26:
I opened the log and pressed a few times (I think at least 6 times). Only got 3 notifications from homekit:

```
211058381 shelly_debug.cpp:231    No log file, sending new entries
217849960 shelly_main.cpp:483     Up 217.84, HAP 0/1/16 ns 1, RAM: 209804/193964; st 48; 0.1: st:0 in_st:0 inm:3 ininv:0; 10.1: svt:10 inv:0 st:0 m:0 lea: -1.000
218113616 mgos_http_server.c:180  0x3ffcd42c HTTP connection from 192.168.42.144:53698
218244018 mgos_http_server.c:180  0x3ffcdcd8 HTTP connection from 192.168.42.144:53699
218260265 mg_rpc.c:315            Shelly.GetInfo via WS_in 192.168.42.144:53699
218277610 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
219325297 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
220245443 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
221269452 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
222229410 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
223318637 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
224239190 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
225263929 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
225849966 shelly_main.cpp:483     Up 225.84, HAP 0/1/16 ns 1, RAM: 208972/193964; st 47; 0.1: st:0 in_st:0 inm:3 ininv:0; 10.1: svt:10 inv:0 st:0 m:0 lea: -1.000
226287931 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
227323022 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
228336244 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
230602037 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
231306573 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
232329785 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
233353758 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
233849610 shelly_main.cpp:483     Up 233.84, HAP 0/1/16 ns 1, RAM: 209184/193964; st 47; 0.1: st:0 in_st:0 inm:3 ininv:0; 10.1: svt:10 inv:0 st:0 m:0 lea: -1.000
234481680 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
235402695 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
236335838 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
236418512 shelly_input.cpp:77     Input 1: change (state 1)
236479665 shelly_input.cpp:77     Input 1: change (state 0)
236518080 shelly_input.cpp:77     Input 1: change (state 1)
237454093 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
237523512 shelly_input.cpp:77     Input 1: long (state 0)
237528638 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 2
238373028 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
239396538 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
240316928 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
241343226 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
241850022 shelly_main.cpp:483     Up 241.84, HAP 0/1/16 ns 1, RAM: 209184/193964; st 48; 0.1: st:0 in_st:0 inm:3 ininv:0; 10.1: svt:10 inv:0 st:0 m:0 lea: 4.312
242262050 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
243285698 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
244282186 shelly_input.cpp:77     Input 1: change (state 1)
244295345 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
244331129 shelly_input.cpp:77     Input 1: change (state 0)
244361747 shelly_input.cpp:77     Input 1: change (state 1)
244424161 shelly_input.cpp:77     Input 1: change (state 0)
244462629 shelly_input.cpp:77     Input 1: change (state 1)
244524286 shelly_input.cpp:77     Input 1: change (state 0)
244528964 shelly_input.cpp:77     Input 1: double (state 0)
244533978 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
244563574 shelly_input.cpp:77     Input 1: change (state 1)
244624341 shelly_input.cpp:77     Input 1: change (state 0)
244661957 shelly_input.cpp:77     Input 1: change (state 1)
244724310 shelly_input.cpp:77     Input 1: change (state 0)
244728983 shelly_input.cpp:77     Input 1: double (state 0)
244733995 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
245348149 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
246360260 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
247279673 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
248304193 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
249327633 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
249850171 shelly_main.cpp:483     Up 249.84, HAP 0/1/16 ns 1, RAM: 209184/193964; st 48; 0.1: st:0 in_st:0 inm:3 ininv:0; 10.1: svt:10 inv:0 st:0 m:0 lea: 5.107
250351752 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
251273408 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
251706656 shelly_input.cpp:77     Input 1: change (state 1)
252298057 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
252712160 shelly_input.cpp:77     Input 1: long (state 0)
252717274 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 2
253321041 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
[... cut ...]
266327576 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.42.144:53699
273850069 shelly_main.cpp:483     Up 273.84, HAP 0/1/16 ns 1, RAM: 207644/193964; st 47; 0.1: st:0 in_st:0 inm:3 ininv:0; 10.1: svt:10 inv:0 st:0 m:0 lea: 21.123
281849625 shelly_main.cpp:483     Up 281.84, HAP 0/1/16 ns 1, RAM: 209208/193964; st 47; 0.1: st:0 in_st:0 inm:3 ininv:0; 10.1: svt:10 inv:0 st:0 m:0 lea: 29.123
289849489 shelly_main.cpp:483     Up 289.84, HAP 0/1/16 ns 1, RAM: 209208/193964; st 47; 0.1: st:0 in_st:0 inm:3 ininv:0; 10.1: svt:10 inv:0 st:0 m:0 lea: 37.123
```

markirb, 2024-08-26:
Your short presses are interpreted as double taps and thus trigger only one ring... ?

jan666, 2024-08-26:
I dont really know how to read the logs. But I left it untouched for 30 minutes before, made one ring and it was not recognized.  That's why I started investigating

markirb, 2024-08-28:
No it does not really make sense. Or maybe only after we look deeper.

There is a detection mechanism for short, long and double press, which is actually not needed for a bell. Maybe this stands in the way.

Can you really isolate the event in time when you rest and try to annotate in your log what you really did?

jan666, 2024-08-29:
> Can you really isolate the event in time when you test and try to annotate in your log what you really did?

I will try to make some better tests when I find time (might take a while)

timoschilling, 2024-09-03:
I can confirm that, I have a similar behavior with an UNI.