mongoose-os-apps/shelly-homekit #1446: Shelly RGBW2 not shown under light sources in Homekit

WRTHacker, 2024-09-01:
Hello experts,
first of all many thanks that the firmware development is ongoing!!

Even with the new firmware I have the problem, that the shelly rgbw2 devices (using rgb+w mode, bridge + rgb + white devide shown in Homekit) are not shown under light sources. Any other vendor led controller device can be found in this Homekit category view. 

Is this known and intended?

Many thanks for support and best regards

markirb, 2024-09-01:
Can you please post your complete configuration screen from web interface

WRTHacker, 2024-09-01:
Hi, here are the screenshots:

- I use the RGB Device for the status display of my alarm system, and to display open windows
- The white device is uses as a nightligt

[image]
[image]

markirb, 2024-09-01:
Normally there is an option HAP service, can you find out why it is not shown for you

[image]

WRTHacker, 2024-09-01:
Even with an unuses RGBW2, reset configuration. There is no such Option!

WRTHacker, 2024-09-01:
However, when I add the test device to homekit, the devices are shown unter light ... think I hade to delete ann readadd the "faulty device" ... will come back if this has helped.

However THE HAP Switch is not there ...

WRTHacker, 2024-09-01:
After deletion and re add the devices occur unter light sources.
However, I remember, when the Homekit controller switches from one hub to another, the bridge is sometimes lost as a device. I will have to observe this with the new firmware and after the re-add.

WRTHacker, 2024-09-23:
Please keep this open until I upgrade to 18.0.x. Typically the change of the HomeKit controller leads to the problem, like it happens during upgrades.

WRTHacker, 2024-10-08:
After upgrading to 18.0(.1) the RGB bridge an the rgb and white device ist stable available.

=> However I have issued with my 1PM and Plus1PM: The 1PM was not reachable, than the switch disappered in Homekit, and the switch function „moved“ to the bridge. I have to remove the device an readd it. However the bridge is not there, but the single switch.

This behaviour occured even before the upgrade to V18 and the latest shelly firmware.

WRTHacker, 2024-10-13:
The initial issue is fixed, but the above one still remains.
=> Will open another issue

closed 2024-10-13 by markirb