mongoose-os-apps/shelly-homekit #1448: Feature Request: Window Support Shelly 2pm

Nastras, 2024-09-03:
Hi, would it be possible for you to support 2pm as an additional mode window?

That would be great. Thanks a lot for your work!

markirb, 2024-09-03:
I do not quite understand. Is roller shutter not the one you are looking for?

Nastras, 2024-09-03:
I not only control window covers with the shelly but also an electric window. It would be great if you could add the Shelly as a window device. Then Siri would also know that it is a window and not a window cover.

I hope it is more understandable now?

Thank you 😉

markirb, 2024-09-03:
I see. This is a new HAP type

Nastras, 2024-09-03:
I don't think it's that new, with Velux Integra you can also control the windows in Homekit and the window cover as an extra device. This has been possible for quite a long time.

It would be very cool if you could implement this so that you can differentiate between windows and window covers.

markirb, 2024-09-03:
It is not new. Just the support would be new.