mongoose-os-apps/shelly-homekit #1508: Homekit showing wrong state for roller shutter on Shelly 2.5

victor-vz, 2024-09-05:
I am using a Shelly 2.5 in roller shutter mode and got it connected via Home Assistant Homekit Controller/Bridge and exposed back to the Home App.

Since a week my Roller Shutter shows wrong states. Either it say opening for multiple hours or shows open in Homekit/HA even though it got closed via an Automation.

I am on the latest firmware on everything and unsure if this is a buck in the Shelly Software or in HA.

victor-vz, 2024-09-05:
Rebooting the Shelly fixes the issue.

Version:2.12.1
Build:
[20240623-075101](tel:20240623-075101)/2.12.1-g417cbdf

gmuth, 2024-09-28:
I have a similar problem. Position (0-100) does not reflect the actual physical state. Usually the shutter get's 'stuck' opened. The state says 0 while the shutter is completely open. Closing requests er rejected or ignored. I guess because it thinks it's already closed. (lates software is used)
Workaround: calibrate again.

markirb, 2024-10-03:
Thanks for posting. I think this is a duplicate of https://github.com/mongoose-os-apps/shelly-homekit/issues/1420

lets continue there. I tried to push a firmware there which might fix it

closed 2024-10-03 by markirb