mongoose-os-apps/shelly-homekit #1509: Flashing Shelly MiniG3

GSumman, 2024-09-06:
I've flashed many shelly devices before. I've recently brought a Shelly MiniG3. It's on the supported devices list. The device is on firmware version 1.3.3 and I'm unable to flash it. No error messages and no matter how long I wait, the device never flashes to the custom firmware. I've also rebooted the device and issue still remains. 
[image]

markirb, 2024-09-07:
Minis are currently only supported with latest beta and thus the url does not work yet, it will once it is out of beta. Please flash manually via file you find in releases section (latest pre-release)

closed 2024-09-07 by markirb

djrams10, 2024-09-12:
> I've flashed many shelly devices before. I've recently brought a Shelly MiniG3. It's on the supported devices list. The device is on firmware version 1.3.3 and I'm unable to flash it. No error messages and no matter how long I wait, the device never flashes to the custom firmware. I've also rebooted the device and issue still remains. <img alt="Screenshot 2024-09-06 175052" width="785" src="https://private-user-images.githubusercontent.com/41484965/365246451-3dd9d2f4-1cb5-442a-8d51-d1b697b86189.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYxNjI0ODEsIm5iZiI6MTcyNjE2MjE4MSwicGF0aCI6Ii80MTQ4NDk2NS8zNjUyNDY0NTEtM2RkOWQyZjQtMWNiNS00NDJhLThkNTEtZDFiNjk3Yjg2MTg5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTEyVDE3Mjk0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEwMWU0NTkyZDUzZjBmNGE3MzY0OThiMTg4MGUwMzVkZTBhMDc4MjQzNmIzMTMwMjk0ZDMzNzFjZjNjMzNiZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8ulcTmKMGyeKqhlo4R2WM-hiZbYOqcmgdNXtvqBtR-0">

Thank you, can you explain to me how you did to install the beta? Thanks

GSumman, 2024-09-12:
> > I've flashed many shelly devices before. I've recently brought a Shelly MiniG3. It's on the supported devices list. The device is on firmware version 1.3.3 and I'm unable to flash it. No error messages and no matter how long I wait, the device never flashes to the custom firmware. I've also rebooted the device and issue still remains. <img alt="Screenshot 2024-09-06 175052" width="785" src="https://private-user-images.githubusercontent.com/41484965/365246451-3dd9d2f4-1cb5-442a-8d51-d1b697b86189.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYxNjI0ODEsIm5iZiI6MTcyNjE2MjE4MSwicGF0aCI6Ii80MTQ4NDk2NS8zNjUyNDY0NTEtM2RkOWQyZjQtMWNiNS00NDJhLThkNTEtZDFiNjk3Yjg2MTg5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTEyVDE3Mjk0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEwMWU0NTkyZDUzZjBmNGE3MzY0OThiMTg4MGUwMzVkZTBhMDc4MjQzNmIzMTMwMjk0ZDMzNzFjZjNjMzNiZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8ulcTmKMGyeKqhlo4R2WM-hiZbYOqcmgdNXtvqBtR-0">
> 
> Thank you, can you explain to me how you did to install the beta? j'ai trouvé le fichier, But I don't know how to update it, Thanks

https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.2-beta3 See this link. Worked a dream.

markirb, 2024-09-12:
just released, so current link is:

https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.2

or you take the manual url from readme, works now to. general update url should work in a few hours