mongoose-os-apps/shelly-homekit #1510: HomeKit show opposite position / Obstruction detection not working

MacUser341, 2024-09-08:
Hello
I have 2 issues :
I have 3 Roller shutters and in 2 of them the positions are correct but in my porch door roller shutter when it’s open completely it shows in HomeKit that it’s close completely - is there a way to change it? I did 2 calibration to this roller shutter.

the second issue that worries me is that I see in the home app that the Obstruction detection isn’t working and I know in the original Shelly app there’s that protection that is something in the way the shutter will stop - is it something I loose if I choose to use this firmware ?

thanks

markirb, 2024-09-12:
Hi, 

there is an obstruction detection. It is triggered when power is 2.5 higher than normal. I guess this is too much for your case?

MacUser341, 2024-09-16:
> Hi,
> 
> there is an obstruction detection. It is triggered when power is 2.5 higher than normal. I guess this is too much for your case?

I’m not sure if it’s too much…how can I check?
But the main problem is the opposite state that I have in 2 of my units as when it 100% open it says that it’s 100% close

markirb, 2024-09-16:
I really have no experience in Roller Shutter. But would this not simply be resolved by setting "Swap Ouputs"?

MacUser341, 2024-09-16:
> I really have no experience in Roller Shutter. But would this not simply be resolved by setting "Swap Ouputs"?

No, the swap options just changed the control for opening and close

markirb, 2024-09-16:
> > I really have no experience in Roller Shutter. But would this not simply be resolved by setting "Swap Ouputs"?
> 
> No, the swap options just changed the control for opening and close

This sounds like you hit "Swap Inputs"? I meant swap Ouputs?

gmuth, 2024-09-28:
I have the same issue. Something confuses the position saved. it would be handy to have a manual override option. e.g enter 100 when open and shelly thinks its closed.

closed 2024-10-02 by markirb