mongoose-os-apps/shelly-homekit #1511: Shelly Plus1PM not updating

DB2FM, 2024-09-09:
One of my Shelly Plus 1PM doesn't update. Neither to the newer 2.12.1 (it's on 2.11.2) nor to 2.12.2-beta3.
Is this a defective unit or can I do something else to update ist?

Log below:
2.11.2 -> 2.12.1
461435709 shelly_ota.cpp:90       Starting firmware update
461448963 esp32xx_ota_backend:209 App: Plus1PM.bin -> app_1 enc 0, FS: fs.img -> fs_1 enc 0
461460427 mgos_ota_core.c:685     0.15% total, bootloader.bin 512 of 23728
461467518 mgos_ota_core.c:1202    Update state: 0
461743719 mgos_ota_core.c:685     1.58% total, partition-table.bin 512 of 3072
461751218 mgos_ota_core.c:1202    Update state: 0
461778389 mgos_ota_core.c:685     1.77% total, otadata.bin 512 of 8192
461785849 mgos_ota_core.c:1202    Update state: 0
462010448 esp32xx_ota_backend:303 Writing FS @ 0x380000
462311756 mgos_ota_core.c:685     2.27% total, fs.img 512 of 458752
462318985 mgos_ota_core.c:1202    Update state: 0
462327476 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.22.44:64727
463193918 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.22.44:64727
464068484 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.22.44:64727
464415904 shelly_main.cpp:449     Up 464.41, HAP 0/0/16 ns 0, RAM: 204184/193040; st 48; disabled
465050799 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.22.44:64727
466357786 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.22.44:64727
466981663 mgos_ota_core.c:685     18.10% total, fs.img 262656 of 458752
466989891 mgos_ota_core.c:1202    Update state: 0
467033549 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.22.44:64727
468043783 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.22.44:64727
469043881 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.22.44:64727
469807560 esp32xx_ota_backend:399 fs_1 verified (429e0768a2f79d22bfe9fcbb5aaf057f1977d88b)
470233740 esp32xx_ota_backend:303 Writing app @ 0x200000
470242317 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.22.44:64727
470552849 mgos_ota_core.c:685     29.98% total, Plus1PM.bin 512 of 1159520
470560068 mgos_ota_core.c:1202    Update state: 0
470567962 mgos_mongoose.c:66      New heap free LWM: 191400
471039036 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.22.44:64727
472093963 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.22.44:64727
472628219 shelly_main.cpp:449     Up 472.62, HAP 0/0/16 ns 0, RAM: 201756/191400; st 49; disabled
473118926 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.22.44:64727
474057011 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.22.44:64727
474783011 mgos_ota_core.c:685     45.80% total, Plus1PM.bin 262656 of 1159520
474790204 mgos_ota_core.c:1202    Update state: 0
478743009 mgos_ota_core.c:685     61.63% total, Plus1PM.bin 524800 of 1159520
478750202 mgos_ota_core.c:1202    Update state: 0
480418119 shelly_main.cpp:449     Up 480.41, HAP 0/0/16 ns 0, RAM: 201756/191400; st 50; disabled
482903021 mgos_ota_core.c:685     77.46% total, Plus1PM.bin 786944 of 1159520
482910454 mgos_ota_core.c:1202    Update state: 0
486993113 mgos_ota_core.c:685     93.29% total, Plus1PM.bin 1049088 of 1159520
487000834 mgos_ota_core.c:1202    Update state: 0
488428368 shelly_main.cpp:449     Up 488.42, HAP 0/0/16 ns 0, RAM: 203188/191400; st 50; disabled
489086010 [0;32mI (489711) esp_image: segment 0: paddr=00200020 vaddr=3f400020 size=30be4h (199652) map[0m
489201272 [0;32mI (489821) esp_image: segment 1: paddr=00230c0c vaddr=3ffb0000 size=03630h ( 13872) [0m
489217935 [0;32mI (489841) esp_image: segment 2: paddr=00234244 vaddr=40080000 size=0bdd4h ( 48596) [0m
489253311 [0;32mI (489871) esp_image: segment 3: paddr=00240020 vaddr=400d0020 size=d2894h (862356) map[0m
489724989 [0;32mI (490351) esp_image: segment 4: paddr=003128bc vaddr=4008bdd4 size=08868h ( 34920) [0m
489753158 [0;32mI (490381) esp_image: segment 5: paddr=0031b12c vaddr=50000000 size=00010h (    16) [0m
489761535 esp32xx_ota_backend:399 app_1 verified (c6de8da6f71f75fdaaae51ed26c9f5ab133c3238)
489772354 mgos_ota_core.c:820     Reached the end of archive
489779905 esp32xx_ota_backend:463 Setting boot partition to app_1
489789115 [0;32mI (490411) esp_image: segment 0: paddr=00200020 vaddr=3f400020 size=30be4h (199652) map[0m
489908509 [0;32mI (490531) esp_image: segment 1: paddr=00230c0c vaddr=3ffb0000 size=03630h ( 13872) [0m
489925180 [0;32mI (490551) esp_image: segment 2: paddr=00234244 vaddr=40080000 size=0bdd4h ( 48596) [0m
489961572 [0;32mI (490581) esp_image: segment 3: paddr=00240020 vaddr=400d0020 size=d2894h (862356) map[0m
490434378 [0;32mI (491061) esp_image: segment 4: paddr=003128bc vaddr=4008bdd4 size=08868h ( 34920) [0m
490462193 [0;32mI (491081) esp_image: segment 5: paddr=0031b12c vaddr=50000000 size=00010h (    16) [0m
490471569 [0;32mI (491091) esp_ota_ops: New OTA data 0: seq 0x00000020, st 0x11, CRC 0xdfa70ee0[0m
490523821 mgos_ota_core.c:1051    Update finished, result 1 (Update applied, rebooting)
490529577 mgos_ota_core.c:1202    Update state: 0
490534445 mgos_ota_core.c:1096    Update requires reboot

2.11.2 -> 2.12.2-beta3
188581289 shelly_ota.cpp:90       Starting firmware update
188594549 esp32xx_ota_backend:209 App: Plus1PM.bin -> app_1 enc 0, FS: fs.img -> fs_1 enc 0
188606003 mgos_ota_core.c:685     0.15% total, bootloader.bin 512 of 23728
188613933 mgos_ota_core.c:1202    Update state: 0
188764796 mgos_ota_core.c:685     1.59% total, partition-table.bin 512 of 3072
188772871 mgos_ota_core.c:1202    Update state: 0
188805549 mgos_ota_core.c:685     1.77% total, otadata.bin 512 of 8192
188813529 mgos_ota_core.c:1202    Update state: 0
189047852 esp32xx_ota_backend:303 Writing FS @ 0x380000
189341462 mgos_ota_core.c:685     2.27% total, fs.img 512 of 458752
189349483 mgos_ota_core.c:1202    Update state: 0
192496064 shelly_main.cpp:449     Up 192.49, HAP 0/0/16 ns 0, RAM: 204708/193288; st 48; disabled
192871619 mgos_ota_core.c:685     18.10% total, fs.img 262656 of 458752
192879634 mgos_ota_core.c:1202    Update state: 0
195729818 esp32xx_ota_backend:399 fs_1 verified (55142c15b00a1f3f7ef99f6170ac3b46a032990e)
196148939 esp32xx_ota_backend:303 Writing app @ 0x200000
196452815 mgos_ota_core.c:685     29.98% total, Plus1PM.bin 512 of 1159536
196460819 mgos_ota_core.c:1202    Update state: 0
200413565 mgos_ota_core.c:685     45.80% total, Plus1PM.bin 262656 of 1159536
200421615 mgos_ota_core.c:1202    Update state: 0
200491337 shelly_main.cpp:449     Up 200.48, HAP 0/0/16 ns 0, RAM: 204728/193288; st 48; disabled
204262995 mgos_ota_core.c:685     61.63% total, Plus1PM.bin 524800 of 1159536
204271029 mgos_ota_core.c:1202    Update state: 0
208123300 mgos_ota_core.c:685     77.46% total, Plus1PM.bin 786944 of 1159536
208131329 mgos_ota_core.c:1202    Update state: 0
208496972 shelly_main.cpp:449     Up 208.49, HAP 0/0/16 ns 0, RAM: 204720/193288; st 49; disabled
212163086 mgos_ota_core.c:685     93.29% total, Plus1PM.bin 1049088 of 1159536
212171137 mgos_ota_core.c:1202    Update state: 0
214295926 [0;32mI (214921) esp_image: segment 0: paddr=00200020 vaddr=3f400020 size=30bf4h (199668) map[0m
214412543 [0;32mI (215031) esp_image: segment 1: paddr=00230c1c vaddr=3ffb0000 size=03630h ( 13872) [0m
214429194 [0;32mI (215051) esp_image: segment 2: paddr=00234254 vaddr=40080000 size=0bdc4h ( 48580) [0m
214464418 [0;32mI (215091) esp_image: segment 3: paddr=00240020 vaddr=400d0020 size=d2894h (862356) map[0m
214935062 [0;32mI (215561) esp_image: segment 4: paddr=003128bc vaddr=4008bdc4 size=08878h ( 34936) [0m
214962880 [0;32mI (215581) esp_image: segment 5: paddr=0031b13c vaddr=50000000 size=00010h (    16) [0m
214971263 esp32xx_ota_backend:399 app_1 verified (58dde0fcd81753004c9682baa8d7e71a9faf08ab)
214983118 mgos_ota_core.c:820     Reached the end of archive
214989912 esp32xx_ota_backend:463 Setting boot partition to app_1
214999121 [0;32mI (215621) esp_image: segment 0: paddr=00200020 vaddr=3f400020 size=30bf4h (199668) map[0m
215119657 [0;32mI (215741) esp_image: segment 1: paddr=00230c1c vaddr=3ffb0000 size=03630h ( 13872) [0m
215136322 [0;32mI (215761) esp_image: segment 2: paddr=00234254 vaddr=40080000 size=0bdc4h ( 48580) [0m
215172133 [0;32mI (215791) esp_image: segment 3: paddr=00240020 vaddr=400d0020 size=d2894h (862356) map[0m
215644729 [0;32mI (216271) esp_image: segment 4: paddr=003128bc vaddr=4008bdc4 size=08878h ( 34936) [0m
215672554 [0;32mI (216291) esp_image: segment 5: paddr=0031b13c vaddr=50000000 size=00010h (    16) [0m
215681928 [0;32mI (216301) esp_ota_ops: New OTA data 0: seq 0x00000022, st 0x11, CRC 0xe76203f5[0m
215734493 mgos_ota_core.c:1051    Update finished, result 1 (Update applied, rebooting)
215740952 mgos_ota_core.c:1202    Update state: 0
215745734 mgos_ota_core.c:1096    Update requires reboot

markirb, 2024-09-09:
Can you try to do two things:

Factory reset and try to downgrade to stock 1.3.3

I mean there are several logs here with this, where only some Shelly's are failing. Also no hints hin the logs. I do not know for sure as I do not have bootlogs but if stock downgrade after factory reset does not work I would say that the flash is defective.

markirb, 2024-09-10:
I will close this, lets talk in https://github.com/mongoose-os-apps/shelly-homekit/issues/1433

There seems to be something. but we do not know yet...

closed 2024-09-10 by markirb