mongoose-os-apps/shelly-homekit #1512: Ni Input Sensor in HomeKit with Mini gen 3

tharwan, 2024-09-10:
I successfully connected my Shelly 1 mini gen 3 to HomeKit with firmware 2.12.2-beta.3. But I do not get the input sensor when switching to „detached“ mode.

Is this already supported?

tharwan, 2024-09-10:
I just realized I am also not seeing the „input 1“ section in the UI.

markirb, 2024-09-10:
You're right it somehow doesn't work. Though from the code and config it should....

tharwan, 2024-09-11:
Anything I could do to help debugging?

markirb, 2024-09-11:
No.
I think I found it, will be fixed soon

markirb, 2024-09-12:
should be fixed in latest beta available in releases

djrams10, 2024-09-12:
> I successfully connected my Shelly 1 mini gen 3 to HomeKit with firmware 2.12.2-beta.3. But I do not get the input sensor when switching to „detached“ mode.
> 
> Is this already supported?

Thank you, can you explain to me how you did to install the beta? Thanks

tharwan, 2024-09-12:
Fixed! Thanks!

tharwan, 2024-09-12:
@djrams10 see #1509

closed 2024-09-12 by tharwan