mongoose-os-apps/shelly-homekit #1513: Shelly plus 1 pm restarting every 1-3 minutes

SebastianStolze, 2024-09-11:
Hey I have a problem with a shelly plus 1 pm, which was running for 172 days without a problem but now it reset every 1-3 minutes and therefore disabling the light. I first updated to the latest firmware version, without any improvement. Reverting to the stock firmware fixed the issue for now. Is there anything I can do to fix this?

I also have a second shelly installed which is still working.

markirb, 2024-09-11:
this sounds like a power problem. can you try to activate wifi power savings?

SebastianStolze, 2024-09-12:
Hi thanks for the ultra fast response. My Wifi Power Savings was already set to 1 for both shellys (the working one and the disconnecting one). Should i set it to 2 or disable?

markirb, 2024-09-12:
For Plus devices 2 is the maximum saving so try this

SebastianStolze, 2024-09-12:
It seems to be working after downgrading to 1.3.3 firmware after some 1.4.x stock firmware struggle. I managed to update to homekit back again and with the power save setting set to 2 it is working right now. Thanks for your work and for your help. I keep the ticket open till tomorrow and close it, when I can confirm it is working.

SebastianStolze, 2024-09-16:
It s working thanks

closed 2024-09-16 by SebastianStolze