mongoose-os-apps/shelly-homekit #1516: shelly 1 mini gen3

DaveRasp, 2024-09-12:
I have tried to download the HomeKit  firmware by both the ota general update link, and the python tool but both failed for my new shelly 1 mini gen3 . I have previously done this before for shelly 1 without issue.

The ota update link failed -404 according to the logs

The py tool saw my other shelly devices (shelly1)  but not this device at 1.3.3 and 1.4.2 firmware.

I see my device with the native shelly software at both AP and on normal network.

markirb, 2024-09-12:
Ota url does work since 15 minutes. just tested it. 
The py tool indeed I think does not work currently as it does not yet have the new model.

closed 2024-09-12 by markirb

simeonreusch, 2024-09-28:
OTA flashing also does not work for me yet (Mini1G3, FW 1.3.3). Flashing with the firmware from the latest release works.

markirb, 2024-09-29:
Can you make a request to following url:

http://A.B.C.D/ota/?url=http://flobby.net:8081/

then I can find out why this does not work

simeonreusch, 2024-09-29:
```
shelly_ota_source_ht:32 Update URL: http://flobby.net:8081/ (0x3fcc94a4) 14:06:54
shelly_http_client.:306 0x3fcc7ca4: HTTP GET http://flobby.net:8081/ 14:06:54
shelly_ota.cpp:238 Starting update (0x3fcc8498), to=600 cd=0/ct=0 ignore_same=0 14:06:54
shelly_http_server.:268 No handler for '/favicon.ico' 14:06:54
shelly_http_client.:606 0x3fcc7ca4: Finished; bytes 0, code 0, redir 0/3, auth 0, status -1: Connection error: -14 14:06:55
shelly_ota.cpp:245 Aborting update: premature end of data 14:06:55
shelly_ota.cpp:172 OTA status error, 0 % 14:06:55
shelly_notification:209 Event from sys: {"component":"sys", "event":"ota_error", "msg":"error", "ts":1727611615.80} 14:06:55
shelly_notification:116 ch type=WS_in info=172.16.100.83:60667 has no DST 14:06:55 
```

DaveRasp, 2024-09-29:
Just to advise from my original email that prompted this thread
I tried it again and it still did not work on my shelly 1 v3 after the - since 15 mins comment
My original shelly 1 v3 was working fine and then I upgraded it to the new shelly software before the HomeKit upgrade. Unfortunately after the upgrade it kept disconnecting from wifi after 30 secs or so then reconnected, I assumed that this was the cause of the error downloading the HomeKit firmware and therefore sent the component back.  I did not buy another one so did not retry it.  Hope this helps but may not be relevant.