mongoose-os-apps/shelly-homekit #1517: Shelly Mini 1 Gen 3 switch not working

wlhrtr, 2024-09-13:
Just flashed from 1.3.3 to 2.12.2.
With 1.3.3 switch in wall and on web-dashboard worked.
With 2.12.2 neither the wall switch not the dashboard switch works.

djrams10, 2024-09-13:
> Just flashed from 1.3.3 to 2.12.2.
> 
> With 1.3.3 switch in wall and on web-dashboard worked.
> 
> With 2.12.2 neither the wall switch not the dashboard switch works. 

Same for me

markirb, 2024-09-14:
Hi,

There seems to be something for some models so far I had only one report of this, for most it seems to work. See https://github.com/mongoose-os-apps/shelly-homekit/discussions/1499#discussioncomment-10590621

Can you please add some more details:
- the log that captures a switch event.
- does it not turn off or on? The report I had was stuck at on
- If you have: system info on stock firmware andd/or picture of the device that shows the exact model and version

djrams10, 2024-09-14:
> Hi,
>
> There seems to be something for some models so far I had only one report of this, for most it seems to work. See [...]

Hi, here are the screenshots with the different information.

The flash is done correctly, except that at the level of ignition, extinction, it does not work.

The light remains fixed red, and when the button is turned on, we do not hear the click. Another thing also I put a switch on the Shelly and it no longer works either.

When I come back to the original firmware everything works normally.

Thanks

[image]
[image]
[image]
[image]
[image]

wlhrtr, 2024-09-14:
used for both the wall switch and dashboard toggle

for 1.4.2 (downgraded befor homekit flash)

```
Connected.
09:46:19
shelly_debug.cpp:236 Streaming logs to 192.168.31.164:64488
09:46:19
shos_dns_sd_respond:236 ws(0x3fcbc474): Announced Shelly1MiniG3-DCDA0CE356B4 any@any (192.168.31.160)
09:46:22
shos_dns_sd_respond:236 ws(0x3fcbc474): Announced Shelly1MiniG3-DCDA0CE356B4 any@any (fe80::deda:cff:fee3:56b4,2001:4bc9:1fb8:fe8a:deda:cff:fee3:56b4)
09:46:22
shos_rpc_inst.c:242 switch.toggle via WS_in 192.168.31.164:64483
09:46:26
shelly_notification:162 Status change of switch:0: {"id":0,"output":true,"source":"WS_in"}
09:46:26
shos_rpc_inst.c:242 switch.toggle via WS_in 192.168.31.164:64483
09:46:27
shelly_notification:162 Status change of switch:0: {"id":0,"output":false,"source":"WS_in"}
09:46:27
shelly_notification:162 Status change of input:0: {"id":0,"state":true}
09:46:36
shelly_notification:162 Status change of switch:0: {"id":0,"output":true,"source":"switch"}
09:46:36
shelly_notification:162 Status change of input:0: {"id":0,"state":false}
09:46:36
shelly_notification:162 Status change of switch:0: {"id":0,"output":false,"source":"switch"}
09:46:36
```

for 2.12.2
```
235916722 shelly_debug.cpp:231    No log file, sending new entries
235923761 mgos_http_server.c:180  0x3fcaf6b8 HTTP connection from 192.168.31.164:65025
240908071 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
241766239 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
242354149 shelly_main.cpp:483     Up 242.34, HAP 0/1/16 ns 1, RAM: 203324/188456; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0
242638315 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
243636292 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
244642466 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
245638937 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
246640913 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
247637208 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
248635733 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
249635845 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
250354025 shelly_main.cpp:483     Up 250.34, HAP 0/1/16 ns 1, RAM: 203324/188456; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0
258354021 shelly_main.cpp:483     Up 258.34, HAP 0/1/16 ns 1, RAM: 203324/188456; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0
266354070 shelly_main.cpp:483     Up 266.34, HAP 0/1/16 ns 1, RAM: 203324/188456; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0
269989473 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
270641902 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
271637735 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
272643096 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
274354077 shelly_main.cpp:483     Up 274.34, HAP 0/1/16 ns 1, RAM: 203324/188456; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0
279836843 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
280641561 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
281639796 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
282354049 shelly_main.cpp:483     Up 282.34, HAP 0/1/16 ns 1, RAM: 203324/188456; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0
282383710 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.31.164:64919
282388981 shelly_output.cpp:67    Output 1: off -> on (RPC)
282547328 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
283571162 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.31.164:64919
283576115 shelly_output.cpp:67    Output 1: on -> off (RPC)
283768658 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
285808829 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
286555677 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.31.164:64919
286560627 shelly_output.cpp:67    Output 1: off -> on (RPC)
286585271 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
286638391 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
287638704 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
[... 61 lines cut ...]
335200528 shelly_output.cpp:67    Output 1: on -> off (RPC)
335220206 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
335639330 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
338439023 mgos_sys_config.c:323   Saved to conf9.json
338449711 shelly_main.cpp:483     Up 338.44, HAP 0/1/16 ns 1, RAM: 203120/188456; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0
346354021 shelly_main.cpp:483     Up 346.34, HAP 0/1/16 ns 1, RAM: 203324/188456; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0
349287191 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
349643004 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
350641275 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
354354041 shelly_main.cpp:483     Up 354.34, HAP 0/1/16 ns 1, RAM: 203324/188456; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0
362354030 shelly_main.cpp:483     Up 362.34, HAP 0/1/16 ns 1, RAM: 201760/188456; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0
[... cut ...]
434354160 shelly_main.cpp:483     Up 434.34, HAP 0/1/16 ns 1, RAM: 201760/188456; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0
442354100 shelly_main.cpp:483     Up 442.34, HAP 0/1/16 ns 1, RAM: 203324/188456; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0
450354099 shelly_main.cpp:483     Up 450.34, HAP 0/1/16 ns 1, RAM: 203324/188456; st 41; 0.1: st:0 in_st:0 inm:1 ininv:0
451665184 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
452716087 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.31.164:64919
```

markirb, 2024-09-14:
Ok, I can see nothing from that so far can you please find out what's printed on your casing? 
Mine says 

S3SW-001X8EU, Ver1, B0124

To which voltage do you have it wired? AC / DC?

djrams10, 2024-09-14:
> Ok, I can see nothing from that so far can you please find out what's printed on your casing? 
> 
> Mine says 
> 
> 
> 
> S3SW-001X8EU, Ver1, B0124
> 
> 
> 
> To which voltage do you have it wired? AC / DC?
> 
> 

I bought it and nothing touches on the ac / dc

[image]
[image]

wlhrtr, 2024-09-14:
@markirb 

Wired like in this diagram:
[image]

Printed on case:
S3SW-001X8EU
Ver1 B2724

djrams10, 2024-09-14:
For my part, the wiring is necessarily good because if I go back to the original firmware it works perfectly, and I had a Shelly 1 flasher before and everything worked well.

djrams10, 2024-09-14:
> Just flashed from 1.3.3 to 2.12.2.
> With 1.3.3 switch in wall and on web-dashboard worked.
> With 2.12.2 neither the wall switch nor the dashboard switch works.

Exactly The Same

markirb, 2024-09-14:
ok... nothing really different on the models.

can you try if it works with this firmware?
(this sets GPIO7 driver strength to maximum)

[fw.zip](https://github.com/user-attachments/files/17002679/fw.zip)

wlhrtr, 2024-09-14:
> can you try if it works with this firmware? (this sets GPIO7 driver strength to maximum)
> 
> [fw.zip](https://github.com/user-attachments/files/17002679/fw.zip)

unfortunately not working on my end

djrams10, 2024-09-14:
> ok... nothing really different on the models.
> 
> 
> 
> can you try if it works with this firmware?
> 
> (this sets GPIO7 driver strength to maximum)
> 
> 
> 
> [fw.zip](https://github.com/user-attachments/files/17002679/fw.zip)
> 
> 

It’s always the same problem unfortunately

markirb, 2024-09-16:
I still am trying to get my head around out what can go wrong here in Hardware or in Software.
So far I have run out of ideas to try unfortunately.

wlhrtr, 2024-09-16:
It looks like an issue with gpio pin in the firmware.
Just flashed back to stock firmware and light went on after updating without me doing anything.

djrams10, 2024-09-16:
I don’t know what else to do to help you. Do not hesitate if you need.

Dag0d, 2024-09-16:
Same for me with a shelly 1 pm gen 3. original firmware works fine. HomeKit firmware keeps the last state of the original firmware. relay is not clicking, nothing. 
no difference if using the website or home to control.

markirb, 2024-09-16:
[shelly-homekit-ShellyMini1Gen3.zip](https://github.com/user-attachments/files/17019170/shelly-homekit-ShellyMini1Gen3.zip)

can you try this and see if it changes anything?

markirb, 2024-09-16:
> It looks like an issue with gpio pin in the firmware. Just flashed back to stock firmware and light went on after updating without me doing anything.

yes. but the gpio number is correct. it wound not work on other devices like mine or tasiota if it wasn't... 
it can only be settings thereof; but nothing the other firmwares seem to use. or it could be mongoose os esp32c3 specific. but I looked through this code.

Dag0d, 2024-09-16:
> [shelly-homekit-ShellyMini1Gen3.zip](https://github.com/user-attachments/files/17019170/shelly-homekit-ShellyMini1Gen3.zip)
> 
> can you try this and see if it changes anything?

If you could do that for the mini 1 pm gen 3 as well I would give it a try.

wlhrtr, 2024-09-16:
> [shelly-homekit-ShellyMini1Gen3.zip](https://github.com/user-attachments/files/17019170/shelly-homekit-ShellyMini1Gen3.zip)
> 
> can you try this and see if it changes anything?

unfortunately still not working

but thank you for your continuous efforts

djrams10, 2024-09-17:
> can you try this and see if it changes anything?

For me I have no change. I would like to point out that I am in France. Thank you for your efforts.

danny1605, 2024-09-19:
I have the same problem

markirb, 2024-09-20:
I think I found it.

First: it will work after you make your shelly hard reboot (cut power)

This only happens when you directly upgrade from stock. Stock seems to call gpio_hold_en before reboot. So we can unlock that after booting.

Just make a hard power cut or wait for the fix which will be coming soon.

djrams10, 2024-09-20:
> I think I found it.
>
> First: it will work after you make your shelly hard reboot (cut power)
>
> This only happens when you directly upgrade from stock. Stock seems to call gpio_hold_en before reboot. [...]

It’s okay, it works perfectly for me. Thank you again for your help.

markirb, 2024-09-20:
If you do not want to cut power you can try this:

[shelly-homekit-ShellyMini1Gen3.zip](https://github.com/user-attachments/files/17079273/shelly-homekit-ShellyMini1Gen3.zip)

wlhrtr, 2024-09-20:
> If you do not want to cut power you can try this:
> 
> [shelly-homekit-ShellyMini1Gen3.zip](https://github.com/user-attachments/files/17079273/shelly-homekit-ShellyMini1Gen3.zip)

just updated from 1.3.3 and it works flawlessly now 

@markirb thank you very much :)

closed 2024-10-03 by markirb