mongoose-os-apps/shelly-homekit #1518: 2.12.2: Shelly Plus RGBW PM: Configuration is not retained

mirtxa, 2024-09-17:
Device: Shelly Plus RGBW PM
Version: 2.12.2

Hi. 

When cycling power the device has lost its configuration and starts to AP mode again. The module loses all its settings and looks like it resets.

markirb, 2024-09-17:
This looks like we still got the reset pin/polarity wrong. Unfortunately I do not have one myself to verify this but I thought it was verified so far, see discussion thread.

You could try and see if it still happens when you push the button during boot, then we have wrong polarity.

altercation, 2024-10-12:
Can confirm this behavior/issue on Shelly Plus RGBW PM:

1) Starting at OEM Firmware 1.3.3 -> Install shelly-homekit firmware
2) WITHOUT power cycling, homekit firmware comes up on same IP and can be configured
3) POWER CYCLING results in reversion to AP setup mode

markirb, 2024-10-12:
Thanks for reporting. Other sites have the PIN number 22 for reset button as well, so if this pin causes this behavior then:

Could you please try to push the button on the device while rebooting and see if that changes behavior?

altercation, 2024-10-12:
> Could you please try to push the button on the device while rebooting and see if that changes behavior?

Ok, following steps taken:

1) Accessed shelly-homekit firmware in AP mode (after last reboot failed to save)
2) Changed settings, connected successfully to wifi, changed name (for reference test)
3) Powered off device (via power strip so easy to perform next step)
4) While depressing reset button on back of unit, powered on, releasing button after 1 second
5) Unit again moved to AP mode.

I should note that when the unit has booted back into AP mode (i.e. settings failed to save) the AP was not named using the shelly-homekit “short” AP name, but rather the long AP name (as with the stock shelly firmware). Connecting to it was unsuccessful until I did a full reset (holding down the reset button for > 10 seconds). After that >10sec reset, the unit reboots with the short AP name and I can connect, but it is of course wiped at that stage.

Happy to do more tests and also happy to send a unit to you at my expense in support of OSS :)

markirb, 2024-10-16:
Thanks for trying. The AP name hints at the factory reset behavior. The different names are a small difference of stock and this os.

Strange. If you want you can just donate to the PayPal link then I will get a RGBWPM and fix this myself in the next days.

markirb, 2024-10-20:
I change the reset polarity in the latest release. Please try and see if this fixes the bug

closed 2024-11-15 by markirb