mongoose-os-apps/shelly-homekit #1523: Aborting update: bad arch: esp8266

fiedl, 2024-09-28:
I'm trying to flash a shelly plug s, which fails with "Aborting update: bad arch: esp8266".

### Device

Shelly PlusPlugS 

### Original Firmware

1.3.3

### Firmware Url

http://rojer.me/files/shelly/shelly-homekit-ShellyPlugS.zip

### Diagnostics

```
shelly_rpc_service.:293 Updating from http://rojer.me/files/shelly/shelly-homekit-ShellyPlugS.zip
20:41:51
shelly_ota_source_ht:32 Update URL: http://rojer.me/files/shelly/shelly-homekit-ShellyPlugS.zip (0x3ffe2960)
20:41:51
shelly_http_client.:306 0x3ffe2180: HTTP GET http://rojer.me/files/shelly/shelly-homekit-ShellyPlugS.zip
20:41:51
shelly_ota.cpp:238 Starting update (0x3ffe7484), to=600 cd=0/ct=0 ignore_same=0
20:41:51
shelly_ota.cpp:356 shelly-plug-s-2.12.2/manifest.json: 1511 bytes (hdr 64 desc 0 CRC 4a11df15)
20:41:51
shelly_ota.cpp:245 Aborting update: bad arch: esp8266
20:41:51
shelly_ota.cpp:172 OTA status error, 0 %
20:41:51
shelly_notification:209 Event from sys: {"component":"sys", "event":"ota_error", "msg":"error", "ts":1727548911.64}
20:41:51
shelly_http_client.:606 0x3ffe2180: Finished; bytes 5648, code 200, redir 0/3, auth 0, status CANCELLED: Cancelled
```

markirb, 2024-09-28:
You are using non plus zip for plus version. Does the auto link not work?

If not, could you help me to adjust the correct mapping?

fiedl, 2024-09-28:
> You are using non plus zip for plus version

Yes, I did, sorry! 

> Does the auto link not work?

No, it returned a 404:

```
shelly_rpc_service.:293 Updating from http://shelly.rojer.cloud/update
22:49:42
shelly_ota_source_ht:32 Update URL: http://shelly.rojer.cloud/update (0x3ffdd6a8)
22:49:42
shelly_http_client.:306 0x3ffe462c: HTTP GET http://shelly.rojer.cloud/update
22:49:42
shelly_ota.cpp:238 Starting update (0x3ffe4d00), to=600 cd=0/ct=0 ignore_same=0
22:49:42
shelly_ota_source_ht:73 0x3ffdd6a8 Error fetching update: ABORTED: Bad HTTP response: 404
22:49:42
shelly_ota.cpp:245 Aborting update: source error: -110: Bad HTTP response: 404
22:49:42
shelly_http_client.:606 0x3ffe462c: Finished; bytes 217, code 404, redir 0/3, auth 0, status ABORTED: Bad HTTP response: 404
22:49:42
shelly_ota.cpp:172 OTA status error, 0 %
22:49:42
shelly_notification:209 Event from sys: {"component":"sys", "event":"ota_error", "msg":"error", "ts":1727556582.63}
```

closed 2024-09-28 by fiedl

markirb, 2024-09-29:
Can you call the following endpoint with your shelly http://A.B.C.D/ota/?url=http://flobby.net:8081/

Then I can find out the right setting and fix it