mongoose-os-apps/shelly-homekit #1524: Shelly Plug S Firmware: 2.12.2

KumanduS, 2024-10-01:
I have a Shelly Plug S running with firmware 2.12.2. I have an automation that turn off the Plug however the devices connected to the Plug are still getting energy. Using the webapplication of the device i get the same result. Toggling the Status does not cut the energy of the plug. Can you please help me solving this problem?

[image]

markirb, 2024-10-02:
Can you try reverting to stock firmware and see if it is the same?

KumanduS, 2024-10-02:
I never try to use the original firmware. I will try do revert the firmware during the weekend and i will report back. Thanks

KumanduS, 2024-10-10:
i reverted to stock firmware, and the problem keeps appearing. So this is not a problem of the firmware, but from the unit itself. I will try to open a ticken in shelly. Thanks for the support, you can close the thread.

closed 2024-10-10 by markirb