mongoose-os-apps/shelly-homekit #1525: Shelly SP1 with Addon and Garage opening sensor showing "blocked" when closing

OTWfuckU, 2024-10-03:
Hi there,

my newly installed SP1 with your fine HomeKit OS is reacting to Siri perfectly. However the issue is, that whenever closing the garage door, I do not get a push stating "closed door". HomeKit is further showing "blocked", although the installed sensor is working fine.

I am guessing there is a fault in the programming.

Would be cool, if you can check it and hopefully remedy accordingly.

Thank you guys.

markirb, 2024-10-03:
Which sensor did you attach where? Closing sensor has to be on SW Input. Opening sensor can optionally be on addon.

OTWfuckU, 2024-10-04:
That is the point. Solely one open sensor on Addon. It works like this with Homeassistant and Shelly Cloud app. But not with the HomeKit.

markirb, 2024-10-04:
Is there a reason why you cannot put it on SW?

See 
https://github.com/mongoose-os-apps/shelly-homekit/issues/1397 for details

markirb, 2024-10-05:
> Yes, I am running (have to) the SP1 with 230 V. This is where you do not attach a magmatic sensor.
> 

Ok that makes sense

> In addition, the config with two sensor does not make sense … wrong logic. It should be solely one close sensor (via addon or otherwise). And once the circuit is not closed anymore, it should show „open“.
> 

I beg to differ. It does make sense, as "open" is not the same as "not closed"

> This is how it works under Shelly OS and the way in makes sense per logic. If you press a button to the middle … the light is still off until the circuit is closed … so pushed on. Same logic to the one sensor solution.

I made a version that can swap the inputs. It does not yet show in Webapp but can be set with: 

[shelly-homekit-ShellyPlus1.zip](https://github.com/user-attachments/files/17265498/shelly-homekit-ShellyPlus1.zip)

curl http://A.B.C.D/rpc/Config.Set -d '{"config": {"gdo": {"sensor_swap": true}}}'

curl http://A.B.C.D/rpc/Config.Save

Can you try?

OTWfuckU, 2024-10-05:
Thanks man, I will try and report back in.

OTW

OTWfuckU, 2024-10-05:
is that the new version?

20240912-182009/2.12.2-gfff6707

markirb, 2024-10-05:
should be 2.12.2-2-g5deadc0-gdo_swapsensors or I made some mistake and it reverts...

OTWfuckU, 2024-10-05:
Most likely the upload did not work. This is what happened, when I uploaded the ZIP file via the Webapp.

Firmware
Version:2.12.2
Build:
20240912-182009/2.12.2-gfff6707
Update:  -1 Archived file size is unknown

markirb, 2024-10-05:
What file do you upload? Mac unzips it if you just click the link. You have to do right click save link as

OTWfuckU, 2024-10-05:
I manage to upload and engaged the curl commands on terminal. HomeKit is still showing "blocked" upon closure => sensor contact.

markirb, 2024-10-05:
After saving you might have to restart the device

OTWfuckU, 2024-10-05:
Sorry to say … same effect.

markirb, 2024-10-05:
Oh I think I made a small mistake. I swapped outputs not inputs. I will push another release, might need until tomorrow sorry

OTWfuckU, 2024-10-05:
No problem … even the best make mistakes … will wait until you fixed it. Thanks for your help and digging into this. Much appreciated.

markirb, 2024-10-06:
Can you try this:

[shelly-homekit-ShellyPlus1.zip](https://github.com/user-attachments/files/17268338/shelly-homekit-ShellyPlus1.zip)

OTWfuckU, 2024-10-06:
good morning,

now it is showing "opening" on HomeKit when pushing switch to "open". Once pushing "close" it says "closing", then jumping to "open" until the sensor circuit is closed, then swapping to "closing" and finally (3 sec after close) to "blocked".

OTWfuckU, 2024-10-06:
In addition, the HomeKit pushes do not work anymore

markirb, 2024-10-06:
The opening Sensor can already be disabled in webif. This should be all you need?

OTWfuckU, 2024-10-06:
Lieber Markus,

ich bin doch ein DaU ("Dümmster anzunehmender User") ... daher auch mein auf mich fluchender Name. Ich kann nur sagen, dass ich einen "Close" Sensor am Addon habe und HomeKit nicht "geschlossen", sondern "Blockiert" und in der Webapp "stopped" anzeigt. Es wäre sinnvoll die Webapp so umzugestalten, dass diese der Shelly App Logik folgt. In der Shelly App als auch im HomeAssistant funktioniert die Anzeige wie gewünscht (habe einen zweiten SP1 für HA verbaut). 

Gerne können wir das auch offline mal besprechen.

Viele liebe Grüße! nach Niedersachen aus Karlsruhe.

OTW

markirb, 2024-10-06:
Genau. Daher den Open Sensor auf disabled stellen, da der sonst was falsches macht. Den closing sensor gibts ja. 

Den anderen richtig einstellen (normally Open/closed je nach relaisart). Timeout auch richtig setzen. 

Und dann sehe ich momentan keinen Grund warum das nicht gehen sollte.

Sonst bitte einmal den Screenshot von den Einstellungen machen.

OTWfuckU, 2024-10-06:
Servus Markus,

hab gerade meinen Nachbarn mal einbezogen ... der kennt sich aus. Hast mir leider den falschen CURL Befehl geschickt. Es ist "gdo1" und nicht "gdo". Jetzt funktioniert der geswapte Zustand. Vielen herzlichen Dank. Lass wissen, wenn ihr den Sensor Swap in die WebApp als Option aufgenommen habt.

Viele liebe Grüße!

OTW

markirb, 2024-10-07:
Ahja. Das hatte ich eigentlich noch 1 Minute nach dem Schreiben korrigiert. Steht auch richtig da

closed 2024-10-07 by markirb

OTWfuckU, 2024-10-07:
🤣 Klassiker … jetzt auch gesehen. Da war ich wohl schneller als sonst. Fällt einem nicht auf, wenn man nicht detailbesessen ist.