mongoose-os-apps/shelly-homekit #1528: OTA flashing of Shelly Mini1G3 not working (FW 1.3.3)

bassplayer404, 2024-10-08:
I have several older Shellys and they are all fine but flashing this Gen3 (FW 1.3.3) does not work. I used this link with the right ip address:
http://192.168.0.117/ota?url=http://shelly.rojer.cloud/update
But that does not work...

markirb, 2024-10-08:
Can you try again? I tried to fix the mapping

bassplayer404, 2024-10-08:
I tried but it doesn´t work...

markirb, 2024-10-08:
Mhh they have quite some naming confusion with the latest ones.

If you want to help you can fire up a local nectat on your machine with

nc -l -p 8080

Afterwards you use the machine's Ip as endpoint.

http://a.b.c.d/ota?http://(machineip):8080

If you then paste the output of nc then I can fix it.

bassplayer404, 2024-10-08:
I would help, but don´t know what a "nc" is and where to paste the code...

depinette, 2024-10-08:
GET / HTTP/1.1
Host: 10.0.0.17:8080
Content-Length: 0
Connection: close
User-Agent: Mini1G3/1.1.99-minig3prod1 (ShellyOS)
X-MGOS-Device-ID: shelly1minig3-84fce636e9ac 84FCE636E9AC
X-MGOS-FW-Version: esp32c3 1.1.99-minig3prod1 20231121-110944/1.1.99-minig3prod1-ga898543

markirb, 2024-10-09:
Thanks. Should work now

bassplayer404, 2024-10-09:
Yes it works! Thank you so much!

closed 2024-10-09 by bassplayer404