mongoose-os-apps/shelly-homekit #1533: Shelly HomeKit 2.12.2 AddOn currently limited to "three" (3) addons like DS18B20 . .

PaulmichlJ, 2024-10-11:
As requested by developer @markirb I open an new issue regarding this limitation.

My Shelly 1+ wasn't able to flash the homekit firmware!
-> I removed the addon an bum - the homekit update runes fine - BUT If I add the addon again - the device isn't able to boot anymore. removed the addon again and the device reboots. revered back to 1.3.3 the addon is working fine again.

FINAL SOLUTION: removed on DS18B20 and tried everything again with just three of them - all fine. therefore I will use two shellys to cover the need of five measure points . . . .

@markirb commented: "_Yes currently the limit is three. As this used to be the old limit. I never got around to test more, but the limit is rather artificial at least from software perspective. For hardware (addon differences) there can be one. Can you open a new issue from this please? It is not hard to remove the limit. Just wondering why it would crash..._"

I tested his updated version which worked - THANK's for the quick help!

markirb, 2024-10-11:
Thanks. Will be fixed from 2.13.0-beta2 onwards (expected in the next days)

boarochristian, 2024-10-13:
> As requested by developer @markirb I open an new issue regarding this limitation.
>
> My Shelly 1+ wasn't able to flash the homekit firmware! -> I removed the addon an bum - the homekit update runes [...]

Hi, I tried connecting the addon to my Shelly Plus 1, but I don't see any options in the device settings. Can you give me some help? on the input item how did you configure it?

PaulmichlJ, 2024-10-15:
One more question regarding the AddOn and the DS18B20: _Would it be possible to add the option to define an "Temperatur Addon Offset" as it was in the original stock firmware?_

markirb, 2024-10-16:
Will be in next beta/release. I'm closing this here for now.

closed 2024-10-16 by markirb

PaulmichlJ, 2024-10-20:
> Will be in next beta/release. I'm closing this here for now.

THANK you so much for adding that so quickly - tested it today. Well done and donated immediately ;-)

PaulmichlJ, 2024-10-20:
> Will be in next beta/release. I'm closing this here for now.

THANK you so much for adding that so quickly - tested it today. Well done and donated immediatelly ;-)