mongoose-os-apps/shelly-homekit #1535: Shelly 1PM: Bridge or switch vanishes in Homekit. Bridge device becomes switch

WRTHacker, 2024-10-13:
After upgrading to 18.0(.1) but it even occurs unter 17.X, even with 2.11 and 2.12 SHelly firmware:

I have issued with my 1PM and Plus1PM: The 1PM was not reachable for weeks. Than suddenly the switch disappered in Homekit, and the switch function „moved“ to the bridge. I have to remove the device an readd it. After this all is ok again, I recover my home setup than with Controller for Home Backups. 

I can't identify an trigger but suggest it is triggered by an switch of the active home hub. I have twoe Apple TV and 5 HomePod mini.

markirb, 2024-10-13:
It is this I think https://github.com/mongoose-os-apps/shelly-homekit/issues/1417

closed 2024-10-13 by markirb

WRTHacker, 2024-10-13:
Hmm, I'm not shure. A second input is not shown. 

When I add the 1PM or Plus 1PM, a bridge and a switch/light is addes. This works for days/weeks. But then one of the devices is shown as "do not answer" in Homekit. After this the describes effects occur.