mongoose-os-apps/shelly-homekit #1536: Shelly Mini1PMG3

SchmiddyTV, 2024-10-13:
Hi, I am trying to upgrade my Shelly Mini1 PM Gen 3 with the firmware for Homekit. According to the start page it is compatible with it, the link also takes me directly to the same article.

Now I'm trying to upgrade the firmware via the URL “http://A.B.C.D/ota?url=http://shelly.rojer.cloud/update”. My browser displays the following text:

“Trying to update from http://shelly.rojer.cloud/update”

After a while I refresh the page and the control panel looks the same as always. I also can't find any settings for HomeKit. 

I have already tried resetting to version 1.3.3 and using the latest version. Unfortunately this does not work.

What am I doing wrong? Does anyone have any ideas?

markirb, 2024-10-13:
Can you try again, maybe mapping was wrong? Did you try the manual link / file?

SchmiddyTV, 2024-10-13:
I tried again with the default URL in the browser, when I reload the page it now tells me that the update is in progress. When it is finished, the page automatically reloads and the control panel looks the same as always and there are no HomeKit settings to be found.

Unfortunately I can't find a file for the 1PM Mini G3. Could you send me the link for the file?

[image]

markirb, 2024-10-14:
I do not know if 1.4.2 update works for Gen3. You can try the latest beta, this should work. If not then lets see further

https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.13.0-beta2

closed 2024-10-15 by markirb