mongoose-os-apps/shelly-homekit #1546: Window motor and Shelly PM2 with custom firmware problem

Nastras, 2024-10-26:
Hello, I have the following problem with my Shelly 2PM and the Homekit firmware. 

I have connected one to an electric window motor in my garage and set it up as a shutter.
If I now want to calibrate the motor, the window only opens about 5cm and closes 5cm. In other words, the motor only moves a very short distance of 30cm. Manually button, the window only moves this short distance after calibration.

So far I have firmly assumed that it is a hardware defect and not due to your firmware. I therefore replaced the window motor and the Shelly. Unfortunately, the behavior has remained despite the new hardware.

After I ruled out the defective hardware, I tried it with the Shelly firmware 1.4.4 and found that the window can be calibrated with this firmware without any problems and opens and closes. I have installed your firmware 1.13.0 and tested the previous version. The problem persists with both. 

Do you have any idea why your firmware reacts to the window motor in this way? I still have 4 window shutters with your firmware and PM2 that work great.

Thanks and best Regards!

 
[image]
[image]

Nastras, 2024-10-26:
[image]
[image]

markirb, 2024-10-26:
No way to know without any logs of what's happening

Nastras, 2024-10-26:
Sorry I forgot to create these. Here are the logs.

Calibration

5956856683 shelly_debug.cpp:231    No log file, sending new entries
5960592271 shelly_main.cpp:492     Up 5960.58, HAP 0/1/16 ns 1, RAM: 209292/180500; st 36; 4.1: c:1 mp:3.56 ip:5.00 mt_ms:5090 cp:0.00 tp:0.00 md:0 lmd:2
5968591984 shelly_main.cpp:492     Up 5968.58, HAP 0/1/16 ns 1, RAM: 209312/180500; st 35; 4.1: c:1 mp:3.56 ip:5.00 mt_ms:5090 cp:0.00 tp:0.00 md:0 lmd:2
5976592012 shelly_main.cpp:492     Up 5976.58, HAP 0/1/16 ns 1, RAM: 209312/180500; st 36; 4.1: c:1 mp:3.56 ip:5.00 mt_ms:5090 cp:0.00 tp:0.00 md:0 lmd:2
5984591993 shelly_main.cpp:492     Up 5984.58, HAP 0/1/16 ns 1, RAM: 209312/180500; st 35; 4.1: c:1 mp:3.56 ip:5.00 mt_ms:5090 cp:0.00 tp:0.00 md:0 lmd:2
5992592037 shelly_main.cpp:492     Up 5992.58, HAP 0/1/16 ns 1, RAM: 209312/180500; st 35; 4.1: c:1 mp:3.56 ip:5.00 mt_ms:5090 cp:0.00 tp:0.00 md:0 lmd:2
6000592004 shelly_main.cpp:492     Up 6000.58, HAP 0/1/16 ns 1, RAM: 209292/180500; st 35; 4.1: c:1 mp:3.56 ip:5.00 mt_ms:5090 cp:0.00 tp:0.00 md:0 lmd:2
6008591989 shelly_main.cpp:492     Up 6008.58, HAP 0/1/16 ns 1, RAM: 209328/180500; st 35; 4.1: c:1 mp:3.56 ip:5.00 mt_ms:5090 cp:0.00 tp:0.00 md:0 lmd:2
6011905658 mgos_http_server.c:180  0x3ffce714 HTTP connection from 192.168.178.25:57514
6012030427 mgos_http_server.c:180  0x3ffce9ec HTTP connection from 192.168.178.25:57515
6012049759 mg_rpc.c:315            Shelly.GetInfo via WS_in 192.168.178.25:57515
6012065644 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57515
6013123708 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57515
6014147945 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57515
6014353094 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.178.25:57515
6014392785 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57515
6014434938 shelly_hap_window_c:330 WC 1: State: idle -> precal0 (0 -> 10)
6014532887 shelly_hap_window_c:450 Begin calibration
6014773636 mgos_sys_config.c:323   Saved to conf9.json
6014779306 shelly_output.cpp:67    Output 1: off -> on (precal0)
6014785209 shelly_hap_window_c:330 WC 1: State: precal0 -> cal0 (10 -> 11)
6015172331 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57515
6015195090 shelly_hap_window_c:466 WC 1: P0 = 3.433
6015994986 shelly_hap_window_c:466 WC 1: P0 = 1.091
6016013208 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57515
6016592052 shelly_main.cpp:492     Up 6016.58, HAP 0/1/16 ns 1, RAM: 208468/180500; st 35; 4.1: c:0 mp:3.56 ip:5.00 mt_ms:5090 cp:0.00 tp:0.00 md:0 lmd:2
6016794942 shelly_hap_window_c:466 WC 1: P0 = 1.799
6017595069 shelly_hap_window_c:466 WC 1: P0 = 2.848
6018394952 shelly_hap_window_c:466 WC 1: P0 = 3.848
6019195096 shelly_hap_window_c:466 WC 1: P0 = 4.104
6019795711 shelly_output.cpp:67    Output 1: on -> off (cal0)
6019801762 shelly_hap_window_c:330 WC 1: State: cal0 -> postcal0 (11 -> 12)
6019896249 shelly_hap_window_c:330 WC 1: State: postcal0 -> precal1 (12 -> 13)
6019995003 shelly_output.cpp:67    Output 2: off -> on (precal1)
6020000880 shelly_hap_window_c:330 WC 1: State: precal1 -> cal1 (13 -> 14)
6020595293 shelly_hap_window_c:497 WC 1: P1 = 4.665
6021395098 shelly_hap_window_c:497 WC 1: P1 = 2.043
6022194984 shelly_hap_window_c:497 WC 1: P1 = 3.372
6022995066 shelly_hap_window_c:497 WC 1: P1 = 4.183
6023794911 shelly_hap_window_c:497 WC 1: P1 = 4.957
6024592211 shelly_main.cpp:492     Up 6024.58, HAP 0/1/16 ns 1, RAM: 208660/180500; st 36; 4.1: c:0 mp:3.56 ip:5.00 mt_ms:5090 cp:0.00 tp:0.00 md:0 lmd:2
6024600319 shelly_hap_window_c:497 WC 1: P1 = 5.439
6025095731 shelly_output.cpp:67    Output 2: on -> off (cal1)
6025103439 shelly_hap_window_c:503 WC 1: calibration done, move_time 5089, move_power 4.043
6025109395 shelly_hap_window_c:330 WC 1: State: cal1 -> postcal1 (14 -> 15)
6025283209 mgos_sys_config.c:323   Saved to conf9.json
6025289787 shelly_hap_window_c:350 WC 1: Tgt pos 0.00 -> 50.00 (postcal1)
6025295783 shelly_hap_window_c:330 WC 1: State: postcal1 -> idle (15 -> 0)
6025302415 shelly_hap_window_c:330 WC 1: State: idle -> move (0 -> 20)
6025394756 shelly_output.cpp:67    Output 1: off -> on (move)
6025400647 shelly_hap_window_c:330 WC 1: State: move -> rampup (20 -> 22)
6025495130 shelly_hap_window_c:552 P = 7.42 -> 4.04
6025501222 shelly_hap_window_c:330 WC 1: State: rampup -> moving (22 -> 23)
6025896919 shelly_hap_window_c:340 WC 1: Cur pos 5.66 -> 7.62, P = 1.07
6026697084 shelly_hap_window_c:340 WC 1: Cur pos 21.38 -> 23.34, P = 1.86
6027497078 shelly_hap_window_c:340 WC 1: Cur pos 37.10 -> 39.06, P = 2.82
6028097069 shelly_hap_window_c:350 WC 1: Tgt pos 50.00 -> 50.85 (fixup)
6028102091 shelly_output.cpp:67    Output 1: on -> off (moving)
6028107890 shelly_hap_window_c:330 WC 1: State: moving -> stop (23 -> 24)
6028298651 mgos_sys_config.c:323   Saved to conf9.json
6028305228 shelly_hap_window_c:330 WC 1: State: stop -> stopping (24 -> 25)
6028313886 shelly_hap_window_c:330 WC 1: State: stopping -> idle (25 -> 0)
6032592161 shelly_main.cpp:492     Up 6032.58, HAP 0/1/16 ns 1, RAM: 208676/180500; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:50.85 tp:50.85 md:0 lmd:1
6040591710 shelly_main.cpp:492     Up 6040.58, HAP 0/1/16 ns 1, RAM: 208656/180500; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:50.85 tp:50.85 md:0 lmd:1

Nastras, 2024-10-26:
Manually closed via the Web Gui from your firmware.

6153907716 shelly_debug.cpp:231    No log file, sending new entries
6160591707 shelly_main.cpp:492     Up 6160.58, HAP 0/1/16 ns 1, RAM: 208268/180500; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:50.85 tp:50.85 md:0 lmd:1
6168592081 shelly_main.cpp:492     Up 6168.58, HAP 0/1/16 ns 1, RAM: 208268/180500; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:50.85 tp:50.85 md:0 lmd:1
6170828402 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6171851959 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6173077519 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6175125574 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6175944470 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6176592116 shelly_main.cpp:492     Up 6176.58, HAP 0/1/16 ns 1, RAM: 208268/180500; st 35; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:50.85 tp:50.85 md:0 lmd:1
6176604133 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6177887402 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6179426178 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6181271036 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6181283837 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.178.25:57499
6181289869 shelly_hap_window_c:350 WC 1: Tgt pos 50.85 -> 0.00 (RPC)
6181299823 shelly_hap_window_c:330 WC 1: State: idle -> move (0 -> 20)
6181320675 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6181394962 shelly_output.cpp:67    Output 2: off -> on (move)
6181400972 shelly_hap_window_c:330 WC 1: State: move -> rampup (20 -> 22)
6181495062 shelly_hap_window_c:552 P = 9.59 -> 4.04
6181501155 shelly_hap_window_c:330 WC 1: State: rampup -> moving (22 -> 23)
6181696935 shelly_hap_window_c:340 WC 1: Cur pos 49.13 -> 47.16, P = 1.82
6181708538 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6182497136 shelly_hap_window_c:340 WC 1: Cur pos 33.41 -> 31.44, P = 1.68
6182909510 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6183297005 shelly_hap_window_c:340 WC 1: Cur pos 17.69 -> 15.72, P = 1.99
6183803041 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6184096972 shelly_hap_window_c:340 WC 1: Cur pos 1.97 -> 0.00, P = 3.55
6184592051 shelly_main.cpp:492     Up 6184.58, HAP 0/1/16 ns 1, RAM: 208680/180500; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:0.00 tp:0.00 md:2 lmd:2
6184605087 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6185980615 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57499
6186595881 shelly_output.cpp:67    Output 2: on -> off (moving)
6186601839 shelly_hap_window_c:330 WC 1: State: moving -> stop (23 -> 24)
6186799432 mgos_sys_config.c:323   Saved to conf9.json
6186806085 shelly_hap_window_c:330 WC 1: State: stop -> stopping (24 -> 25)
6186816898 shelly_hap_window_c:330 WC 1: State: stopping -> idle (25 -> 0)
6192592057 shelly_main.cpp:492     Up 6192.58, HAP 0/1/16 ns 1, RAM: 208680/180500; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:0.00 tp:0.00 md:0 lmd:2

Nastras, 2024-10-26:
Manually open via the Web Gui from your firmware.

6290232512 shelly_debug.cpp:231    No log file, sending new entries
6296591989 shelly_main.cpp:492     Up 6296.58, HAP 0/1/16 ns 1, RAM: 208856/179588; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:0.00 tp:0.00 md:0 lmd:2
6304592026 shelly_main.cpp:492     Up 6304.58, HAP 0/1/16 ns 1, RAM: 209280/179588; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:0.00 tp:0.00 md:0 lmd:2
6312591993 shelly_main.cpp:492     Up 6312.58, HAP 0/1/16 ns 1, RAM: 209280/179588; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:0.00 tp:0.00 md:0 lmd:2
6315007796 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6316032115 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6316851282 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6317875692 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6318899096 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6320128126 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6320591980 shelly_main.cpp:492     Up 6320.58, HAP 0/1/16 ns 1, RAM: 209280/179588; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:0.00 tp:0.00 md:0 lmd:2
6320742799 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6322381170 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6322791014 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6324019565 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6324942937 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6324956726 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.178.25:57547
6324963474 shelly_hap_window_c:350 WC 1: Tgt pos 0.00 -> 100.00 (RPC)
6324996505 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6325010593 shelly_hap_window_c:330 WC 1: State: idle -> move (0 -> 20)
6325094799 shelly_output.cpp:67    Output 1: off -> on (move)
6325100767 shelly_hap_window_c:330 WC 1: State: move -> rampup (20 -> 22)
6325195097 shelly_hap_window_c:552 P = 8.37 -> 4.04
6325201225 shelly_hap_window_c:330 WC 1: State: rampup -> moving (22 -> 23)
6325996992 shelly_hap_window_c:340 WC 1: Cur pos 13.52 -> 15.48, P = 0.00
6326013130 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6326688765 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6326797167 shelly_hap_window_c:340 WC 1: Cur pos 29.40 -> 31.20, P = 1.43
6327597024 shelly_hap_window_c:340 WC 1: Cur pos 44.96 -> 46.92, P = 2.59
6327673740 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6328397032 shelly_hap_window_c:340 WC 1: Cur pos 60.68 -> 62.64, P = 3.54
6328592405 shelly_main.cpp:492     Up 6328.58, HAP 0/1/16 ns 1, RAM: 209256/179588; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:64.61 tp:100.00 md:1 lmd:1
6328705139 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6329197028 shelly_hap_window_c:340 WC 1: Cur pos 76.40 -> 78.37, P = 4.04
6329997088 shelly_hap_window_c:340 WC 1: Cur pos 92.12 -> 94.09, P = 4.02
6330295853 shelly_output.cpp:67    Output 1: on -> off (moving)
6330301741 shelly_hap_window_c:330 WC 1: State: moving -> stop (23 -> 24)
6330508641 mgos_sys_config.c:323   Saved to conf9.json
6330515129 shelly_hap_window_c:330 WC 1: State: stop -> stopping (24 -> 25)
6330527068 shelly_hap_window_c:330 WC 1: State: stopping -> idle (25 -> 0)
6336592360 shelly_main.cpp:492     Up 6336.58, HAP 0/1/16 ns 1, RAM: 209276/179588; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:100.00 tp:100.00 md:0 lmd:1
6344591956 shelly_main.cpp:492     Up 6344.58, HAP 0/1/16 ns 1, RAM: 209256/179588; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:100.00 tp:100.00 md:0 lmd:1
6352284787 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6352593135 shelly_main.cpp:492     Up 6352.58, HAP 0/1/16 ns 1, RAM: 209276/179588; st 36; 4.1: c:1 mp:4.04 ip:5.00 mt_ms:5089 cp:100.00 tp:100.00 md:0 lmd:1
6353306871 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6354330455 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547
6355354618 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:57547

Nastras, 2024-10-26:
I hope you can recognize something?

markirb, 2024-10-27:
Not quite but maybe...

I think it is the same as:

https://github.com/mongoose-os-apps/shelly-homekit/issues/985

I will close this issue here and continue there, I think I at least have a hint what you can try

closed 2024-10-27 by markirb