mongoose-os-apps/shelly-homekit #1547: 2.13.0 Upgrade for Plus Devices

kloibs, 2024-10-26:
Hi there,

I‘m having trouble updating my Plus devices to 2.13.0. Yes, I read the release notes regarding reverting to stock first, but unfortunately that’s where my issue lies. 

When I use the revert to stock option in the UI my Shelly will reboot and show the following screen indefinitely (see attached)

I am unable to add the Shelly in the Shelly app or flash it back to Shelly HomeKit firmware without physical access to the relay itself.

Is there something wrong with the revert to stock UI button? I prefer to keep all of my Shellys up to date but if that’s the case then I’ll avoid it for as long as possible.

Thank you to everyone for the revived development!
[image]

markirb, 2024-10-26:
Looks more Like Browser Cache? What happens if you just refresh?

kloibs, 2024-10-26:
Nothing happens. Unfortunately the only way to proceed was for me to physically access the Shelly and reset it using the little button on the back.

This was the case on two separate Shelly Plus 1PMs 👍🏼

markirb, 2024-10-26:
Ok strange then. 

I mean it sounds like there's some part of your config that stock does not like. 

If you reset in Shelly HomeKit before reverting then it should not happen. But then you loose your config... but this avoids the physical interaction

kloibs, 2024-10-26:
It’s happened on two separate relays so far, so could be my config potentially, yes.

Either way when it comes to having to choose between taking all of my relays out of the wall to update them or reconfigure my automations, less than ideal.

Just wanted to make it known 👍🏼

markirb, 2024-10-26:
Yeah. Let's see if others face this as well.

I guess we cannot do much about it though. From the issues somehow 1PM seems to be the most picky... I wonder why 😀

kloibs, 2024-10-26:
I’ve got a mix in my setup: 2.5, +2PM, 1, +1, 1PM, +1PM, and other than what was mentioned in the release notes the update was fine with everything else 🥳

metallus42, 2024-10-26:
Same here! +1PM. Didnt reset on the HomeKit OS as to not loose my config and now I am stuck at the "Connecting Screen". I rlly dont want to pull it out of the wall but I guess thats my only option... right?

kloibs, 2024-10-26:
@markirb I can try with my other Plus series devices if you think this is a Plus 1PM specific issue, let me know!

markirb, 2024-10-27:
I currently do not think it is a general problem.

Might be related to:
https://github.com/mongoose-os-apps/shelly-homekit/issues/1433

I think it can only be related to one file.
Before you do anything on a new devices can you send me the response of (this does contain private data, you can send it via email in my GitHub profile):

curl http://A.B:C:D/RPC/FS.Get -d '{"filename": "conf9.json"}'

I can see if anything is strange there. 

Then you might try to do the process, to tell me if it does work or not. Make it with a Shelly that is easily accessible incase you need to access It physically. 

If it does not work I would also be interested to see any error messages in the browser console.

I mean the Shelly seems to be in a state where it does not seem to give the Webif.
It still might be available to a Reset via url (on Stock):

curl http://A.B:C:D/RPC/Shelly.FactoryReset

metallus42, 2024-10-27:
Hi @markirb,

the First +1 I did try to update was not accessible today. Not even conneted to my WiFI anymore. Maybe it timed out and shut down? I cycled the power and am now at the point I was wen I made my first comment here.
[image]

`curl http://a.b.c.d/RPC/Shelly.FactoryReset` sadly does not work. It gives
`Not Found% `.

`curl http://a.b.c.d/RPC/FS.Get -d '{"filename": "conf9.json"}'` gives me no output on all of my devices running Shelly HomeKit. Or did you want me to do this on one with stock FW?

kloibs, 2024-10-27:
> I currently do not think it is a general problem.
>
> Might be related to: #1433
>
> I think it can only be related to one file. Before you do anything on a new devices can you send me the response of [...]

I should have some time to test it this week. Cheers!

markirb, 2024-10-28:
> Hi @markirb,
> 
> the First +1 (Shelly A) I did try to update was not accessible today. Not even conneted to my WiFI anymore. Maybe it timed out and shut down? I cycled the power and am now at the point I was when I made my first comment here. <img alt="Bildschirmfoto 2024-10-27 um 22 45 01" width="1500" src="https://private-user-images.githubusercontent.com/74405969/380533883-e695ba03-8dce-4310-a290-2de66bca94d9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAxMTE0NDYsIm5iZiI6MTczMDExMTE0NiwicGF0aCI6Ii83NDQwNTk2OS8zODA1MzM4ODMtZTY5NWJhMDMtOGRjZS00MzEwLWEyOTAtMmRlNjZiY2E5NGQ5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI4VDEwMjU0NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThjYjBlYTNiMWZkYzFkNzQ1ZGM5MjhjMDZjZGZjNzZjOTgxNTE1ZDI4NzgyNmFhY2M3MWI3MzE2MjllMWY2NGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.s7-gTNA4ykPWch9DwlafoKfeuNAKDO8frwjQyYSj_IA">
> 
> `curl http://a.b.c.d/RPC/Shelly.FactoryReset` sadly does not work. It gives `Not Found% `.

I figured out the RPC has to be lower case so `curl http://a.b.c.d/rpc/Shelly.FactoryReset` ?

> 
> `curl http://a.b.c.d/RPC/FS.Get -d '{"filename": "conf9.json"}'` gives me no output on all of my devices running Shelly HomeKit. I tried with all my +1 and it worked for none of them.

That is really strange. Also that this should work after reset....
Does
`http://a.b.c.d/rpc/FS.List"` work?

> 
> Update: Reseted another one of the +1 (Shelly B) via the Button and then again via the Web UI, then connected back to my WiFI. Homekit Config is lost but I now have the output of `curl http://a.b.c.d/RPC/FS.Get -d '{"filename": "conf9.json"}'`. I have sent it per email! This time the reset worked and I was able to update stock from 1.3.3 to 1.4.4 and then update to shellyhomekit 1.13.
> 
> So maybe in the current state of my and @kloibs (?) shellys this file is defective or somehow not correcty accessible?

Yes this is somehow likely. We have to find out why. I think it is also linked to the other issue I already linked

> 
> Ill try to reset one via the web UI only since that could give us a way to solve the issue without needing physical access...

metallus42, 2024-10-28:
> I figured out the RPC has to be lower case so curl 'http://a.b.c.d/rpc/Shelly.FactoryReset' ?

Shelly A: `{"code":500,"message":"unauthorized"}%  `

>Does 'http://a.b.c.d/rpc/FS.List"' work?

Just making sure: Doing it with the " at the end always brings me in dquote no matter what IP I use. So that was a typo?
Doing it without:
Shelly A: {"code":404,"message":"No handler for FS.Lis"} 
Shelly B: '{"code":404,"message":"No handler for FS.Lis"}'
Shlly C (Untouched  ShellyPlus1 running Shelly Homekit 2.12.2 for months): '{"code":404,"message":"No handler for FS.Lis"}'

Let me know in case I should try anything else :)

markirb, 2024-10-28:
Typo...

lindingerm, 2024-10-30:
Same Problem here.

1 ShellyPlus1 worked.
The other ShellyPlus1 hangs after "Revert to Stock 1.3.3". Just the "Connecting to Shelly" screen.

With http://a.b.c.d/rpc/Shelly.FactoryReset the same as above: {"code":500,"message":"unauthorized"}

tothbe, 2024-10-30:
I am a bit lost with this update. I am using Gen 1 Shelly 1 devices. As I read this back to stock and upgrade thing is only valid for Plus devices. So for Gen 1 Shelly 1 I can just simply upgrade them with upgrade button? Thanks for any feedback.

lindingerm, 2024-10-31:
After i reseted my second ShellyPlus1 the Update worked.

Now with my third ShellyPlus1 th Problem changed: The Shelly is at original firmware v1.4.4 but cant update to shelly-homekit firmware it boots always back to stock webUI.

markirb, 2024-10-31:
> I am a bit lost with this update. I am using Gen 1 Shelly 1 devices. As I read this back to stock and upgrade thing is only valid for Plus devices. So for Gen 1 Shelly 1 I can just simply upgrade them with upgrade button? Thanks for any feedback.

Exactly

markirb, 2024-10-31:
> After i reseted my second ShellyPlus1 the Update worked.
> 
> 
> 
> Now with my third ShellyPlus1 th Problem changed: The Shelly is at original firmware v1.4.4 but cant update to shelly-homekit firmware it boots always back to stock webUI.

Yet another case.
You did try to factory reset? 
Do you have any logs?

lindingerm, 2024-10-31:
I tried a factory reset, different browsers and multiple times to update. It comes always back to the stock webUI.
As this is a critical device for my smart home, i have now switched to HomeAssistant.

metallus42, 2024-10-31:
I have reseted and updated one Plus1 (Shelly C) successfully without needing physical access AND w/o needing to reconfigure it in Homekit!
1. Use reset Button (not the one under the homekit tile, the one in the System tile)
2. Connect to AP, 192.168.33.1, reconfigure your Wifi
3. Go to whatever IP your shelly now has in your LAN
4. Firmware tile -> Revert
5. Reload and the shelly is now on stock FW
6. Update stock FW to latest
7. http://A.B.C.D/ota?url=http://shelly.rojer.cloud/update it
8. Reload and change the Name back to what it was before. For me Homekit was connected still. 
9. Setup your Password again (if u wish so)

This makes the process a bit tideous but possible w/o phyiscal interaction. I wonder if this could be automated?

markirb, 2024-10-31:
> Same Problem here.
> 
> 
> 
> 1 ShellyPlus1 worked.
> 
> The other ShellyPlus1 hangs after "Revert to Stock 1.3.3". Just the "Connecting to Shelly" screen.
> 
> 
> 
> With http://a.b.c.d/rpc/Shelly.FactoryReset the same as above: {"code":500,"message":"unauthorized"}
> 
> 
> 
> 

Did you have a password set?

johannesvogel, 2024-10-31:
Also happened to both of my Shelly Plus1. Had to reset by getting them out of the wall, since the other reset methods on the shelly website (turning off power 1 time and pressing switch 5 times / turning off power 5 times) did not work either. Was then able to upgrade the Shelly firmware and then install the shelly-homekit update.

lindingerm, 2024-11-01:
> Did you have a password set?

No password.

markirb, 2024-11-01:
Please anyone if we want to have the reason for this give me the content of this call before performing any updates. Send content via email in my GitHub profile.

curl http://a.b.c.d/RPC/FS.Get -d '{"filename": "conf9.json"}'

tschutschi, 2024-11-01:
> Please anyone if we want to have the reason for this give me the content of this call before performing any updates. Send content via email in my GitHub profile.
> 
> curl http://a.b.c.d/RPC/FS.Get -d '{"filename": "conf9.json"}'

Done!

markirb, 2024-11-01:
So there seems to be another case:

1. Revert to 1.3.3 works
2. Update to 1.4.4 works
3. Upgrade to Shelly Homekit fails.

In this case a simple reboot was observed to help.
However this is completely different than the "Connecting to shelly" issue

datadorf, 2024-11-04:
I have the same issue, after reverting I get this "Connecting to Shelly" screen.

datadorf, 2024-11-06:
I could not revert to stock firmware, with pressing the reset button, so I installed a new shelly plus 1 now.

vntus, 2024-11-12:
I had a lot of trouble flashing the firmware on a Shelly Plus I4. After reverting from the homekit to the shelly firmware and updating it to version 1.4.4. it was impossible for me to install the homekit firmware 2.13.1. I did several factory resets, tried to install previous versions, etc. and finally, after physically accessing the device and pressing the reset button for 10 seconds, I managed to install the homekit firmware. Now everything works perfectly.

CyberMrProper, 2024-11-15:
Any news on this issue? it's happening to me. I tried with soft and hard reset (physical button), nothing seems to work to make the installation a success, it is always going back to stock firmware after the update reaches 100%.

markirb, 2024-11-15:
> Any news on this issue? it's happening to me. I tried with soft and hard reset (physical button), nothing seems to work to make the installation a success, it is always going back to stock firmware after the update reaches 100%.

this is a new case, which was not described here so far. What Shelly Device is this? Which version are you upgrading from?

SebastianStolze, 2024-11-16:
I had also trouble with one of my shellys upgrading from 1.4.4 to the latest firmware. The only thing that helped in the end was doing a factory reset with the shelly app itself, if I did it over the ui it won't make any difference.

markirb, 2024-12-08:
For upgrade issues with Plus1PM there might be a clue now:

https://github.com/mongoose-os-apps/shelly-homekit/issues/1567

fubu2k, 2024-12-09:
Shelly 1 Plus:
Revert to 1.3.3 works
Update to 1.4.4 works
Upgrade to Shelly Homekit fails with http://a.b.c.d/ota?url=http://shelly.rojer.cloud/update

Hard and Soft-Reset did not work. After the Homekit-Upgrade the original Shelly GUI 1.4.4 is shown

galdo320, 2024-12-22:
FYI, it seems like the issue is fixed on my end. I didn't encounter any problems with my Shelly Plus 1 PM during these steps:

Revert to firmware version 1.3.3
Update to version 1.4.4
Upgrade to Shelly Homekit using http://a.b.c.d/ota?url=http://shelly.rojer.cloud/update
I’ve seen others mention issues, but everything worked smoothly for me.

mossimac, 2024-12-30:
Is there a thread for the Shelly Plus1 on 1.4.4 not flashing to the homekit firmware? I feel like I've tried everything. Downgrading seems to pretend to work, but then refreshes and still on 1.4.4.

hejsiri, 2025-02-13:
I have the same problem on Shelly 2PM

[image]

daneb255, 2025-03-10:
I had 3 Shellys, two of them worked, one didn't (even after several attempts)

Steps:
Revert to firmware version 1.3.3
Update to version 1.4.4
Upgrade to Shelly Homekit using http://a.b.c.d/ota?url=http://shelly.rojer.cloud/update

still on factory 1.4.4 shelly firmware ..

sello89, 2025-03-15:
> Is there a thread for the Shelly Plus1 on 1.4.4 not flashing to the homekit firmware? I feel like I've tried everything. Downgrading seems to pretend to work, but then refreshes and still on 1.4.4. 

Same problem for me. Did you find a solution for your problem?

markirb, 2025-03-16:
If anyone having this problem in Germany can you send me the device so I can see the error. Will send it back afterwards.

sello89, 2025-03-16:
> If anyone having this problem in Germany can you send me the device so I can see the error. Will send it back afterwards.

Yes sure! Please send me your adresse and i will sent you my Shelly Plus 1.

sello89, 2025-03-23:
I know Shelly often got problems with the update regarding to the IP/ Wlan setting. Especially when there is set an fixed IP adress.

Could somebody with this problem try to set up a other wlan hotspot or deactivate the fixed ip adress and then try the reset and load again?

Worth a try i would say.

skyedrummer, 2025-04-01:
After reverting in shelly-homekit interface to stock to update to 2.14 from 2.13, ended up in an endless cycle of failed shelly-homekit installs. The only thing that allowed me to install homekit again was to remove from the wall and press physical reset button on plus1 for +10 secs, then managed to install 2.13.0. Can’t update to 2.14 any way, ota or otherwise. Have tried update to 2.14 on 2 other plus1 - doesn’t update.

freswa, 2025-05-17:
Run into the same issue. Just wanted to add that at least the 2PM can be resettet by powering them off and on (did that by pulling the fuse) and then pressing any button 5 times very quickly. Maybe it helps someone else in the process :)

skyedrummer, 2025-06-14:
Is there ANY update on flashing the Plus1? Just tried it again and nothing, doesn't update, it says it does in the OG Shelly interface and then I'm back to the Shelly interface. Should we be assuming that the firmware just doesn't work on the plus1?? I've now had to remove over 6 plus1 from the wall and replace with shelly1 - complete PITA. Going forward, should I be trying to buy older shelly1 because the plus1 will not accept this firmware?

taitems, 2025-09-22:
I've tried:
- `1.7` to `latest`
- `1.7` to specific firmware zip
- `1.6.2` to `latest`
- `1.6.2` to specific firmware zip
- `1.5.1` to `latest`

None of which work. Logs are annoying in Shelly because when the OTA "succeeds" it reboots, so here's a screen recording.

https://github.com/user-attachments/assets/1bb1d82e-07e8-48be-9fd2-67017a830ddc