mongoose-os-apps/shelly-homekit #1550: Homekit No Response

chickenpesto, 2024-10-30:
Hi all, 

have a shelly 1 plus, everything has worked up until actually switching the light on in homekit. It simply says no response. On the IP address site, it says that it is paired but no connections. 
Has anyone else had this problem and figured out what is wrong?
Thanks

markirb, 2024-10-31:
Which firmware? Did you reboot?

filic83, 2024-11-03:
Hello I have same problem. In UI web interface it is paired but there is no active connection. Reboot helps and in 30minutes whole situation repeats. It is same for shelly1, shelly1 mini gen3, shelly 2.5. Sw version 2.13.0

pfrasun, 2024-11-04:
Hi! I had same issue with several devices after 2.13.0 upgrade. I tried restarts etc. but only thing which helped was reverting to original FW and reset. After that process I updated all devices having issues to 2.13.0 and reconnected them to HomeKit. Since that devices have been stable (until now) and have no connection issues. I know it is not a perfect solution, but only one I found working for me.

tvavrys, 2024-11-10:
I have this issue with `Shelly25` for months. Power saving is disabled.  It's like the whole device goes to hibernation and the only way how to make it work in Apple Home again is to access the Web UI of the shelly device from the desktop, wait until it fully renders (connects) and then it's accessible from Apple Home again. Uptime says days.

pfrasun, 2024-11-16:
At least for me ping worked. Devices were in WiFi network, visible and listed as active devices by router and UI was also accessible, but devices were loosing HomeKit connection after few hours of operation after restart.
Power states did not change anything, on the end only thing which help was restoring shelly FW, reset with factory FW and update to mongoose. After doing it all devices are back to more predictable behaviour.
During analysis I also found one Shelly 1 Plus having some issues with connectivity and removed it permanently. It was same known issue `mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.`

aequitas, 2024-11-18:
Experiencing the same issue, only with 'plus' devices, not with some older shelly 1 and 2.5's. 

- Can connect using the web interface, no interuptions
- Can stream logs, no interuptions
- Can ping the device, no interuptions or higher pings during the issues
- In Apple Home, the devices constantly switch from working normally to 'No response', every half minute or so
- Power savings is disabled
- All devices on latest 2.13.2 firmware

iTzJacopoo, 2024-12-14:
same problem here, im using 1plus and even after reboot and repairing it doesn’t work.
it says “no response” no matter what, i tried separing 2ghz channel or connecting it to a different network but nothing…

markirb, 2024-12-15:
> At least for me ping worked. Devices were in WiFi network, visible and listed as active devices by router and UI was also accessible, but devices were loosing HomeKit connection after few hours of operation after restart.
> 
> Power states did not change anything, on the end only thing which help was restoring shelly FW, reset with factory FW and update to mongoose. After doing it all devices are back to more predictable behaviour.
> 
> During analysis I also found one Shelly 1 Plus having some issues with connectivity and removed it permanently. It was same known issue `mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.`, I was assuming that this device may cause trouble to WiFi / Router, but removing was not making situation better, so I decided to try different track and reseted problematic devices.

Those RPC warnings is only garbage on the UART pins of the device. This should not influence behavior

vandinter, 2024-12-21:
same (or similar issue) since 2.13: Several older shellys 1 and 2 running smoothly with previous firmwares for 1-2 years are now regularly disconnected / not available in HK. When reachable via UI the report reverted in WiFi.

zobe123, 2024-12-25:
Also same (or similar issue) since 2.13: Several older shellys 1 and 2 running smoothly with previous firmwares for 1-2 years are now regularly disconnected / not available in HK. When reachable via UI the report reverted in WiFi.

closed 2025-01-23 by github-actions[bot]