mongoose-os-apps/shelly-homekit #1552: Shelly 1 v3 dead after swap inputs enabled

mikecitt, 2024-11-01:
I wanted to set up Shelly 1 as a Garage Door Opener, and during testing, I enabled the "Swap Inputs" option. After that, it became unavailable. I wasn't even able to reset it to factory settings. Do you have any suggestions?

markirb, 2024-11-01:
Yes. Factory reset. Will be fixed in next version.

https://github.com/mongoose-os-apps/shelly-homekit/discussions/1545#discussioncomment-11058710

closed 2024-11-01 by markirb

mikecitt, 2024-11-01:
@markirb I did try, but as I wrote, I'm not able to reset it as well.

markirb, 2024-11-01:
After 5 powerups it should reset itself

mikecitt, 2024-11-01:
@markirb That’s exactly what I tried, and nothing. I’ve broken 2 devices. :(

markirb, 2024-11-01:
One of those will work:

https://github.com/mongoose-os-apps/shelly-homekit/wiki/Recovery

markirb, 2024-11-01:
Shelly 1 has no button.

Input will also not work.

So your only option is the reboot counter so 5 reboots. This should work

markirb, 2024-11-01:
I would switch off 5 sec switch on 5 secs and repeat this 5 times or more

markirb, 2024-11-01:
If all this does not work you need a serial interface or you get a Shelly addon...

mikecitt, 2024-11-01:
after few random restarts it got to work! thank you so much!