mongoose-os-apps/shelly-homekit #1553: Offset sensor doesn't work with decimals

lucaspinelli85, 2024-11-04:
Ciao a tutti ho notato che se imposto un offset con numero intero ad esempio 1,2,3 ecc... il valore della temperatura si corregge, ma se imposto un numero non intero con il decimale esempio 1.3, 0.6, 2.1 ecc... il valore della temperatura non si corregge
[image]

markirb, 2024-11-06:
should work in 2.13.1

lucaspinelli85, 2024-11-06:
Ok it works!!!Ho un’altra domanda: perché in app viene mostrata la temperatura con precisione di 0,5 gradi invece di usare tutti i decimali ed avere la precisione di 0,1 gradi? In che modo viene fatto questo arrotondamento?GrazieInviato da iPhone di Luca

markirb, 2024-11-07:
This is a limitation of apples Home App. Third party apps like eve can do this.