mongoose-os-apps/shelly-homekit #1556: Support for Shelly Plus RGBW PM

JG271, 2024-11-09:
Hello,

I would like to switch the four channels of the Shelly Plus RGBW PM individually in white mode. This doesn't seem to be possible at the moment. Is it possible to expand the firmware to do this?

Thanks a lot!
Jens

### Discussed in https://github.com/mongoose-os-apps/shelly-homekit/discussions/1502

<div type='discussions-op-text'>

<sup>Originally posted by **PabloIT** June 26, 2024</sup>
Will the new Shelly Plus RGBW PM be supported by Mongoose?
https://www.shelly.com/en/products/shop/shelly-plus-rgbw-pm</div>

markirb, 2024-11-09:
It is implemented in a pull request. 

You can get it here and test and provide feedback:

https://github.com/mongoose-os-apps/shelly-homekit/actions/runs/11425337810

JG271, 2024-11-11:
Unfortunately, no change can be detected. All lights are still switched on at the same time. The manual push buttons are no longer working. I also tested 2.13.0-rgbw2_switch4, #1558. Unfortunately, this also did not bring any improvement.
Is there a possibility that I made a mistake when flashing?

markirb, 2024-11-12:
Not sure if we are on the same page.

What do you expect vs what do you see?

JG271, 2024-11-13:
Hey, I hope I can explain it clearly. The module has four individual outputs to switch four lights separately. When you switch on one of the four channels separately via the web interface, all four outputs are always switched on at the same time. It doesn't matter which of the four channels is switched on, all four lights always come on, which means that all outputs are activated simultaneously via any of the four switches.
Furthermore, it is currently not possible to switch the lights on or off using a physical switch via the device inputs.
I hope I was able to describe the problem in a reasonably understandable way.
Thank you in advance!

markirb, 2024-11-13:
The first problem sound like the bug solved in https://github.com/mongoose-os-apps/shelly-homekit/issues/1558

For the second one - this should be possible. Can you show me a screenshot of your web interface?

JG271, 2024-11-13:
Hey, thanks for the feedback.
I was able to solve the second problem with the switches in the meantime, wiring error.
The behavior is now as described in the error #1558.
I installed the build described in #1558 but unfortunately it didn't make any difference.
Does the module need to be reset to factory settings first or what else could I be doing wrong?

Thanks again for your support!

markirb, 2024-11-13:
Should just work. Version should contain the hash 3b4b0e

closed 2024-11-13 by markirb