mongoose-os-apps/shelly-homekit #1557: Shelly1 Plus not turning on the light when activating physical switch

CyberMrProper, 2024-11-09:
From time to time my Shelly1 Plus when I try to turn on the light from the physical switch, it doesn’t turn my light on. If I try the switch again it works… it happens maybe once a week or so (it is configured in edge mode)

markirb, 2024-11-13:
Strange indeed. 

Edges cannot be missed… maybe it is because the input is bouncing and is not devounced…

Can you maybe try to reproduce with a log file?

CyberMrProper, 2024-11-13:
Hard to reproduce with the log opened. I have both, shelly1 and shely1plus at home installed , this issue never happened to me with shelly1, I started experiencing this issue when I bought shelly1plus devices so something is different in that hardware / software combination.

markirb, 2024-11-14:
Yes hard to reproduce.... does it happen if you push really often?

CyberMrProper, 2024-11-14:
Not really, usual case, I enter the room and I try to tun on the light and nothing happens. Then I click again and the light comes on.  Not happening very often, but enough to annoy me :)

iyerusad, 2024-11-14:
Seeing potentially similar, abliet no amount of toggling seems to do anything on my end.

(copied from https://github.com/mongoose-os-apps/shelly-homekit/pull/1542#issuecomment-2477186715):
Deployed 2.13.2. Tried toggle and edge modes, doesn't appear to do anything. Sticking to Edge mode for now (seems most ideal for testing). Nothing in Debug log when toggle physical switch, tried rebooting Shelly same thing. Software control (web, homekit) successfully works, just physical toggle doesn't appear to doing anything. Confirmed working in Factory firmware last night (before migrating to 2.13.1).

Fairly new to Shelly so apologizes if missing something obvious.

[image]

Config:
--
- **Shelly +RGBW PM**, 2.13.2 Homekit build
- **Device Profile:** White / "Lights x4" mode. O1 used, O2-O4 HAP hidden
- **24V DC input:** (positive/red into `+` and negative/white into `┴` of Shelly)
- **O1:** (3) LED strips, negatives spliced into one going to `O1` of Shelly
  - Positives of LED strip spliced into `+` port of the Shelly (effectively 24V input positive)
- **I1:** Load side of Single Pole switch into `I1` of Shelly
  - Line/Live side is spliced into negative/white into `┴` of Shelly (effectively 24V input negative)

[image]

markirb, 2025-02-06:
I see. It detects a Long press, I think this makes all the difference....

Can you share your settings page, so I can see what case applies?

CyberMrProper, 2025-02-06:
Here you have it 

[image]

CyberMrProper, 2025-02-06:
Also check that in the logs there are some additional entries

Input 1: single (state 0)

That are not followed by the relay activation.  I see that according to my config it should be always “change” instead of single/long

CyberMrProper, 2025-02-06:
I think this is somehow related to https://github.com/mongoose-os-apps/shelly-homekit/issues/1417

For some of my shellies sometimes the config comes as follows ( it goes away if I save and reboot) but it comes back randomly. 

[image]

markirb, 2025-02-09:
This is not related to the ghost input as this does not control anything.

I think it is more related to your switch "bouncing". As far as I read the code this can be a problem....

CyberMrProper, 2025-02-09:
Weird, this is happening in different homes with different physical switch brands. Besides, with the stock firmware I’ve never experienced this issue. Something could be off in the way events are being handled.

markirb, 2025-02-09:
Every switch bounces.

It’s up to the code. 
I think currently this code does not handle it well.
Shelly stock firmware seems to.
At least that’s my current suspicion when I look at the code