mongoose-os-apps/shelly-homekit #1558: ShellyPlusRGBWPM Turns on/off random lights when on/off is used

steezeman, 2024-11-10:
I just installed this firmware on my new ShellyPlusRGBWPM device. So far, it seems great, but there is one glaring issue. When I use the dimmer functionality in the Home app, it works fine, but when I tap the icons in the Home app to turn the lights on or off, seemingly random lights turn on or off. I played with the system for half an hour, but I can't see a clear pattern.

[image]

[image]

Clicking the lightbulb icon (not the rest of the object) is how I create the behavior

[image]

I tried rebooting several times. I also upgraded the firmware to the one linked in https://github.com/mongoose-os-apps/shelly-homekit/issues/1556

What else do you need from me to repro this?

markirb, 2024-11-10:
This is strange. Can you add the device logs, maybe this shows something

steezeman, 2024-11-10:
```
1681050030 shelly_output.cpp:82    Output 3: 0.979513 (transition)
1681055611 shelly_output.cpp:73    Duty: 0.986
1681061578 shelly_output.cpp:82    Output 3: 0.985909 (transition)
1681067510 shelly_output.cpp:73    Duty: 0.992
1681073464 shelly_output.cpp:82    Output 3: 0.991560 (transition)
1681078159 shelly_output.cpp:73    Duty: 0.997
1681084133 shelly_output.cpp:82    Output 3: 0.997456 (transition)
1681091464 shelly_light_bulb_co:56 Transition finished, end state: [w=1.00]
1681095107 shelly_output.cpp:73    Duty: 1.000
1681099712 shelly_output.cpp:79    Output 3: ON (transition)
1681678653 mgos_sys_config.c:323   Saved to conf9.json
1681702346 shelly_main.cpp:492     Up 1681.68, HAP 0/1/16 ns 1, RAM: 203396/187680; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 50, h: 0, sa: 100, ct: 400; 11.3: sta: on, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 100, h: 0, sa: 100, ct: 400
1686560454 shelly_hap_light_bu:198 State changed (HAP): off => on
1686568144 shelly_light_bulb_c:102 Starting transition: [w=0.00] -> [w=1.00], 2000 ms
1686572734 shelly_white_control:38 Output 1: 0.00 => 1.00
1686582119 shelly_output.cpp:73    Duty: 0.009
1686587866 shelly_output.cpp:82    Output 4: 0.008622 (transition)
1686592597 shelly_output.cpp:73    Duty: 0.014
1686597988 shelly_output.cpp:82    Output 4: 0.014007 (transition)
1686603542 shelly_output.cpp:73    Duty: 0.019
```

A small snippet: This should just be changing one output. Output 1 goes 0 > 1 and output 4 turns on as a result

markirb, 2024-11-10:
The message „output 1“ does not mean channel 1, but it is the first output of the white controller responsible for channel 4. so the log message is correct actually….

steezeman, 2024-11-10:
So in this case, I'm toggling one of the outputs on/off. The light doesn't respond. Then I change the brightness using the dimmer slider, and it starts responding, but weirdly it takes a second channel with it, i.e. two channels respond in tandem to the dimming request. See how output 1 and 4 are interleaved in this message?

[image]

[image]

steezeman, 2024-11-10:
Further confounding matters, it seems like this doesn't happen when brightness isn't set to 100%. Is it possible that there is an integer wrapping bug somewhere in the code?

markirb, 2024-11-11:
No. But I think I know the reason. 
RGBWPM is the first Plus (ESP32) device with PWM. There was a bug fix for ESP8266 regarding behavior at 100%.

I reverted the fix in this commit:
https://github.com/mongoose-os-apps/shelly-homekit/pull/1542/commits/3b4b0e992f00d3da5a34fe7b5717ebda36b10eab

If you take the current build from here:
https://github.com/mongoose-os-apps/shelly-homekit/actions/runs/11773647636

I think it should work. Can you try?

btw: do you have any problems with resetting after reboot? (https://github.com/mongoose-os-apps/shelly-homekit/issues/1518)

steezeman, 2024-11-11:
Sure enough, that seems to have solved it. I haven't seen any problems with the device resetting like #1518 

Thanks for the <24hr fix!

markirb, 2024-11-14:
Fixed in 2.13.2

closed 2024-11-14 by markirb