mongoose-os-apps/shelly-homekit #1566: Shelly Plus RGBW PM in White Mode without functioning Input Modes

adriaanb, 2024-11-18:
Hello,

I recently set up my first Shelly (Plus RGBW PM) device and flashed it immediately, loving the firmware so far, thank you for the great work!

I did, however, run into an issue that I hope you can help with. So far, I can only control my device via HomeKit. Despite testing all all four input channels on my Shelly RGBW PM, I cannot register any events or switch the device state using the input channels.

Here’s my setup:

* A single white LED strip is connected to channel 1 on the Shelly Plus RGBW PM.
* A non-latching switch is connected between I01 and Ground.
* This configuration works as expected with the stock firmware.

Assuming that channel 1 maps onto I01, I tried all available input modes on the first device enumerated in the web interface, but was still unable to register any events. Furthermore, it seems as though I can only detach one input device at a time, as no configuration pane for any subsequent detached input is available.

Is there something I might be overlooking, or is this a known issue? Any guidance would be greatly appreciated!

markirb, 2024-11-19:
Hi,

Known issue. See here:

https://github.com/mongoose-os-apps/shelly-homekit/pull/1542

Workaround for now: hard reboot (cut power)

Will be fixed in one of the next releases.

markirb, 2024-12-03:
fixed in 2.13.3

closed 2024-12-03 by markirb

juancarlosisb, 2024-12-20:
Hi, I have the same problem on my 3 Shelly Plus RGBW PM.

They are all updated to firmware 2.13.3, and all have white LED strips connected to outputs 1 and 2, controlled by buttons 1 and 2 but none of them work with the wall button, they only work from homekit. I have also tried with all previous firmware versions and it doesn't work. The wall buttons only work with the original firmware.

Has anyone managed to fix this error?
It seems that in the latest firmware 2.13.3 this is not fixed yet.

juancarlosisb, 2024-12-20:
> Hi, I have the same problem on my 3 Shelly Plus RGBW PM.
>
> They are all updated to firmware 2.13.3, and all have white LED strips connected to outputs 1 and 2, controlled by buttons 1 and 2 but none [...]

juancarlosisb, 2024-12-20:
> Hi, I have the same problem on my 3 Shelly Plus RGBW PM.
> They are all updated to firmware 2.13.3, and all have white LED strips connected to outputs 1 and 2, controlled by buttons 1 and 2 but none [...]

UPDATE:

Sorry, as mentioned here #1542 , it is necessary to completely turn off the power and turn it back on for everything to work correctly, restarting the device by software is not enough. I spent many hours with this problem and it was solved in the simplest way :-) . Thank you very much for everything.