mongoose-os-apps/shelly-homekit #1567: Shelly Plus 1PM Update 2.13.0 -> 2.13.2 fail

Salle88, 2024-11-19:
Hello together, check update Check from 2.13.0 there is 2.13.2 available. After successfull installation and reload the version doesn't change.
Can somebody help me? 

The problem two times from 12 shelly's.

Salle88, 2024-11-19:
[image]

[image]

[image]

[image]

markirb, 2024-11-19:
What is the model?

Salle88, 2024-11-19:
ShellyPlus1PM

Salle88, 2024-11-28:
I would be grateful.

Salle88, 2024-12-03:
> not sure yet. only difference to other models is the partition reading. I can try to throw this out...

Included in 2.13.3 ?

Salle88, 2024-12-04:
59774279 shelly_debug.cpp:231    No log file, sending new entries
64447429 shelly_main.cpp:492     Up 64.44, HAP 0/1/16 ns 1, RAM: 210440/204644; st 43; 0.1: st:0 in_st:0 inm:1 ininv:0
65809397 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.96:55877
65820215 mgos_mongoose.c:66      New heap free LWM: 204592
65828193 mgos_mongoose.c:66      New heap free LWM: 202364
67029704 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.96:55877
67447126 mgos_http_server.c:180  0x3ffce3a4 HTTP connection from 192.168.178.96:55879
67466050 mgos_mongoose.c:66      New heap free LWM: 201660
67474157 mgos_mongoose.c:66      New heap free LWM: 201576
67484961 mgos_mongoose.c:66      New heap free LWM: 201532
67494207 mgos_mongoose.c:66      New heap free LWM: 201464
67508382 mgos_mongoose.c:66      New heap free LWM: 198936
67571243 mgos_http_server.c:180  0x3ffcd970 HTTP connection from 192.168.178.96:55880
67587249 mg_rpc.c:315            Shelly.GetInfo via WS_in 192.168.178.96:55880
67603046 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.96:55880
68668263 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.96:55880
69590266 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.96:55880
71639065 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.96:55880
72447782 shelly_main.cpp:492     Up 72.44, HAP 0/1/16 ns 1, RAM: 209972/198936; st 42; 0.1: st:0 in_st:0 inm:1 ininv:0
80447635 shelly_main.cpp:492     Up 80.44, HAP 0/1/16 ns 1, RAM: 210400/198936; st 42; 0.1: st:0 in_st:0 inm:1 ininv:0
86082152 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.96:55880
87100902 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.96:55880
88170711 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.96:55880
88447437 shelly_main.cpp:492     Up 88.44, HAP 0/1/16 ns 1, RAM: 210400/198936; st 42; 0.1: st:0 in_st:0 inm:1 ininv:0
88465114 mgos_http_server.c:180  0x3ffce130 HTTP connection from 192.168.178.96:55881
88474873 mgos_ota_core.c:269     Starting, timeout 600, commit timeout 0, mem 202184
88483068 mgos_mongoose.c:66      New heap free LWM: 197752
88489598 mgos_mongoose.c:66      New heap free LWM: 197372
88509268 mgos_ota_core.c:656     FW: Plus1PM esp32 2.13.3 20241203-192047/2.13.3-g1cb5084 sig 0x00
88515246 shelly_main.cpp:369     == Stopping HAP service (1)
88519926 mgos_ota_core.c:771     begin: waiting for user
88528103 shelly_main.cpp:384     HAP server state: 2
88539139 HAPPlatformTCPStrea:309 HAPPlatformTCPStreamClose ts 0x3ffcdaf0 nc 0x3ffcd8a8 0/0/16
88544287 mgos_mongoose.c:66      New heap free LWM: 195864
88554571 HAPPlatformTCPStrea:112 0x3ffc0a98 0.0.0.0:9000 HAP connection closed, ns 0/0/16 i 1
88564661 shelly_main.cpp:384     HAP server state: 0
88569692 shelly_main.cpp:357     === Destroying accessories
89160856 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.96:55880
90071367 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.96:55880
91094661 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.96:55880
93473830 shelly_ota.cpp:90       Starting firmware update
93487071 esp32xx_ota_backend:209 App: Plus1PM.bin -> app_1 enc 0, FS: fs.img -> fs_1 enc 0
93498551 mgos_ota_core.c:685     0.15% total, bootloader.bin 512 of 23728
93506040 mgos_ota_core.c:1202    Update state: 0
93636411 mgos_ota_core.c:685     1.65% total, partition-table.bin 512 of 3072
93643942 mgos_ota_core.c:1202    Update state: 0
93673061 mgos_ota_core.c:685     1.84% total, otadata.bin 512 of 8192
93680587 mgos_ota_core.c:1202    Update state: 0
93859539 esp32xx_ota_backend:303 Writing FS @ 0x390000
94161469 mgos_ota_core.c:685     2.36% total, fs.img 512 of 393216
94169098 mgos_ota_core.c:1202    Update state: 0
94177316 mgos_mongoose.c:66      New heap free LWM: 195384
96627078 shelly_main.cpp:492     Up 96.62, HAP 0/0/16 ns 0, RAM: 200652/195384; st 42; disabled
97451638 mgos_ota_core.c:685     18.82% total, fs.img 262656 of 393216
97459212 mgos_ota_core.c:1202    Update state: 0
99353797 esp32xx_ota_backend:399 fs_1 verified (4173d6212f5882d8dc10bbbd3730e728ec68d309)
99746228 esp32xx_ota_backend:299 Skip writing app (digest matches)
99759959 mgos_ota_core.c:685     27.05% total, Plus1PM.bin 512 of 1161792
99767606 mgos_ota_core.c:1202    Update state: 0
101012982 mgos_ota_core.c:685     43.51% total, Plus1PM.bin 262656 of 1161792
101021128 mgos_ota_core.c:1202    Update state: 0
102260035 mgos_ota_core.c:685     59.97% total, Plus1PM.bin 524800 of 1161792
102267671 mgos_ota_core.c:1202    Update state: 0
103500048 mgos_ota_core.c:685     76.43% total, Plus1PM.bin 786944 of 1161792
103507614 mgos_ota_core.c:1202    Update state: 0
104446940 shelly_main.cpp:492     Up 104.44, HAP 0/0/16 ns 0, RAM: 202340/195384; st 43; disabled
104728267 mgos_ota_core.c:685     92.88% total, Plus1PM.bin 1049088 of 1161792
104735826 mgos_ota_core.c:1202    Update state: 0
105255644 mgos_ota_core.c:820     Reached the end of archive
105269910 esp32xx_ota_backend:463 Setting boot partition to app_1
105279115 [0;32mI (105910) esp_image: segment 0: paddr=00200020 vaddr=3f400020 size=30d94h (200084) map[0m
105396272 [0;32mI (106030) esp_image: segment 1: paddr=00230dbc vaddr=3ffb0000 size=036d0h ( 14032) [0m
105413040 [0;32mI (106050) esp_image: segment 2: paddr=00234494 vaddr=40080000 size=0bb84h ( 48004) [0m
[... cut ...]
105918613 [0;32mI (106550) esp_image: segment 4: paddr=00312f48 vaddr=4008bb84 size=08ab8h ( 35512) [0m
105948466 [0;32mI (106580) esp_image: segment 5: paddr=0031ba08 vaddr=50000000 size=00010h (    16) [0m
105957830 [0;32mI (106590) esp_ota_ops: New OTA data 0: seq 0x00000049, st 0x11, CRC 0xcfba7c04[0m
106014083 mgos_ota_core.c:1051    Update finished, result 1 (Update applied, rebooting)
106020424 mgos_ota_core.c:1202    Update state: 0
106025216 mgos_ota_core.c:1096    Update requires reboot

Salle88, 2024-12-04:
Log from Update fron 2.13.2 ... same bug from 2.13.0 🤷‍♂️ 
Two devices from stock 1.4.4 same bug, after complete update - Renoir and the old software

markirb, 2024-12-07:
sorry, I had so far no time looking into this. 

here is a version without reading partition. can you see if that fixes it?
[shelly-homekit-ShellyPlus1PM.zip](https://github.com/user-attachments/files/18048854/shelly-homekit-ShellyPlus1PM.zip)

markirb, 2024-12-07:
oh wow... I think I finally spotted the error...  we set a reset handler on the LED.... I will fix that now...

Salle88, 2024-12-07:
It works !!! Thanks !

Salle88, 2024-12-07:
Update:

After reverting the last shelly to stock 1.4.4 and updating it with your file the update was succesfull. 🙏

closed 2025-01-06 by github-actions[bot]