mongoose-os-apps/shelly-homekit #1568: Shelly 1 Mini G3 Garage Door Opener - Never gets to open

htims1989, 2024-11-20:
I'm trying to use a Shelly 1 Mini G3 as a garage door opener. Opening and closing the door works, but the contact sensor doesn't seem to work correctly - it never gets into an "open" state.

For wiring I've got:
Live into L
Neutral into N
Garage door opener into O and I
Contact sensor to SW and N

The the magnet is close to the contact sensor when the door is closed (normally closed).

For settings:
Close sensor: Normally Closed
Open sensor: Disabled
Movement time: 20 seconds

When I trigger an open by pressing toggle (with the door closed) it starts opening and the status changes to opening but midway through the status changes to closed. If I press toggle again once the door is fully open, the status goes to opening and then closed but before the door has closed.

I captured this from the log in case it's useful:

```
3371627959 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.28:63825
3372631126 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.28:63825
3373358054 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.1.28:63825
3373363198 shelly_output.cpp:67    Output 1: off -> on (GDO:open)
3373369582 shelly_hap_garage_d:321 GDO 1: Tgt State: closed -> open (1 -> 0) (RPC)
3373375701 shelly_hap_garage_d:235 GDO 1: Cur State: closed -> opening (1 -> 2)
3373393328 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.28:63825
3373631126 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.28:63825
3373668602 shelly_output.cpp:67    Output 1: on -> off (pulse_off)
3375075897 shelly_main.cpp:492     Up 3375.07, HAP 0/1/16 ns 1, RAM: 203424/176912; st 37; 5.1: cur:opening tgt:open cl:1 op:-1
3380439052 shelly_hap_garage_d:321 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
3380445217 shelly_hap_garage_d:235 GDO 1: Cur State: opening -> closed (2 -> 1)
3383075982 shelly_main.cpp:492     Up 3383.07, HAP 0/1/16 ns 1, RAM: 203424/176912; st 37; 5.1: cur:closed tgt:closed cl:1 op:-1
3391075941 shelly_main.cpp:492     Up 3391.07, HAP 0/1/16 ns 1, RAM: 203424/176912; st 36; 5.1: cur:closed tgt:closed cl:1 op:-1
3399075916 shelly_main.cpp:492     Up 3399.07, HAP 0/1/16 ns 1, RAM: 203424/176912; st 35; 5.1: cur:closed tgt:closed cl:1 op:-1
3401778038 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.28:63825
3402635303 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.28:63825
3403626990 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.28:63825
3407075968 shelly_main.cpp:492     Up 3407.07, HAP 0/1/16 ns 1, RAM: 203424/176912; st 35; 5.1: cur:closed tgt:closed cl:1 op:-1
3415075969 shelly_main.cpp:492     Up 3415.07, HAP 0/1/16 ns 1, RAM: 203424/176912; st 34; 5.1: cur:closed tgt:closed cl:1 op:-1
3423075970 shelly_main.cpp:492     Up 3423.07, HAP 0/1/16 ns 1, RAM: 203424/176912; st 34; 5.1: cur:closed tgt:closed cl:1 op:-1
3431075979 shelly_main.cpp:492     Up 3431.07, HAP 0/1/16 ns 1, RAM: 203424/176912; st 33; 5.1: cur:closed tgt:closed cl:1 op:-1
3439075922 shelly_main.cpp:492     Up 3439.07, HAP 0/1/16 ns 1, RAM: 203424/176912; st 33; 5.1: cur:closed tgt:closed cl:1 op:-1
3441132129 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.28:63825
3441627635 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.28:63825
3442628158 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.28:63825
3443638668 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.28:63825
3444404942 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.1.28:63825
3444410071 shelly_output.cpp:67    Output 1: off -> on (GDO:open)
3444416452 shelly_hap_garage_d:321 GDO 1: Tgt State: closed -> open (1 -> 0) (RPC)
3444422756 shelly_hap_garage_d:235 GDO 1: Cur State: closed -> opening (1 -> 2)
3444441163 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.28:63825
3444626926 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.28:63825
3444715397 shelly_output.cpp:67    Output 1: on -> off (pulse_off)
3447075821 shelly_main.cpp:492     Up 3447.07, HAP 0/1/16 ns 1, RAM: 203424/176912; st 32; 5.1: cur:opening tgt:open cl:1 op:-1
3451439035 shelly_hap_garage_d:321 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
3451445353 shelly_hap_garage_d:235 GDO 1: Cur State: opening -> closed (2 -> 1)
```

 I'm not sure if this is a me/wiring issue or a bug. I have not tried it with stock firmware. Ideally I would be using a Shelly 1 and 12v DC but I had a spare mini and my meross opener kept dropping off wifi so looking to replace it!

markirb, 2024-11-21:
Could be this:

https://github.com/mongoose-os-apps/shelly-homekit/issues/1566

Can you try?

htims1989, 2024-11-21:
Switch and 5 flips didn't work either so think it's dead. I just pressed the "Revert" button, saw it progressing and get to 96% but then it never came back onto the network and the shelly SSID is not being broadcast.

htims1989, 2024-11-22:
I mean, I have tried all of the reset options mentioned here multiple times and I can't get it to broadcast an SSID:
https://support.shelly.cloud/en/support/solutions/articles/103000095415-how-to-factory-reset-a-shelly-device-

Also I have now installed and flashed another one (I had 2 spare) and it is acting exactly the same as the first, even after a power cycle. Is there any more detail I can provide?

markirb, 2024-11-22:
According to wire diagram of shelly you wiring is wrong for input. 

This is no advice. Be careful I am under impression you are using this on high voltage?!?

htims1989, 2024-11-23:
I took another look at the wiring and think I agree and have disconnected the sensor. I think I'll just order a Shelly 1 and run it on 12V.

closed 2024-11-23 by htims1989