mongoose-os-apps/shelly-homekit #1571: Plus1+Addon Garage Door sensor bug

mrautomations, 2024-11-23:
I have a Shelly Plus 1 with the Plus Addon.

Door sensor connected: GRD and DIGITAL IN

It doesn't work.

If I change from Garage Door to Switch, the sensor works fine!
But with Garage Door mode, it doesn't recognize the sensor.
I don't want to connect the sensor without the Addon because I am using 240VAC to power the Shelly.

markirb, 2024-11-23:
If you have closing sensor attached to addon then you need to use the swap sensor input option. Then it should work.

mrautomations, 2024-11-23:
Hello, thank you for the help but I can't test that out because another bug is driving me nuts.

I don't know if you want me to open another issue with this but my Shelly is losing connection with wifi as soon as I connect my Addon!

Everything works fine without the Addon, but if I attach it to the Shelly Plus 1, I can't connect to it anymore. Not the WiFi IP nor the internal AP with its own Wifi.

I have to detach the Addon and reset the Shelly to gain connection to its AP again.

I have two Shellys Plus 1 with Addon and both show this same behaviour. It's driving me nuts. :(

Yesterday both were working fine when I bought them and did the initial setup. Today – they were disconnected from power all night –, they are behaving like this.

markirb, 2024-11-23:
Can you try on stock firmware?

I doubt this is a sw issue I think maybe hardware/power or wiring but just in case...

mrautomations, 2024-11-23:
It works flawlessly with stock firmware.

mrautomations, 2024-11-23:
I think I know why it worked yesterday and not today.

I reverted to stock firmware as you said and it worked fine.
I installed mongose and it worked fine again like yesterday!
As soon as I powered it off, it never reconnected to wifi again.

So, yesterday it worked because I was coming from the upgrade from stock firmware, like today when I redid the procedure.

But as soon as I powered it off it somehow loses the ability to reconnect. But this only happens with the Addon attached.

markirb, 2024-11-23:
I see. Can you try if this all works in non-gdo mode? Maybe there is still a bug in gdo/sensor swap..

Would not know the difference elsewise.

mrautomations, 2024-11-23:
Sorry to only answer now.

The bug persists in Switch mode.

mrautomations, 2024-11-25:
Also, can you explain what "swap sensor input" does? Because it isn't mentioned in the documentation.

Thank you

markirb, 2024-11-25:
I added this to documentation.

For the actual bug, I currently do not have an Idea. The code for Shelly Plus 1PM and Plus 1 is totally the same regarding the add-on....

This has to be some Pin Conflict, or something similar.

mrautomations, 2024-12-29:
Is there anything I can do with my Shellys to help figure out this problem? I am not a programmer, but I can try to help with my hardware in some way if you give me the instructions.

Thanks

markirb, 2024-12-29:
I got an addon and an +1PM.

To be more precise: if you boot with the addon from power state: does it come up in AP mode or not at all? Which firmware version ist this again?

mrautomations, 2024-12-29:
Yes, I have a them too and never had a problem.

This bug is with Plus1 and the Addon.
Not at all. I have to reset it for it to work.
It's the last firmware version.

markirb, 2024-12-29:
But without reset if you take of the add-on it still does not work?

mrautomations, 2024-12-30:
If I remove the addon, the bug persists. A reset must be done to revive it.

I think I might know where the bug is but I need to test again in a few hours.

I think the bug only occurs when we inject current through NEUTRAL!
Is this possible?
If current is arriving through LINE, everything works fine. If we invert the plug in the wall socket, and the current now arrives through NEUTRAL at the Shelly, it bricks and never recovers until we reset it.
I will confirm this in a few hours.

mrautomations, 2024-12-30:
I made a few tests a few hours ago and it bricked. I did a reset.

Now, I am doing a lot of tests and it is working without a problem! 😭 I don't understand

mrautomations, 2024-12-30:
This is driving me crazy.

I am testing now with another Shelly 1 with addon and the bug is still there in this one.

It must be a small detail or thing i done differently.
Sometimes if i remove the addon the AP network appears and i don't need to reset it! But it is not consistent. 🤔