mongoose-os-apps/shelly-homekit #1573: No devices found in zeroconf? Can't flash Shelly 1 Gen 3?

bendinwire, 2024-11-25:
I have two Shelly 1 Gen 3s here running latest stock firmware 1.4.4. I'm trying to flash them with this firmware but it doesn't seem to see them as destinations when I query:

``` python3 /Users/tornado/Downloads/shelly-homekit-master/tools/flash-shelly.
py -la
Scanning for Shelly devices...
```

similarly when I try to do it via http `http://10.0.1.178/ota?url=http://shelly.rojer.cloud/update` it just gives me this response:
`Trying to update from http://shelly.rojer.cloud/update`

Any suggestions?

Thanks

markirb, 2024-11-25:
Suggestion is to wait until this model is supported 

If you have the Pinout then it will take only a few days

bendinwire, 2024-11-26:
I think the pinouts are on this page if that is what you're looking for?
https://kb.shelly.cloud/knowledge-base/shelly-1-gen3

[image]

markirb, 2024-11-27:
No, the internal pinout is important, this is how the esp chip inside is wired. This is not public information

markirb, 2024-12-07:
we have pinning available now. please see here for early alpha.
https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.14.0-alpha0
continue feedback in disunion thread please

closed 2024-12-07 by markirb