mongoose-os-apps/shelly-homekit #1576: Shelly RGBW2 - constant reboots on 2.13

volovva, 2024-11-27:


volovva, 2024-11-27:
20822925 shelly_debug.cpp:231    No log file, sending new entries
26034393 shelly_main.cpp:486     Up 26.01, HAP 0/1/12 ns 1, RAM: 21460/12464; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
34034499 shelly_main.cpp:486     Up 34.01, HAP 0/1/12 ns 1, RAM: 21460/12464; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
42034388 shelly_main.cpp:486     Up 42.01, HAP 0/1/12 ns 1, RAM: 21460/12464; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
50034259 shelly_main.cpp:486     Up 50.01, HAP 0/1/12 ns 1, RAM: 20916/12464; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
58034342 shelly_main.cpp:486     Up 58.01, HAP 0/1/12 ns 1, RAM: 21460/12464; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
66034298 shelly_main.cpp:486     Up 66.01, HAP 0/1/12 ns 1, RAM: 21460/12464; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
74034326 shelly_main.cpp:486     Up 74.01, HAP 0/1/12 ns 1, RAM: 21460/12464; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
82034170 shelly_main.cpp:486     Up 82.01, HAP 0/1/12 ns 1, RAM: 21460/12464; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400

(reset here)
22787040 mgos_mongoose.c:66      New heap free LWM: 19984
22830746 mgos_mongoose.c:66      New heap free LWM: 19752
22849999 mgos_mongoose.c:66      New heap free LWM: 19656
26121673 shelly_main.cpp:486     Up 26.10, HAP 0/1/12 ns 1, RAM: 22120/19656; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
34121802 shelly_main.cpp:486     Up 34.10, HAP 0/1/12 ns 1, RAM: 22120/19656; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
37275091 mgos_mongoose.c:66      New heap free LWM: 19456
40290782 mgos_mongoose.c:66      New heap free LWM: 19384
42121945 shelly_main.cpp:486     Up 42.10, HAP 0/1/12 ns 1, RAM: 22120/19384; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
44783016 mgos_http_server.c:180  0x3fff6a7c HTTP connection from 192.168.9.173:59846
44808833 mg_rpc.c:311            Shelly.GetInfoExt via WS_in 192.168.1.xxx:59846
44840789 mgos_mongoose.c:66      New heap free LWM: 11976
50121916 shelly_main.cpp:486     Up 50.10, HAP 0/1/12 ns 1, RAM: 21508/11976; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
58121982 shelly_main.cpp:486     Up 58.10, HAP 0/1/12 ns 1, RAM: 21508/11976; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400

volovva, 2024-11-27:
2475437 shelly_debug.cpp:231    No log file, sending new entries
2481777 shelly_output.cpp:73    Duty: 0.000
2486914 mgos_mongoose.c:66      New heap free LWM: 21248
2493289 shelly_output.cpp:73    Duty: 0.000
2498256 shelly_output.cpp:73    Duty: 0.810
2503886 shelly_output.cpp:78    Output 4: 0.810328 (transition)
2508562 shelly_output.cpp:73    Duty: 0.000
2512835 shelly_output.cpp:73    Duty: 0.000
2517050 shelly_output.cpp:73    Duty: 0.000
2521592 shelly_output.cpp:73    Duty: 0.822
2527117 shelly_output.cpp:78    Output 4: 0.822066 (transition)
2533152 shelly_output.cpp:73    Duty: 0.000
2537740 shelly_output.cpp:73    Duty: 0.000
2542200 shelly_output.cpp:73    Duty: 0.000
2546617 shelly_output.cpp:73    Duty: 0.835
2552356 shelly_output.cpp:78    Output 4: 0.834575 (transition)
2557204 shelly_output.cpp:73    Duty: 0.000
2562529 shelly_output.cpp:73    Duty: 0.000
2570565 shelly_output.cpp:73    Duty: 0.000
2575694 mgos_mongoose.c:66      New heap free LWM: 20584
2581993 shelly_output.cpp:73    Duty: 0.852
2587644 shelly_output.cpp:78    Output 4: 0.852068 (transition)
2592504 shelly_output.cpp:73    Duty: 0.000
2597177 shelly_output.cpp:73    Duty: 0.000
2602007 shelly_output.cpp:73    Duty: 0.000
2606635 shelly_output.cpp:73    Duty: 0.865
2612182 shelly_output.cpp:78    Output 4: 0.864572 (transition)
2616534 shelly_output.cpp:73    Duty: 0.000
2620929 shelly_output.cpp:73    Duty: 0.000
2625174 shelly_output.cpp:73    Duty: 0.000
2629608 shelly_output.cpp:73    Duty: 0.876
2635240 shelly_output.cpp:78    Output 4: 0.876080 (transition)
2640636 shelly_output.cpp:73    Duty: 0.000
2648615 shelly_output.cpp:73    Duty: 0.000
2653691 shelly_output.cpp:73    Duty: 0.000
2658144 shelly_output.cpp:73    Duty: 0.890
2663821 shelly_output.cpp:78    Output 4: 0.890328 (transition)
2668346 shelly_output.cpp:73    Duty: 0.000
2672578 shelly_output.cpp:73    Duty: 0.000
2678488 shelly_output.cpp:73    Duty: 0.000
2684784 shelly_output.cpp:73    Duty: 0.904
2690429 shelly_output.cpp:78    Output 4: 0.903504 (transition)
2695687 shelly_output.cpp:73    Duty: 0.000
2700553 shelly_output.cpp:73    Duty: 0.000
2704884 shelly_output.cpp:73    Duty: 0.000
2709299 shelly_output.cpp:73    Duty: 0.916
2714963 shelly_output.cpp:78    Output 4: 0.915920 (transition)
2719331 shelly_output.cpp:73    Duty: 0.000
2723561 shelly_output.cpp:73    Duty: 0.000
2729352 shelly_output.cpp:73    Duty: 0.000
2737777 shelly_output.cpp:73    Duty: 0.930
2743668 shelly_output.cpp:78    Output 4: 0.929913 (transition)
2748568 shelly_output.cpp:73    Duty: 0.000
2752908 shelly_output.cpp:73    Duty: 0.000
2757411 shelly_output.cpp:73    Duty: 0.000
2762015 shelly_output.cpp:73    Duty: 0.942
2767551 shelly_output.cpp:78    Output 4: 0.942271 (transition)
2771902 shelly_output.cpp:73    Duty: 0.000
2776871 shelly_output.cpp:73    Duty: 0.000
2782772 shelly_output.cpp:73    Duty: 0.000
2787385 shelly_output.cpp:73    Duty: 0.955
2793053 shelly_output.cpp:78    Output 4: 0.954926 (transition)
2800218 shelly_light_bulb_co:55 Transition finished, end state: [w=0.00]
2804034 shelly_output.cpp:73    Duty: 0.000
2809765 shelly_output.cpp:73    Duty: 0.000
2817762 shelly_output.cpp:73    Duty: 0.000
2822726 shelly_output.cpp:73    Duty: 0.973
2828272 shelly_output.cpp:78    Output 4: 0.972589 (transition)
2835786 shelly_light_bulb_co:55 Transition finished, end state: [w=0.00]
2839626 shelly_output.cpp:73    Duty: 0.000
[... 8 lines cut ...]
2889464 shelly_output.cpp:73    Duty: 1.000
2895014 shelly_output.cpp:78    Output 4: 1.000000 (transition)
3185871 mgos_mongoose.c:66      New heap free LWM: 20360
3244070 mgos_sntp.c:63          SNTP reply from 216.239.35.8: time 1732724200.440331, local 3.231375, delta 1732724197.208956
4060259 HAPPlatformTCPStrea:121 0x3fff62cc 192.168.9.103:64991 Incoming HAP connection, ns 0/0/12
4070298 HAPPlatformTCPStrea:276 0x3fff62cc 192.168.9.103:64991 Accepted HAP connection, ns 0/1/12 ts 0x3fff3f3c
5172713 mgos_mongoose.c:66      New heap free LWM: 20216
5191368 mgos_mongoose.c:66      New heap free LWM: 20168
5384537 mgos_mongoose.c:66      New heap free LWM: 19880
5555585 mgos_mongoose.c:66      New heap free LWM: 19560
6113529 mgos_http_server.c:180  0x3fff6354 HTTP connection from 192.168.9.173:60072
9214955 mgos_sys_config.c:323   Saved to conf9.json
9387999 mgos_sys_config.c:323   Saved to conf9.json
9561066 mgos_sys_config.c:323   Saved to conf9.json
9798855 mgos_sys_config.c:323   Saved to conf9.json
9825119 shelly_main.cpp:486     Up 9.80, HAP 0/1/12 ns 1, RAM: 21372/18912; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
9831106 mgos_mongoose.c:66      New heap free LWM: 18912
10890774 mgos_sys_config.c:323   Saved to conf9.json
12119643 esp_main.c:138          SDK: pm open,type:0 0
16999818 shelly_main.cpp:486     Up 16.98, HAP 0/1/12 ns 1, RAM: 21740/18912; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
24999742 shelly_main.cpp:486     Up 24.98, HAP 0/1/12 ns 1, RAM: 21740/18912; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
[... cut ...]
40999597 shelly_main.cpp:486     Up 40.98, HAP 0/1/12 ns 1, RAM: 21740/18912; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
48999701 shelly_main.cpp:486     Up 48.98, HAP 0/1/12 ns 1, RAM: 21740/18912; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: on, b: 100, h: 0, sa: 100, ct: 400
56999905 shelly_main.cpp:486     Up 56.98, HAP 0/1/12 ns 1, RAM: 21740/18912; st 0; 11.1: sta: off, b: 0, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100

markirb, 2024-11-28:
Oh. I See. There was a fix which was accidentally deleted for plus devices because there the fix did not work. Will be fixed again

markirb, 2024-12-03:
fixed in 2.13.3 (again)