mongoose-os-apps/shelly-homekit #1583: OTA flashing of Shelly +2PM not working (FW 0.10.2-beta4)

mattchrocha, 2024-12-03:
Hello, I am trying to flash Shelly +2PM (FW 0.10.2-beta4) via OTA (.../ota?url=http://shelly.rojer.cloud/update) and it is not working.
I think this could be similar to issue #1528 , so I fired up netcat and this is the output, in case this helps:
GET / HTTP/1.1
Host: 192.168.1.36:8080
Content-Length: 0
Range: bytes=0-
Connection: close
X-MGOS-Device-ID: shellyplus2pm-5443b23f5f44 5443B23F5F44
User-Agent: Plus2PM/0.10.2-beta4 (MongooseOS/202202232310; esp32)
X-MGOS-FW-Version: esp32 0.10.2-beta4 20220527-091739/0.10.2-beta4-gecc3a61

mattchrocha, 2024-12-03:
I did click on "check for update" multiple times , and the message was always "the device is up to date". I am looking for it, but even via the API or trying to indicate the zip for FW 1.3.3 via ?ota= doesnt do anything either :(

markirb, 2024-12-03:
Please ask Shelly support.
If you cannot update stock then you cannot update to this firmware

mattchrocha, 2024-12-04:
Problem solved, i used the official Shelly phone app to properly update the FW. I am sorry for the inconvenience!! Thank you very much for the help. Closing this issue.

closed 2024-12-04 by mattchrocha