mongoose-os-apps/shelly-homekit #1595: Not able to update shellyuni from 20221027-121732/v1.12.1-ga9117d3

chef76, 2024-12-19:
Hi,

 i have done this 5 times without problem, but today i`m not able to update my shellyuni anymore.
 The answer to following request: http://A.B.C.D/ota?url=http://rojer.me/files/shelly/2.13.3/shelly-homekit-ShellyUNI.zip is {"status":"updating","has_update":false,"new_version":"","old_version":"20221027-121732/v1.12.1-ga9117d3"}

Looks like it doesn't find the new firmware?!? Or isn't it possible to update from this version?

Can somone please help me?

markirb, 2024-12-21:
Please update to latest stock version first this should be v1.14.0-gcb84623

chef76, 2024-12-21:
It doesn't offer me that firmware?!?

[image]

markirb, 2024-12-21:
you can try this tool: http://archive.shelly-tools.de/

chef76, 2024-12-21:
Tranks for your help, but this doesn't work either:

http://192.168.2.144/ota?url=http://archive.shelly-tools.de/version/v1.14.0/SHUNI-1.zip

Same result:

[image]

markirb, 2024-12-21:
Did you reset already?

chef76, 2024-12-21:
Reset, cold start, nothing worked. Took a new one out of the box and it worked first try.

Thanks for your help!

closed 2024-12-21 by chef76

chef76, 2024-12-21:
After a Factory Reset and Setting it up from scratch, it offered the  new firmware version and update to    HomeKit worked 🥳