mongoose-os-apps/shelly-homekit #1602: Wiki image for WiFi Settings out of date

jsiegenthaler, 2024-12-31:
The Wiki image /resources/Wifi-Settings.png in the following page does not represent the current v2.13.3 UI
https://github.com/mongoose-os-apps/shelly-homekit/wiki/WiFi-Settings

Can someone please update the Wifi-Settings.png image
I would but I don't know where the /resources folder is located...

markirb, 2024-12-31:
Thanks for you efforts.
There may be more outdated information.

Can you try?
https://github.com/mongoose-os-apps/shelly-homekit.wiki.git
I think you can find them in this repository: (URL is on sidebar).

jsiegenthaler, 2025-01-02:
@markirb 
That url gives 404 page not found
The wiki is here: https://github.com/mongoose-os-apps/shelly-homekit/wiki

All the following are not found:
https://github.com/mongoose-os-apps/shelly-homekit/resources 404
https://github.com/mongoose-os-apps/shelly-homekit/wiki/resources 404
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Firmware/resources rolls back to wiki

jsiegenthaler, 2025-01-02:
@markirb maybe the resources folder is in the main file tree but not visible to me?

markirb, 2025-01-02:
The url I posted is a git repository which you have to clone via git clone

jsiegenthaler, 2025-01-02:
Thanks. I cloned the repo, updated the png file, made a new branch, went to publish the branch and github tells me I have no permissions to publish a new branch to shelly-homekit.wiki

jsiegenthaler, 2025-01-03:
I have no permissions to create new branches (Updated-WiFi-Settings-image-in-Wiki) nor to update the master branch. 
[image]
However on other projects I am working on I can create branches and publish without any problem.

markirb, 2025-01-06:
also on other wikis? I do not think wikis can handle branches. only commits to master

jsiegenthaler, 2025-01-08:
I do not have any other wikis managed via cloned github repos. 
Honestly, I give up. 
Someone else can invest the time and effort to update the images.

markirb, 2025-01-08:
Upload it here then I will change it

jsiegenthaler, 2025-01-08:
[image]

markirb, 2025-01-10:
thanks. I uploaded this. And I corrected power saving description

closed 2025-01-10 by markirb