mongoose-os-apps/shelly-homekit #1603: HomeKit:  Shelly Duo – RGBW GU10 - “This accessory not responding.”

dgvinton, 2025-01-01:
The first time I flashed the bulb the I was able to manage it through the HomeKit UI.  After a day or two, HomeKit reported “This accessory not responding.”  The web UI worked just fine and reported a good WiFi connection.  Rebooting the bulb and kicking it off the WiFi to reconnect made no difference.

I restored the bulb to factory settings, re-flashed, and re-added it to HomeKit.  This time the HomeKit UI immediately reported  “This accessory not responding.”

I have 20+ Shelly Duo Bulbs in use and have not experienced anything like this.  This my first GU10 based Shelly bulb.  

Please advise,

Don

markirb, 2025-01-11:
Did you solve this? What is your wifi power saving at?

dgvinton, 2025-01-12:
Hi Markus,

Thank you for following up. It's now working, but I don't know if I solved
the problem.

The steps taken to get the GU10 working are as follows:

1) Power on the GU10 and browse its webpage.  Power saving was set to 2.
The Home App reports it's not responding.
2) I set the power saving value to 0.  The Home App reports it's not
responding.
3) I reset the GU10 HomeKit Setting and removed the device from the Home
App.
4) With Power saving set to 0, I attempted to add the GU10.  The Home App
reported, "Unable to Add Accessory."
5) With Power saving set back to 2, I successfully added the GU10 back to
Home.

So, at this point, the GU10 is working.

If/when this happens again, what further troubleshooting steps do I need to
take?

Best regards,

Don